arendst/Tasmota #7402: Floating point number for moisture sensor

robin13, 2020-01-02:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  

This is related to https://github.com/arendst/Tasmota/issues/7182 and https://github.com/arendst/Tasmota/pull/7309
The moisture is given as a percentage, but the percentage is rounded to an integer.  In many situations (e.g. if the analogue range is wet -> dry 720 -> 200), a mere 100 steps is loosing a lot of granularity.  As the total possible range for the analogue input is 0->1023, it would be good to have one decimal point after the value to give the full possible granularity of the range.
The issue is compounded that with a moisture sensor which has 200 for wet and 720 for dry, the effective useful range for a pot plant may be only 300 -> 400.  Without further precision this results in only 19 steps in the range where there could be 100.

**Describe the solution you'd like**  

I think about here (https://github.com/arendst/Tasmota/pull/7309/commits/c1f30da81ee71fcfb9b293997fb4d4b1dc75fe7f#diff-346bd9707370e3b0529b922b797ef43bR122) it would be necessary to use a float rounded to one decimal place

robin13, 2020-01-03:
@effelle Maybe you are also interested in this one? :)

effelle, 2020-01-03:
I'm on vacancy :P
Anyway, all `Adcparam` commands work with integer result, even if we use double precision as input values. I'll have a look but is not on my priority list.

joba-1, 2020-01-04:
I would definitely try to avoid floats. How about using promille instead?

GrendelWWU, 2020-01-06:
As far, as I understand this, we're talking about a moisture sensor connected to A0? So, I guess, the best practice will be using the raw data 0-1023. Or you can try this sensor via I2C: https://www.tindie.com/products/miceuz/i2c-soil-moisture-sensor/
(Yes, I know, you'll probably need to compile the stuff yourself....)

effelle, 2020-01-06:
@joba-1 yes, if I'll change it I will use a per mille. 

@GrendelWWU I prefer use the pure Analog value, but some users from time to time ask for a direct conversion/calibration similar to what we have added for analog LDR and TEMP. Normally is a better implementation for HAB. 

`Aadcparam 6` is simple and even if I understand Robin's request I'm not sure I'll add a more precise version. I'll think on it.
For now if you want a better calibration `scalex` and a `var` on a rule is still the best choice. 
What do you think @ascillato?

ascillato, 2020-01-06:
> For now if you want a better calibration scalex and a var on a rule is still the best choice.
What do you think @ascillato?

I agree. As a workaround for now, a rule can be used to set the range that is needed. Also, other workaround is to read A0 pin and do the calculations on a home automation software.

An idea for an optimization to add 1 decimal precision to the moisture conversion and not adding more flash size (due to float variables), can be just **multiplying the input of adc by 10** ( 0-1024 to 0-10240 range - as the actual variables are uint16 - limit 65536 ), then do the **scale conversion**, then take the **lowest digit as decimal** and **divide by 10** the result, having as ouput: [scaledvalue.decimal]

What do you think?

effelle, 2020-01-06:
Could be the right call. I'll put it on optimization queue list.

stefanbode, 2020-01-07:
First the moisture sensors are more a guess than just a correct measurement. I assume you will not get any additional value with more digits. Anyhow the easiest way to implement this is to set units from percent to per Mille. In this case the base is 1000. I assume this will have a minimum impact on the code.

effelle, 2020-01-07:
> First the moisture sensors are more a guess than just a correct measurement.
Is what I'm saying Stefan and is why is on low priority for now.

robin13, 2020-01-07:
Thanks for the thoughts and considerations here! :)
Taking all that into account, I'm also :+1:  for the per mille solution!

sfromis, 2020-01-17:
While such a moisture sensor most probably is far from having a precision needing decimals, I suppose that the point still could be to be able to see a more smooth curve, like when I like to plot values with two decimals from a temperature sensor which is only correct within a margin of +/- 0.5 degrees. A small change is likely to be of interest even while accepting that the value may be an order of magnitude more off. So I like that Tasmota has the TempRes command :-)

mike2nl, 2020-01-18:
As addition to a sensor via I2C.
Link:
https://www.adafruit.com/product/4026

Or via google images with more types of i2c soil sensors.
Link:
https://www.google.com/search?rlz=1C1VFKB_enNL689NL689&biw=1163&bih=932&tbm=isch&sxsrf=ACYBGNRuMqzy_0LBXX4VYfNX8RLrB18tWw%3A1579363146417&sa=1&ei=SisjXqWWGcXKkwXYqJCgDg&q=moisture+sensor+i2c&oq=moisture+sensor+i2c&gs_l=img.3..0i8i30.57378.58519..59077...0.0..0.169.754.0j6......0....1..gws-wiz-img.......35i39j0j0i67j0i24.wH70dWJ6tCM&ved=0ahUKEwil48Cawo3nAhVF5aQKHVgUBOQQ4dUDCAc&uact=5

arendst, 2020-01-23:
Update moisture calculation using ``AdcParams 6,0,1023,0,100`` and below formula:

``
moisture = ( ((double)Settings.adc_param2 - (double)adc) / ( ((double)Settings.adc_param2 - (double)Settings.adc_param1)) * ((double)Settings.adc_param3 - (double)Settings.adc_param4) + (double)Settings.adc_param4 );
``

ascillato2, 2020-01-23:
Closing this issue as the feature has been added.

Thanks :+1:

closed 2020-01-23 by ascillato2