arendst/Tasmota #7404: rc_switch bug

to-scho, 2020-01-02:
### PROBLEM DESCRIPTION
There is a tiny bug in the rc_switch library that prevents the usage for "Conrad RS-200" based devices. Howto handle such bug? Should I do a PR to rc_switch repo and at some time it's mirrored to Tasmota? Or do you want a PR to tasmota directly?

-- Issue description --
In "void RECEIVE_ATTR RCSwitch::handleInterrupt()" there is in line 672 

"if (diff(duration, RCSwitch::timings[0]) < 200) {" 

which does not handle the gap between two transmission right when RF packet is received first time . An easy fix is 

"if ((repeatCount==0) || (diff(duration, RCSwitch::timings[0]) < 200)) {" 

Furthermore the "Conrad RS-200" protocol should be added to the "PROGMEM proto[] = {..." starting at line 78:

"
...
  { 150, {  2, 62 }, {  1,  6 }, {  6,  1 }, false },    // protocol 7 (HS2303-PT, i. e. used in AUKEY Remote)
  { 200, {  3, 130}, {  7, 16 }, {  3,  16}, false},     // protocol 8 Conrad RS-200 RX
  { 200, { 130, 7 }, {  16, 7 }, { 16,  3 }, true}       // protocol 9 Conrad RS-200 TX
};"

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic
- [x] Tasmota binary firmware version number used: 8.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-01-02:
> Should I do a PR to rc_switch repo and at some time it's mirrored to Tasmota? Or do you want a PR to tasmota directly?

As you have found and solved the bug, you should make a PR to both, Tasmota and rc_switch repo in order to fast fix both ends.

Thanks for sharing your findings :+1:

ascillato2, 2020-01-02:
Closing this issue as it is solved.

Thanks :+1:

closed 2020-01-02 by ascillato2