arendst/Tasmota #7410: MI desk lamp crashes

Sonne12, 2020-01-02:
### PROBLEM DESCRIPTION
I flased Tasmota 8.1.0.1 as a MI Desk Lamp. I can see the counter values in the serial log but after a few seconds of playing with the rotary encoder it crashes and restarts. After a few restarts Wifi is also dead and i need to reflash the firmware.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP12F Module
- [x] Tasmota binary firmware version number used: 8.1.0.1
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: PlatformIO/Visual Studio Code
- [x] Flashing tools used: flash download tool 3.6.7
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
16:14:08 RSL: stat/tasmota/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
16:14:08 RSL: stat/tasmota/RESULT = {"Module":{"66":"Mi Desk Lamp"}}

  Configuration output here:
16:14:08 RSL: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"37":"PWM1"},"GPIO5":{"38":"PWM2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"150":"Rotary1a"},"GPIO13":{"151":"Rotary1b"},"GPIO14":{"17":"Button1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
16:15:31 RSL: stat/tasmota/STATUS = {"Status":{"Module":66,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:15:31 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Exception","Uptime":"0T00:48:00","StartupUTC":"2020-01-02T14:27:31","Sleep":50,"CfgHolder":4617,"BootCount":6,"SaveCount":49,"SaveAddress":"FB000"}}
16:15:31 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0.1(tasmota)","BuildDateTime":"2020.01.01 23:25:08","Boot":6,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"324/699"}}
16:15:31 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!Box 7430 YX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
16:15:31 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":568,"Free":432,"Heap":27,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000407","8FDAE397","043683A1","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
16:15:31 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-7394","IPAddress":"192.168.1.103","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:F3:EB:FB:FC:E2","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:15:31 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FBFCE2","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
16:15:31 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 02 15:15:31 2020","Local":"Thu Jan 02 16:15:31 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:04"}}
16:15:31 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-01-02T16:15:31"}}
16:15:31 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-01-02T16:15:31","Uptime":"0T00:48:00","UptimeSec":2880,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":19,"Color":"250B","HSBColor":"0,0,0","Channel":[15,4],"CT":234,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"FRITZ!Box 7430 YX","BSSId":"98:9B:CB:7A:79:DE","Channel":1,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:06"}}}
16:15:31 RSL: stat/tasmota/STATUS12 = {"StatusSTK":{"Exception":0,"Reason":"Exception","EPC":["402061f4","00000000","400043e6"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40100eb6","40100f44","401009e4","40100e7c","4000050c","400043df","400044ab","40104952","40104dc3","40246790","4010488b","40103750","40216bc8","40000020","40103438","402467a0","4000444e","401068c7","4000410f","40004a3c","40106b58","4020c679","4023dbe6","4021cdc4","402194c3","40229098","40100d64","4020375b","402293ca","40240ab4","40100949"]}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
00:00:00 APP: Set Serial to 8N1 115200 bit/s
00:00:00 Projekt tasmota Tasmota Version 8.1.0.1(tasmota)-2_6_1
00:00:00 WIF: verbinden mit AP1 FRITZ!Box 7430 YX in Modus 11N als tasmota-7394...
00:00:05 WIF: verbunden
00:00:05 HTP: Web-Server aktiv bei tasmota-7394 mit IP-Adresse 192.168.1.103
15:26:15 RSL: stat/tasmota/RESULT = {"POWER":"ON"}
15:26:15 RSL: stat/tasmota/POWER = ON
15:26:24 RSL: stat/tasmota/RESULT = {"POWER":"OFF"}
15:26:24 RSL: stat/tasmota/POWER = OFF
15:26:26 RSL: stat/tasmota/RESULT = {"POWER":"ON"}
15:26:26 RSL: stat/tasmota/POWER = ON
15:26:33 RSL: stat/tasmota/RESULT = {"POWER":"OFF"}
15:26:33 RSL: stat/tasmota/POWER = OFF
15:26:40 RSL: stat/tasmota/RESULT = {"POWER":"ON"}
15:26:40 RSL: stat/tasmota/POWER = ON

Exception (0):
epc1=0x402061f4 epc2=0x00000000 epc3=0x400043e6 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffbb0 end: 3fffffc0 offset: 01a0
3ffffd50:  40100eb6 056e25d1 00000000 40100f44
3ffffd60:  e0035005 00000000 0000001f 401009e4
3ffffd70:  000005e0 00000000 3fffc228 00000022
3ffffd80:  3fffc200 40100e7c 3fffc258 4000050c
3ffffd90:  400043df 00000030 00000016 ffffffff
3ffffda0:  400044ab 3fffc718 3ffffe90 08000000
3ffffdb0:  60000200 08000000 00000003 00000000
3ffffdc0:  0000ffff 00042035 00002035 000fa000
3ffffdd0:  3fff1cb4 00000000 3ffeff78 00000030
3ffffde0:  00000001 40104952 3ffed9e0 ffffffff
3ffffdf0:  40104dc3 3ffed9e0 3ffed110 40246790
3ffffe00:  3ffffe20 3ffffe10 0000000c 00040000
3ffffe10:  4010488b 00000021 00000000 0000000f
3ffffe20:  0000001b 00000020 00000000 00040000
3ffffe30:  40103750 3fffff2a 00000003 40216bc8
3ffffe40:  3ffffeb0 3ffffea0 00000008 3fffc278
3ffffe50:  323a3531 32333a37 40000020 3fffc278
3ffffe60:  40103438 3fffc200 00000022 402467a0
3ffffe70:  00000000 4000444e 66e978d5 3fff1a62
3ffffe80:  00000002 401068c7 00000001 60000200
3ffffe90:  00000002 4000410f 80000000 00000305
3ffffea0:  3fffc718 40004a3c 000000fa 3fff1cb4
3ffffeb0:  3fffc718 40106b58 000000fa 4020c679
3ffffec0:  4023dbe6 3fff2bb4 3fff1c20 4021cdc4
3ffffed0:  30304538 30303839 30303030 402194c3
3ffffee0:  00001000 3fff1b00 00002623 00ffeb00
3ffffef0:  3ffffeed 00000000 3ffef8cc 3fffff10
3fffff00:  3fff1cb4 3fff1b68 3fff1b8c 40229098
3fffff10:  00000000 00000000 4bc6a7f0 00000000
3fffff20:  0098008e 00000000 40100d64 00016f99
3fffff30:  00000000 00000000 3ffef8a0 3fff310c
3fffff40:  01a90922 3fffff34 00000000 000003ff
3fffff50:  3fffdad0 00000000 00000000 3ffe86cc
3fffff60:  3fff1a4d 3fff1be0 00017091 4020375b
3fffff70:  00000004 0000004c 000000fa 3fff310c
3fffff80:  3fffdad0 3fff1be0 00016f97 402293ca
3fffff90:  3fffdad0 00000000 3fff30cc 3fff310c
3fffffa0:  3fffdad0 00000000 3fff30cc 40240ab4
3fffffb0:  feefeffe feefeffe 3ffe8700 40100949
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d

```

### TO REPRODUCE
using the rotary encoder

### EXPECTED BEHAVIOUR
no crash

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-01-02:
What happens if you keep the light intensity low resulting in less power usage; exception 0 mostly looks like a power issue.

Sonne12, 2020-01-02:
It's the same. I also disconnected my test LED. If I rotate slower, it wil
crash faster.

arendst, 2020-01-03:
Pls re-try using the latest dev with above commit.

s-hadinger, 2020-01-03:
Theo, I also pushed `LightPowerIRAM()` variant to IRAM so the interrupt handler runs fully in IRAM

closed 2020-01-03 by arendst

Sonne12, 2020-01-03:
Thanks, now it works like expected!

PR #7426 (2020-01-03): Add LightPowerIRAM() for rotary interrupt handler
## Description:

Commit b82d1fdcc3fdf53d33becb38ed2b93ad4a022152, the interrupt handler still called `LightPower()` which is not in IRAM. I added `LightPowerIRAM()` variant in IRAM. Only rotary uses it, the linker will evict this function from IRAM if not used elsewhare.

**Related issue (if applicable):** fixes #7410 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [ [... cut]