arendst/Tasmota #7412: poweronstate 3 doesn't work in Tuya Dimmer

Smandurlo, 2020-01-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

ERC309 zemismart receiver, flashed with tasmota 8.1.0 as shown in the wiki. Everything works as expected, but the poweronstate 3 doesn't work. If the switch is ON, restarting the device the switch remains ON, but after a power loose, the switch stays OFF.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): ERC309
- [ ] Tasmota binary firmware version number used: tasmota.bin 8.1.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: espeasy
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
{"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
stat/tasmotazemi/RESULT = {"Module":{"54":"Tuya MCU"}}
stat/tasmotazemi/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"107":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"108":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
stat/tasmotazemi/STATUS = {"Status":{"Module":54,"FriendlyName":["Tasmota"],"Topic":"tasmotazemi","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:46:57 MQT: stat/tasmotazemi/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:07:39","StartupUTC":"2020-01-02T16:39:18","Sleep":50,"CfgHolder":4617,"BootCount":2,"SaveCount":13,"SaveAddress":"F7000"}}
17:46:57 MQT: stat/tasmotazemi/STATUS2 = {"StatusFWR":{"Version":"8.1.0(lite)","BuildDateTime":"2019-12-25T12:36:09","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"336/699"}}
17:46:57 MQT: stat/tasmotazemi/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["VillaTorreZisa",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
17:46:57 MQT: stat/tasmotazemi/STATUS4 = {"StatusMEM":{"ProgramSize":462,"Free":540,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":3,"Features":["00000809","8F082383","041083A0","20021706","010001C0","00000000"],"Drivers":"1,2,3,4,9,10,16,20,21","Sensors":"3"}}
17:46:57 MQT: stat/tasmotazemi/STATUS5 = {"StatusNET":{"Hostname":"tasmotazemi-1244","IPAddress":"192.168.178.21","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"EC:FA:BC:4B:84:DC","Webserver":2,"WifiConfig":4}}
17:46:57 MQT: stat/tasmotazemi/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.26","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4B84DC","MqttUser":"Felice","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
17:46:57 MQT: stat/tasmotazemi/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 02 16:46:57 2020","Local":"Thu Jan 02 17:46:57 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:04"}}
17:46:57 MQT: stat/tasmotazemi/STATUS10 = {"StatusSNS":{"Time":"2020-01-02T17:46:57"}}
17:46:57 MQT: stat/tasmotazemi/STATUS11 = {"StatusSTS":{"Time":"2020-01-02T17:46:57","Uptime":"0T00:07:39","UptimeSec":459,"Vcc":3.386,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"VillaTorreZisa","BSSId":"E0:28:6D:80:9A:CD","Channel":1,"RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
turn ON the switch, unplug it from the electricity, plug in again. The switch stays off

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
It should turn on

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Smandurlo, 2020-01-02:
there is a topic with similar problem, but it is a dimmer and the problem is connected with the brightness: https://github.com/arendst/Tasmota/issues/7225
I believe they are connected somehow

stefanbode, 2020-01-07:
There is an option for GPIO Probe and changing the relay to the recognized state. I assume this is switching off your device. Let me check in the wiki which option might help  here. I have disabled the probe in all devices.

stefanbode, 2020-01-07:
SetOption63 Is your friend. Please look at wiki what it do.

Smandurlo, 2020-01-07:
hello, thanks for your reply. Setoption63 was 0. Tried with 1 but the problem isn't fixed: if the relay is ON, after power up, the relay stays OFF. If I restart the device the relay stays ON.

Remeslenik, 2020-01-08:
I have the same problem. The problem is that the dimmer consists of two controllers. Wifi module and controller that tracks crosszero and directly controls dimming. They communicate through UART under the TUYA protocol. As you can see from the logs, when the Wifi module starts, it does not transmit the current state to the second controller.

00:00:00 TYA: Send "55aa0001000000" **- Product ID Request**
00:00:00 TYA: Set WiFi LED 2 (-1)
00:00:00 TYA: Send "55aa000300010205" **- WIFI is configured but not connected to the router**
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Chulan in mode 11N as tasmota-0144...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on tasmota-0144 with IP address 192.168.0.104
00:00:05 RSL: INFO1 = {"Module":"RJWF-02A","Version":"7.1.2.6(7aab540-tasmota)","FallbackTopic":"cmnd/DVES_3D0090_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:05 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota-0144","IPAddress":"192.168.0.104"}
00:00:05 RSL: INFO3 = {"RestartReason":"Software/System restart"}
00:00:05 RSL: RESULT = {"POWER":"OFF"}
00:00:05 RSL: POWER = OFF
00:00:05 TYA: Set WiFi LED 3 (0)
00:00:05 TYA: Send "55aa000300010306" **- WIFI is configured and connected to the router**

17:23:51 TYA: Send "55aa00000000ff" **- Start heartbeat**

There is no data transfer to adjust the dimmer. I think this is a firmware problem for TUYA.

Smandurlo, 2020-01-08:
I don't even know if the original tuya firmware has the same problem too,
never used with it. Should we wait for a tasmota fix or it is a hardware
problem that cannot be fixed?

Remeslenik, 2020-01-08:
Need tasmota fix. But I don’t know if the developers will fix the problem. 
As a temporary solution you can use the rules:

```
Rule1 ON System#Boot DO event toggling=%mem2% ENDON
on event#toggling=0 do  TuyaSend1 1,0 endon
on event#toggling=1 do  TuyaSend2 2,%mem1% endon

Rule2 ON Dimmer#State DO mem1 %value% ENDON
ON Power1#State DO mem2 %value% ENDON
```

Smandurlo, 2020-01-08:
thanks for your help, but I tried to use those rules and they don't work.
Additionally my erc09 started to bootlool. I will wait for a fix eventually
and I will live happily without the restore status on power lost :-)
Luckily it is not a real problem in my area!

Remeslenik, 2020-01-08:
They work 100%. Problems with the syntax? Try it like this:
Rule1 ON System#Boot DO event toggling=%mem2% ENDON on event#toggling=0 do Backlog TuyaSend2 2,%mem1%; TuyaSend1 1,0 endon on event#toggling=1 do Backlog TuyaSend2 2,%mem1%; TuyaSend1 1,1 endon ON Dimmer#State DO mem1 %value% ENDON ON Power1#State DO mem2 %value% ENDON

Smandurlo, 2020-01-08:
mine is not a dimmer, just on/off, the dimmer is the erc310. Maybe this is
the problem?

Remeslenik, 2020-01-08:
Then try like this:
Rule1 ON System#Boot DO event toggling=%mem1% ENDON on event#toggling=0 do TuyaSend1 1,0 endon on event#toggling=1 do TuyaSend1 1,1 endon on Power1#State DO mem1 %value% ENDON

Smandurlo, 2020-01-10:
yep! This one works perfectly. Thank you :-)

meingraham, 2020-01-10:
```
Rule1
  on System#Boot do event toggling=%mem1% endon
  on event#toggling do TuyaSend1 1,%value% endon
  on Power1#State do mem1 %value% endon
```

Remeslenik, 2020-01-10:
Yes, this is more rational, thanks. But all this does not solve the problem. Tasmota, at start it does not transmit the power state and the dimmer state to TUYAmcu. Need a fix.

meingraham, 2020-01-10:
@shantur ⏫

shantur, 2020-01-10:
Its a Tasmota bug.
Seems like Tasmota is executing poweronstate a bit to early in initialization process. As seen in logs too, TuyaMCU module is initialized after poweronstate is executed.
I will try to get a PR for the fix

shantur, 2020-01-10:
@Smandurlo : Can you please try the attached firmware and see if it fixes your issue.

[firmware.bin.zip](https://github.com/arendst/Tasmota/files/4048287/firmware.bin.zip)

Remeslenik, 2020-01-10:
I can not flash through the console. I'll test firmware for dimmer tomorrow. I unsolder and flash directly.

meingraham, 2020-01-10:
@Remeslenik 

Download `http://thehackbox.org/tasmota/tasmota-minimal.bin` and perform an OTA File upload of this file. Then do the same with the firmware file @shantur provided. You should not need to serial flash your device.

Remeslenik, 2020-01-11:
@meingraham Thanks, flashed. 

@shantur 
For @Smandurlo should work. But with a dimmer problem. Previously, this is due to the fact that when the tasmota sends the dimmer data, it turns on.
It is now 3 a.m. Tomorrow I will study and attach logs. Thanks for the work.

Smandurlo, 2020-01-11:
Guys, since there is this difference between dimmer and on/off model (mine
was boot looping with a rule for dimmer) and it is installed behind a wall
mirror not easy to remove, I would love to test it, but only if I am sure
100% that is not going to make any crazy behaviour with the a non dimmer
model like mine.
Could you please confirm?

Thanks!

Remeslenik, 2020-01-11:
I tested on a dimmer and wrote in my topic #7225

multijohn, 2020-01-12:
Now with 8.1.0 i have a similar problem, but only when the switch is in off state.
I have a classic european on/off switch with a micro module moes wifi relay (tuya).
to explain,
When the switch is in off position and turn on the lamp from homeassistant, if i cut the power and back on again, the lamp goes always at off state!
If i do the same with the switch in on position and turn off the lamp from homeassistant , after power comes, lamp is off and if do the same with on command from homeassistant, after power recover lamp always is on.

my settings

poweronstate 3
switchmode 0

ascillato2, 2020-01-15:
@Smandurlo 

Hi, any news on this? Have you tried latest version from development branch? Some fixes have been added to the codebase.

Smandurlo, 2020-01-15:
I tried the 8.1.0.3 and it doesn't fix the problem.

ascillato2, 2020-02-14:
Not a bug

It is designed to not change states on a restart. Only on a power cycle. 

See https://github.com/arendst/Tasmota/issues/7737#issuecomment-586283615

closed 2020-02-14 by ascillato2

FourFoot, 2020-02-26:
Did anyone find a solution to this as it doesn't seem to be resolved in 8.1.0?
I only became aware of it as whilst I was away we had a power cut, when the power was restored all lamps connected to Tuya dimmers were on. PowerOnState is set to 0, so they should have been off when power was restored?

ascillato, 2020-02-26:
You need to see the Tasmota log. May be you have mqtt retained messages. For more help, please address this to the Tasmota support chat. Thanks