arendst/Tasmota #7415: LED Flickering H801

MarcelBTTF, 2020-01-02:
### PROBLEM DESCRIPTION
I have connected 5 LED strips (warm white) to a H801 controller and activated SetOption68 = 1 and set values for Color 23,61,71,69,46 and switch the individual channels ON/OFF, the LED strips start to flicker uncontrolled and differently. The same happens when changing the Color values. The flickering decreases after a short time, but is still noticeable.
If LedTable is set to 0, it is much better, but then the LEDs can only be dimmed in a small range, no gamma correction and the flickering is no longer present during dimming, but is still visible afterwards!
Command Sleep = 0 has no influence on the behaviour described above.
No flickering I have noticed when I set option15 = 1 and set the individual channels with the command PWM<x> 0..255, but with this variant I have no LED functionality anymore and cannot be the solution for the problem!

### REQUESTED INFORMATION
- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): H801
- [X] Tasmota binary firmware version number used: 8.1.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: FlashESP8266
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
18:05:00 SRC: WebConsole from 192.168.178.30
18:05:00 CMD: Gruppe 0, Index 1, Befehl "BACKLOG", Daten "Module"
18:05:00 SRC: Backlog
18:05:00 CMD: Gruppe 0, Index 1, Befehl "MODULE", Daten ""
18:05:00 MQT: stat/sonoff/RESULT = {"Module":{"20":"H801"}}:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```
18:01:14 SRC: WebConsole from 192.168.178.30
18:01:14 CMD: Gruppe 0, Index 1, Befehl "STATUS", Daten "0"
18:01:14 MQT: stat/sonoff/STATUS = {"Status":{"Module":20,"FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4",""],"Topic":"sonoff","ButtonTopic":"0","Power":31,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:01:14 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:02","StartupUTC":"2020-01-02T16:59:12","Sleep":50,"CfgHolder":4617,"BootCount":24,"SaveCount":329,"SaveAddress":"F9000"}}
18:01:14 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019.12.25 12:47:07","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"329/699"}}
18:01:14 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MROnline",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0002800B","2805C8000180060000005A64000000000000","00040008","00000000"]}}
18:01:14 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":567,"Free":436,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440EF","FlashMode":3,"Features":["00000407","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
18:01:14 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5458","IPAddress":"192.168.178.50","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"3C:71:BF:31:95:52","Webserver":2,"WifiConfig":4}}
18:01:14 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_319552","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
18:01:14 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 02 17:01:14 2020","Local":"Thu Jan 02 18:01:14 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:04"}}
18:01:14 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2020-01-02T18:01:14"}}
18:01:14 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2020-01-02T18:01:14","Uptime":"0T00:02:02","UptimeSec":122,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER1":"ON","Channel1":1,"POWER2":"ON","Channel2":1,"POWER3":"ON","Channel3":1,"POWER4":"ON","Channel4":1,"POWER5":"ON","Channel5":1,"Color":"3,3,3,3,3","Wifi":{"AP":1,"SSId":"MROnline","BSSId":"E8:DF:70:64:EF:E0","Channel":11,"RSSI":100,"Signal":-31,"LinkCount":1,"Downtime":"0T00:00:06"}}}
18:01:19 WIF: Prüfe Verbindung...
18:01:19 WIF: verbunden

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Set Color 23,61,71,69,46, switch the channels ON/OFF individually at random, always wait a moment. Set Color 13,26,3,23,3 and repeat the steps. 

### EXPECTED BEHAVIOUR
H801 Controller with SetOption68 = 1 and LedTable = 1.
No flickering when dimming or switching ON/OFF the LED channels.

### SCREENSHOTS
[image]

s-hadinger, 2020-01-02:
Thanks, I noticed subtle flickering too with Arilux LC01. I hadn't time to check with a scope what's happening yet.

stefanbode, 2020-01-07:
The PWM is controlled by the SDK. Unlikely that this causes the flicker. As far as I understand you see the flicker only during dimming up and down. In this case the tasmota way how to change the pwm could be an issue. There a recent changes to the code to make this much smoother with a 100Hz instead of 20Hz change process. Anyhow is this really a flickering or is the dumm down process not as smooth as you expect?

stefanbode, 2020-01-07:
Flickering can also be initiated by the power supply. If you can change this experimental this is also an idea to get more insight in the root cause

diebobo, 2020-01-12:
I have also encountered flickering with my RGB strips offcourse most noticble in the lower PWM range.. I attached my Logic Analyzer to a NodeMCU with nothing else attached so no influences. Running 8.1.0 and also 8.1.0.3 I with 5 PWM channels in RGBWW setting ( less channels makes no difference btw ) can clearly see that the PWM Pulse  lenght is at random moments off.. That means that a channel 99% of the time is for example 3us and then jumps on the next Pulse to 13us and then back to 3us.. There is no pattern to be found when its off.. The analyzer is set at 10 Mhz so there is nou rounding errors with that us .. Can attach screenshots if wanted or do additional tests.. 

First time experience with tasmota so no idea how it was in previous versions...

s-hadinger, 2020-01-12:
Thanks. Screenshots would be useful. Can you try with 1 PWM active and all off, then with 2 pwm active. I had a gut feeling it was more obvious at low values.

ascillato2, 2020-01-15:
@diebobo 

Hi, any news on this?

ascillato2, 2020-01-16:
@MarcelBTTF 

Any news on this? Have you tried latest Tasmota v8.1.0.4?

sfromis, 2020-01-17:
Testing with my H801, I saw marked slow on/off flickering (period like ½ sec) on 8.1, with a CW/WW strip (no RGB) on certain levels of moderate brightness, with color temperature slider in the middle. After upgrading to 8.1.0.4 I was not able to get this blinking again, even trying to do similar things.

However, I still see quite visible, and much faster, flickering when going near minimum brightness. This is not constant, but cycling between stable and flickering over a periods around a couple of seconds.

MarcelBTTF, 2020-01-18:
I could now record the behaviour of a friend with an oscilloscope. You can see the PWM signal from channel 4, if you switch channel 5 on, channel 4 starts to flicker, which is perceived as flickering of the LEDs. The behaviour can be reproduced with all channels. I will now test the new Tasmota v8.1.0.4 version to see if the flickering still occurs. 

[flicker.zip](https://github.com/arendst/Tasmota/files/4080360/flicker.zip)

s-hadinger, 2020-01-19:
Thanks, I will try to make a capture with a smaller time scale. There is definitely an issue in the pulse duration. I still don't know if it's the initial pulse that has a wrong duration or a second pulse that is randomly happening.

s-hadinger, 2020-02-01:
I think I found the cause of the flickering, but it will require changes in Arduino Core.

https://github.com/esp8266/Arduino/issues/7054

closed 2020-02-04 by arendst

kugelkopf123, 2020-05-22:
Just if someone search for it:
Because of the fact that blinking in powerup is a hardware problem, I have done some research on the logic processor. The built-in NXP 74HC245D has a pin for standby. With the attached circuit, the IC works only about 100ms after powerup. So the flash can be avoided.  OE is the Pin from the NXP.
[image]

nordeep, 2022-11-10:
For those who are here, like me. As you may see B channel(GPIO15) doesn't flickering on power up. Refer to [ESP8266 gpin table](https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/) - GPIO15 - pulled to GND.
As I guess NXP 74HC245D translate IN state to OUT as is on power up. R,G,W1,W2 - have HIGH or uncertain condition on boot. One of the way to fix it - add external PULL_DOWN resistor to the necessary GPIOs. Unfortunately that is extremely uncomfortable due small smd components. The bright side:
GPIO15 - B channel - already has internal PULL_DOWN
GPIO14 - W1 and GPIO4 - W2 - [located on opposite sides of the chip](https://tasmota.github.io/docs/Pinouts/).
I used 20kOm resistors. I'm using controller only as `cwww`.
[image]

nordeep, 2022-11-10:
> Looks like there are two revisions of the board, the image in the template for [H801](https://templates.blakadder.com/h801.html) has a different layout where you connect.

Yes. The board has index - H807WIFI-210819. The box itself has the same look. NXP 74HC245D on the over side of PCB.

PR #7649 (2020-02-04): Fix PWM flickering at low levels (#7415)
## Description:

Local patch for solving PWM flickering at low levels. See: https://github.com/esp8266/Arduino/issues/7054

Flickering is caused by phase changes in PWM pulses, draining current at different moments. Since most power supplies have limited capacitors, it creates visible changes in brigthness.

Note: there is still a slight flickering for 1 second after boot, probably due to high interrupt activity when connecting to wifi.

**Related issue (if applicable):** fixes #7415

## Checklist:
  - [x] The pull request is done against the latest dev [... cut]