arendst/Tasmota #7424: Tuya WiFi Blind Switch with RF remote control

mgoeller, 2020-01-03:
**Have you looked for this feature in other issues and in the docs?**  
Yes.

**Is your feature request related to a problem? Please describe.**  
I control my shutters from Elero with a plastic RF remote. I want to keep the possibility to control my shutters without Tasmota, so decided to buy the following product: [Tuya Smart Life WiFi Blind Curtain Switch with Remote](https://www.aliexpress.com/item/4000291503955.html?spm=a2g0s.9042311.0.0.71b74c4d0IONzh). This module is based on the Tuya TYWE2S, that I am able to flash with tasmota. There is a separate RF daughter board labeled CMT2219A connected to GPIO4/5 for communication with the ESP (?).

**Describe the solution you'd like**  
I would like to be able to continue using the included remote after flashing with Tasmota.

**Describe alternatives you've considered**  
I have considered using a Raspberry Pi and the official Elero Centero Transmitter Stick.

**Additional context**  
[image]

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-01-03:
If you could flash Tasmota on the ESP8266 chip, and you know the pins, you can set those gpios as serial port in Tasmota and by rules make Tasmota do the behaviour you want when receiving those RF codes.

Seems that there is nothing to add. Tasmota as it is should work. You will need to find the baudrate and serialconfig of the communication. If you need help on that you can ask in the Tasmota support chat.

Jason2866, 2020-01-03:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

mgoeller, 2020-01-03:
Thank you @ascillato . From the web interface I configured it as an "generic module" and set GPIO4 to Serial TX (148) and GPIO5 to Serial RX (149). How can I set the Baud Rate for the serial connection and how can I see the serial output in the web console?

@Jason2866 in which of the channels should I ask this question? I was asking pretty much the same question today in #general at 12:48pm but didn't get an answer.

Jason2866, 2020-01-03:
@mgoeller  12:48pm which timezone?  
Please read the documentation for the other questions you have. All explained and listed in
Section `Commands`

closed 2020-01-03 by ascillato2

mgoeller, 2020-01-04:
The support chat cannot help because the communication between the CMT2219A and the ESP is done via SPI and not serial. Can we reopen this feature request for supporting the CMT2219A?

freynder, 2020-08-20:
@mgoeller: are you sure the ESP communicates directly with the CMT2219A chip? The daughter board also contains an STM 8s003f3p6 MCU which may stand in between?

mgoeller, 2020-08-20:
As far as I remember, as soon as the remote was paired it kept on working even after flashing with Tasmota. So no need to do anything, the remote probably triggers the same GPIOs as the buttons do in the end (or directly the relays via the STM you mentioned).

jcisio, 2020-12-27:
Hi @mgoeller, could you tell me how did you manage to flash it? I have a similar one, maybe more recent, but there is nothing about the vcc/gnd/tx/rx pins.
[image]
[image]
[image]

pajakh, 2021-02-18:
@jcisio 
I have the same model, did you flash it? I cannot find GPIO 0 and VCC.

sergiojfa, 2022-02-04:
[image]
Good Morning.
Can anyone identify these components?

lalo-uy, 2022-02-04:
Q5 seems to read 3D -> MMBT44. NPN transistor
The other chips ar not readable.
May you to a more in focus pic, otr transcribe the markings?

sergiojfa, 2022-02-04:
Eduardo Spremolla thanks for replying...

I found this photo here.
I have one of these modules, but my module doesn't have markings on these two components U1 and U4.
If I had, I would look for it on the internet.

rojoricardo, 2022-02-24:
> Hi @mgoeller, could you tell me how did you manage to flash it? I have a similar one, maybe more recent, but there is nothing about the vcc/gnd/tx/rx pins. [image] [image] [image] If I guess correctly, the esp8266 is below the RF module. I don't need RF, so I could cut that part, but I don't know if it will damage the device. The reset button is on the vertical board in the first photo.

Hey, I have that same module, i had to transplant an ESP8266 to get tasmota in it, but i still have trouble to get communication between the RF module and Tasmota and Relays, have you had any progress with it?

ulugris, 2023-07-06:
I got the same device as @jcisio, @pajakh, @sergiojfa and @rojoricardo (MOES WRM-108-MS). I have managed to get it working with Tasmota together with the MOES RF remote by replacing the ESP WiFi module. It seems that the RF subsystem is independent of the ESP: each button on the RF remote activates a GPIO pin on the ESP board, which in turn controls the relays.

I simply removed the original WB2S module (blue board) with a hot air rework station, and replaced it by a TYWE2S, which I previously flashed with Tasmota using an FTDI.

After some fiddling with the multimeter, I discovered that the GPIO3 pin of the TYWE2S (RX) controls the LED on the RF daughter board, and GPIO13 is triggered by both the pushbutton on the vertical board and the "pause" button of the RF remote. Therefore, I used the following configuration:

```
Template {"NAME":"WRM-108-MS","GPIO":[0,0,0,544,32,33,0,0,225,34,224,0,0,0],"FLAG":0,"BASE":18}
```

Then, I issued the standard commands to enable and configure shutter mode:

```
SO80 1
Interlock 1,2
Interlock 1
ShutterRelay1 1
ShutterButton1 1 up 0
ShutterButton1 2 down 0
```

Finally, I added an extra rule I got [here](https://github.com/stefanbode/Sonoff-Tasmota/issues/182#issuecomment-528348209) for the "pause" button of the RF remote:

```
Rule1 ON Button3#State DO Shutterstop ENDON
Rule1 ON
```