arendst/Tasmota #7425: Shutter mode not observed using safe-circuit configuration

malbinola, 2020-01-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Up/Down arrow buttons in webpage and physical buttons connected to dedicated [pins](https://tasmota.github.io/docs/#/devices/Sonoff-Dual-R2?id=use-external-switches) seems that only activate relay1 and relay2 separately without shutter logic.

For example, if i press the Up button, relay1 is activated, opening time starts counting but at the end of the opening time a power1 OFF is not fired, leaving the relay exit powered. Same when clicking on the Down button, only the realy2 is activated (erroneously according to safe-circuit mode where both relays should be active) and then not powered off until a click on the Web ui button (in this case no close counting is done).

If i use the slider everything works correctly and shutter logic is succesfully applied: relay1 is powered on when sliding from closed to opened and then powered off at the end time; both relays are activated when dragging from opened to closed.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Dual R2
- [x] Tasmota binary firmware version number used: 8.1.0.2 Core/SDK Version 2_6_1/2.2.2-dev
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: PlatformIO
- [x] Flashing tools used: FW upgrade page
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

15:05:02 RSL: tasmota/stat/RESULT = {"NAME":"Sonoff Dual R2","GPIO":[17,255,0,255,0,22,18,0,21,56,0,0,0],"FLAG":0,"BASE":39}
15:05:02 RSL: tasmota/stat/RESULT = {"Module":{"0":"Sonoff Dual R2"}}
15:05:02 RSL: tasmota/stat/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"22":"Relay2"},"GPIO9":{"18":"Button2"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

15:06:14 RSL: tasmota/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:06:14 RSL: tasmota/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:43","StartupUTC":"2020-01-03T14:04:31","Sleep":50,"CfgHolder":4617,"BootCount":10,"SaveCount":74,"SaveAddress":"F5000"}}
15:06:14 RSL: tasmota/stat/STATUS2 = {"StatusFWR":{"Version":"8.1.0.2(tasmota)","BuildDateTime":"2020-01-02T14:42:23","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"341/699"}}
15:06:14 RSL: tasmota/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["razzarendena",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","40000000","00000000"]}}
15:06:14 RSL: tasmota/stat/STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":476,"Heap":28,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","87CAC287","040403A1","22B617CD","01001BC0","00007981"],"Drivers":"1,2,3,4,5,6,8,9,12,20,24,26,27,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
15:06:14 RSL: tasmota/stat/STATUS5 = {"StatusNET":{"Hostname":"tasmota-7252","IPAddress":"192.168.1.211","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"60:01:94:E5:5C:54","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:06:14 RSL: tasmota/stat/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E55C54","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
15:06:14 RSL: tasmota/stat/STATUS7 = {"StatusTIM":{"UTC":"Fri Jan 03 14:06:14 2020","Local":"Fri Jan 03 15:06:14 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:04","Sunset":"16:51"}}
15:06:14 RSL: tasmota/stat/STATUS10 = {"StatusSNS":{"Time":"2020-01-03T15:06:14","Shutter1":{"Position":0,"direction":0}}}
15:06:14 RSL: tasmota/stat/STATUS11 = {"StatusSTS":{"Time":"2020-01-03T15:06:14","Uptime":"0T00:01:43","UptimeSec":103,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"razzarendena","BSSId":"98:9B:CB:A4:CF:10","Channel":6,"RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

--> Up button clicked:

15:30:21 HTP: Main Menu
15:30:23 SRC: WebGui from 192.168.1.108
15:30:23 SHT: Switched relay: 1 by I
15:30:23 SHT: Wait for Motorstop..
15:30:23 SRC: Shutter
15:30:23 SHT: Switched relay: 0 by Shutter
15:30:23 RSL: tasmota/stat/RESULT = {"POWER1":"OFF"}
15:30:23 RSL: tasmota/stat/POWER1 = OFF
15:30:24 RSL: tasmota/stat/RESULT = {"POWER1":"OFF"}
15:30:24 RSL: tasmota/stat/POWER1 = OFF
15:30:24 SHT: Shutter 1: Real. 100, Start: 0, Stop: 10000, dir 1, motordelay 0, rtc: 0.05 [s], freq 0
15:30:24 RSL: tasmota/tele/RESULT = {"Shutter1":{"Position":1,"direction":1}}
[... 115 lines cut ...]
1. Configure device using template:
{"NAME":"Sonoff Dual R2","GPIO":[17,255,0,255,0,22,18,0,21,56,0,0,0,0],"FLAG":0,"BASE":39}
2. Activate shutter mode from console:
SetOption80 1
restart 1
ShutterRelay1 1
Backlog PulseTime1 0; PulseTime2 0
Interlock OFF
ShutterOpenDuration 5
ShutterCloseDuration 5
ShutterSetClose
3. Click on the Up button of the Web UI or push the hardware button 0

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

The relay1 is powered off automatically after shutter reaches its opening time.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2020-01-05:
There was already a change mode to the code that could be responsible for this bug. Have you checked you are on the latest release. The change is 3-4 days old

stefanbode, 2020-01-05:
Please use the current Develoment branch. I’m quite sure this already will fix your issue

malbinola, 2020-01-05:
> Have you checked you are on the latest release. The change is 3-4 days old

Hi @stefanbode, yes my project code is aligned to development branch; i've also done an update right before posting the issue. (for eg. i have [the brand new PR](https://github.com/arendst/Tasmota/pull/7403) for button controls in my fw)

Do you refer to the issue caused by the wrong test in a if sentence (missing  ==)?

However i can update the code to current situation and check again the behaviour.
Tell me if i can provide you more clues, i'm working on a test environment. :)

stefanbode, 2020-01-05:
Yes the missing == was the bug. The behavior you see only occur if the driver thinks that you use a stripper motor. In this Case the Position is defined  by the counter and not by time. I know what is wrong ( shutter mode is wrong) but I do not know why it changed. As far as I can see you all do right.

malbinola, 2020-01-05:
Ok done, i've updated my code to latest commits in development branch, compiled new .bin file and upgraded my Sonoff Dual R2. My tries:

**ShutterMode 1 (short-circuit safe)**
Unfortunately the issue is still present.
Up/Down buttons do not follow shutter logic, using slider everything is working fine.

**ShutterMode 0**
Everything is ok, Up/Down, hw buttons and slider are working correctly.

In order to do a clean start, i've reset the device (using `reset 1`) before changing the operative mode.

stefanbode, 2020-01-05:
#7364 shows similar bug.maybe not 100% fixed

stefanbode, 2020-01-05:
Ok agree that the issue is slightly different. The position is now recorded , but the relay is not changed to off as it should be. Anyhow the end position seems to be detected. Sorry, still in offline mode on holiday without the option to test intensively.

stefanbode, 2020-01-05:
Can you uncomment line 316 in the shutter_27drv.ino file and send the log again with log level 4. Thx

stefanbode, 2020-01-05:
Also please just for confirmation show the output of Shutterrelay1 1

malbinola, 2020-01-06:
> Sorry, still in offline mode on holiday without the option to test intensively.

Do not worry, it's normal in these days. :)

> Also please just for confirmation show the output of Shutterrelay1 1

`
17:49:29 CMD: shutterrelay1 1
17:49:29 SHT: Shutter 1 (Relay:1): Init. Pos: 10000 [100 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 10000, Opentime 5.0 [s], Closetime 5.0 [s], CoeffCalc: c0: 300, c1 500, c2: 700, c3: 900, c4: 1000, binmask 3, is inverted 0, shuttermode 1,motordelay 0
17:49:29 RSL: tasmota/stat/RESULT = {"ShutterRelay1":1}
`

Yes, you are right, stop instant is detected and the debug log printed.
The problem is the condition to skip the power off at line [352](https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_27_shutter.ino#L352):
I've added some other debug output just before testing that condition, these are the obtained values:
Settings.shutter_startrelay[i] --> 1
power --> 0
I hope it helps!

--> Shutter closed, Up button clicked:
`
17:51:09 SRC: WebConsole from 192.168.1.108
17:51:09 CMD: Group 0, Index 1, Command "D", Data ""
17:51:09 RSL: tasmota/stat/RESULT = {"Command":"Unknown"}
17:51:12 SRC: WebGui from 192.168.1.108
17:51:12 SHT: Switched relay: 1 by I
17:51:12 SHT: Wait for Motorstop..
17:51:12 SRC: Shutter
17:51:12 SHT: Switched relay: 0 by Shutter
17:51:12 RSL: tasmota/stat/RESULT = {"POWER1":"OFF"}
17:51:12 RSL: tasmota/stat/POWER1 = OFF
17:51:13 RSL: tasmota/stat/RESULT = {"POWER1":"OFF"}
17:51:13 RSL: tasmota/stat/POWER1 = OFF
17:51:13 SHT: Shutter1 Real 100, Start 0, Stop 10000, Dir 1, Delay 0, Rtc 0.05 [s], Freq 0
17:51:13 RSL: tasmota/tele/RESULT = {"Shutter1":{"Position":1,"Direction":1}}
17:51:13 SHT: Shutter1 Real 1000, Start 0, Stop 10000, Dir 1, Delay 0, Rtc 0.50 [s], Freq 0
17:51:13 RSL: tasmota/tele/RESULT = {"Shutter1":{"Position":10,"Direction":1}}
17:51:14 SHT: Shutter1 Real 3000, Start 0, Stop 10000, Dir 1, Delay 0, Rtc 1.50 [s], Freq 0
17:51:14 RSL: tasmota/tele/RESULT = {"Shutter1":{"Position":30,"Direction":1}}
17:51:15 SHT: Shutter1 Real 5000, Start 0, Stop 10000, Dir 1, Delay 0, Rtc 2.50 [s], Freq 0
17:51:15 RSL: tasmota/tele/RESULT = {"Shutter1":{"Position":50,"Direction":1}}
17:51:16 SHT: Shutter1 Real 7000, Start 0, Stop 10000, Dir 1, Delay 0, Rtc 3.50 [s], Freq 0
17:51:16 RSL: tasmota/tele/RESULT = {"Shutter1":{"Position":70,"Direction":1}}
17:51:17 SHT: Shutter1 Real 8900, Start 0, Stop 10000, Dir 1, Delay 0, Rtc 4.45 [s], Freq 0
17:51:17 RSL: tasmota/tele/RESULT = {"Shutter1":{"Position":89,"Direction":1}}
**17:51:18 SHT: Stop Condition detected: real: 10000, Target: 10000, direction: 1**
17:51:18 SHT: Shutter1 Real 10000, Start 0, Stop 10000, Dir 1, Delay 0, Rtc 5.00 [s], Freq 0
17:51:18 RSL: tasmota/stat/SHUTTER1 = 100
17:51:18 RSL: tasmota/tele/RESULT = {"Shutter1":{"Position":100,"Direction":0}}
17:51:18 CFG: Saved to flash at FA, Count 59, Bytes 4096
17:51:26 WIF: Checking connection...
17:51:26 WIF: Connected
`

--> Up button clicked again:
`
17:51:58 SRC: WebGui from 192.168.1.108
17:51:58 SHT: Switched relay: 1 by I
17:51:58 SHT: Wait for Motorstop..
17:51:58 SRC: Shutter
17:51:58 SHT: Switched relay: 0 by Shutter
17:51:58 RSL: tasmota/stat/RESULT = {"POWER1":"OFF"}
17:51:58 RSL: tasmota/stat/POWER1 = OFF
17:51:58 SHT: Skipping switch off relay 2
17:51:58 SRC: Shutter
17:51:58 SHT: Switched relay: 0 by Shutter
17:51:58 RSL: tasmota/stat/RESULT = {"POWER1":"OFF"}
17:51:58 RSL: tasmota/stat/POWER1 = OFF
17:51:58 RSL: tasmota/stat/RESULT = {"POWER1":"OFF"}
17:51:58 RSL: tasmota/stat/POWER1 = OFF
17:52:07 WIF: Checking connection...
17:52:07 WIF: Connected
`

stefanbode, 2020-01-07:
This is really strange. Power is a global tasmota variable that holds the binary bitmask of the relays switched on. In you case relay 1 is on and operation, therefore power must be 1 one not 0. Currently I have no idea why the global state is wrong. I need more thinking.

stefanbode, 2020-01-07:
Ok it look like there is a miss configuration with buttons. I see relay goes off if you switched the up button. This should not happen. For sure there is a bug that the shutter operates. Please take a look in the wiki how to define the buttons with follow and so on. Second question: is this a one time click button or a button that change position and stay there. The first was never implemented so far. This is more shuttermode2 with momentary switches that just send pulses.

stefanbode, 2020-01-07:
There is a new shuttercommand introduced the last days. I think this is the better option to control the shutter with buttons instead of directly addressing the relay. When home I will check what to change to avoid at least the hang. From the log I see both relay are off. Therefore the shutter should not move anyway. But in the log you can see that the shutter starts operating. This is definitely wrong.

malbinola, 2020-01-08:
> Please take a look in the wiki how to define the buttons with follow and so on.

Ok, i've configured buttons using `ShutterButton1 100 50 74 100 0 0 0 1 1 0 50 24 0 0 0 0 1 1` and now from a preliminary test, hardware buttons connected to GPIOs works observing ShutterMode1 logic. I hadn't guessed that PR completely change button logic.

Now the only aspect left a little bit strange is that Up/Down arrow button displayed in the Web home-page (just below shutter slider) they continue acting with the behaviour explained above; relays are powered individually. I believe that PR implementation needs to be completed, maybe @to-scho can tell us something.

> Second question: is this a one time click button or a button that change position and stay there.

Yes, it's a push-button that does not keep state, button return back to its initial position when releasing the finger.

stefanbode, 2020-01-09:
Ok in this case when it is a push-button, this is not supported in a shuttermode 1 configuration. The  buttons in the web interface are more or less wrong in some config because they operate always the relays. This can for sure be complicated. Anyhow I do not understand at the end you configuration. Is you only have pulse buttons and the second gives the direction and the first the power. How does this work. Pluse Buttons one for up and one for down are supported.

to-scho, 2020-01-09:
As Stefan said, web buttons are another story than GPIO buttons.

I submitted PR #7464 that make web buttons shutter control buttons instead of relais buttons. 

@Stefan, can you please review whether this might break other shutter modes?

malbinola, 2020-01-09:
> How does this work. Pluse Buttons one for up and one for down are supported.

Sorry @stefanbode, i try to explain it in a better way: the purpose is to have a push-button module (like [this](https://cdn.manomano.com/living-international-doppio-pulsante-interbloccato-l4037-T-3211308-10556407_1.jpg)) that have 2 buttons (one for Up and on for Down) mounted in the wall box near the shutter. This module will be connected to Sonoff Dual R2 exits for hardware buttons (a common gnd, button0 and button1, like described [here](https://tasmota.github.io/docs/#/devices/Sonoff-Dual-R2?id=use-external-switches)). Now after adding ShutterButton configuration works! :)

Then i would like to use ShutterMode 1 in order to avoid any short-circuit problems with shutter motors. Due to Sonoff R2 does not have clean exits, i will place an external relay (with COM NC NO) connected in a proper way where Sonoff relay1 will give the power and relay2 will control the external relay coil in order to change motor direction.

> The buttons in the web interface are more or less wrong in some config because they operate always the relays.

Do you mean now with these new PRs? Right? I'm quite sure to have tested both shuttermodes with success using your [previous fork](https://github.com/stefanbode/Sonoff-Tasmota).

Thanks @to-scho, i'll give your PR #7464 a try.

to-scho, 2020-01-09:
Config described by @malbinola is fine, I am using same setup beside Sonoff Basics instead of the Dual. 

Be aware of two pitfalls to get an external relay working:
a) Usually relays modules are 5V power supply ones. Even when the relay works at the beginning at 3.3V it will suddenly stop operation after a few weeks. This happened reproducibly here. You can get 5V for the relay module supply at one pin of the SOT89 drop down voltage converter at the sonoff PCB - AT YOUR OWN RISK.
b) You can get relays modules in an active high or active low version. Only the seldom active high ones will work when supplied by 5V since the GPIOs will only deliver 3.3V. 3.3V is fine for an active high one but with an active low version the relay will always stay active.

Another possibility is to find a module that has a 3.3V relay mounted (I haven't found ones for reasonable price). But there are separate 3.3V relais easily available to convert a 5V module.

malbinola, 2020-01-09:
@to-scho thank you for your right observations. I think that i will use a relay with 220V coil directly powered by Relay1/2 exits (also at 220V) of my Dual R2. I would like to keep Sonoff PCB modification at minimum so in case of failure i can easily replace it with a spare Dual R2.

Next, i've updated my code and after some preliminary test seems that ShutterMode 1 is fully working now; both hardware and web Up/Down buttons and relay activation logic are ok. Good work!

Then i switched back to ShutterMode 0 and problems begin. The opening seems to work well but, during close operation, after clicking the Down button relays are powered on alternately and suddenly everything stops. Another click to Down button shows same behaviour.

Am i missing something related to ShutterButton configuration?
This is the output for `ShutterButton1`:
 `RSL: tasmota/stat/RESULT = {"ShutterButton1":"1 100 50 74 100 - - - 1 1|2 0 50 24 0 - - - 1 1"}`

Here some debug log that may help.

From position 0 shutter is being opened, Up web button is clicked. No problems here.
```
21:29:28 SRC: WebConsole from 192.168.1.108
21:29:28 CMD: Group 0, Index 1, Command "A", Data ""
21:29:28 RSL: tasmota/stat/RESULT = {"Command":"Unknown"}
21:29:34 SRC: WebGui from 192.168.1.108
21:29:34 SHT: WebButton 1, ShutterPosition1 ToggleUp
21:29:34 SRC: WebGui from 192.168.1.108
21:29:34 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data "ToggleUp"
21:29:34 SHT: Pos. in: payload ToggleUp (8), payload -99, idx 1, src 7
21:29:34 SHT: Pos. in: payload ToggleUp (8), payload 100, idx 1, src 7
21:29:34 SHT: lastsource 7:, real 0, target 10000, payload 100
21:29:34 SHT: Wait for Motorstop..
21:29:34 SRC: Shutter
21:29:34 SHT: Switched relay: 0 by Shutter
21:29:34 RSL: tasmota/stat/RESULT = {"POWER1":"OFF"}
21:29:34 RSL: tasmota/stat/POWER1 = OFF
21:29:35 SRC: Shutter
21:29:35 SHT: Switched relay: 0 by Shutter
21:29:35 RSL: tasmota/stat/RESULT = {"POWER1":"OFF"}
21:29:35 RSL: tasmota/stat/POWER1 = OFF
21:29:35 SRC: Shutter
21:29:35 SHT: Switched relay: 0 by Shutter
21:29:35 RSL: tasmota/stat/RESULT = {"POWER2":"OFF"}
21:29:35 RSL: tasmota/stat/POWER2 = OFF
21:29:35 SRC: Shutter
21:29:35 SHT: Switched relay: 1 by Shutter
21:29:35 RSL: tasmota/stat/RESULT = {"POWER1":"ON"}
21:29:35 RSL: tasmota/stat/POWER1 = ON
21:29:35 RSL: tasmota/stat/RESULT = {"ShutterPosition1":100}
21:29:35 CFG: Saved to flash at F4, Count 171, Bytes 4096
21:29:35 SHT: Shutter1 Real 100, Start 0, Stop 10000, Dir 1, Delay 0, Rtc 0.05 [s], Freq 1000
21:29:35 RSL: tasmota/tele/RESULT = {"Shutter1":{"Position":1,"Direction":1}}
21:29:36 SHT: Shutter1 Real 1600, Start 0, Stop 10000, Dir 1, Delay 0, Rtc 0.80 [s], Freq 1000
21:29:36 RSL: tasmota/tele/RESULT = {"Shutter1":{"Position":16,"Direction":1}}
21:29:37 SHT: Shutter1 Real 3700, Start 0, Stop 10000, Dir 1, Delay 0, Rtc 1.85 [s], Freq 1000
21:29:37 RSL: tasmota/tele/RESULT = {"Shutter1":{"Position":37,"Direction":1}}
21:29:38 SHT: Shutter1 Real 5700, Start 0, Stop 10000, Dir 1, Delay 0, Rtc 2.85 [s], Freq 1000
21:29:38 RSL: tasmota/tele/RESULT = {"Shutter1":{"Position":57,"Direction":1}}
21:29:39 SHT: Shutter1 Real 7600, Start 0, Stop 10000, Dir 1, Delay 0, Rtc 3.80 [s], Freq 1000
21:29:39 RSL: tasmota/tele/RESULT = {"Shutter1":{"Position":76,"Direction":1}}
21:29:40 SHT: Shutter1 Real 9700, Start 0, Stop 10000, Dir 1, Delay 0, Rtc 4.85 [s], Freq 1000
21:29:40 RSL: tasmota/tele/RESULT = {"Shutter1":{"Position":97,"Direction":1}}
21:29:40 SRC: Shutter
21:29:40 SHT: Switched relay: 1 by Shutter
21:29:40 RSL: tasmota/stat/RESULT = {"POWER1":"OFF"}
21:29:40 RSL: tasmota/stat/POWER1 = OFF
21:29:40 SRC: Shutter
21:29:40 SHT: Switched relay: 0 by Shutter
21:29:40 RSL: tasmota/stat/RESULT = {"POWER2":"OFF"}
21:29:40 RSL: tasmota/stat/POWER2 = OFF
21:29:40 SHT: Shutter1 Real 10000, Start 0, Stop 10000, Dir 1, Delay 0, Rtc 5.00 [s], Freq 1000
21:29:40 RSL: tasmota/stat/SHUTTER1 = 100
21:29:40 RSL: tasmota/tele/RESULT = {"Shutter1":{"Position":100,"Direction":0}}
21:29:40 CFG: Saved to flash at FB, Count 172, Bytes 4096
21:29:41 WIF: Checking connection...
21:29:41 WIF: Connected
21:29:44 APP: Serial logging disabled
[... 32 lines cut ...]
21:31:25 SHT: Wait for Motorstop..
21:31:25 SRC: Shutter
21:31:25 SHT: Switched relay: 2 by Shutter
21:31:25 RSL: tasmota/stat/RESULT = {"POWER1":"OFF"}
21:31:25 RSL: tasmota/stat/POWER1 = OFF
21:31:25 RSL: tasmota/stat/RESULT = {"POWER2":"OFF"}
21:31:25 RSL: tasmota/stat/POWER2 = OFF
21:31:25 SRC: Shutter
21:31:25 SHT: Switched relay: 1 by Shutter
21:31:25 RSL: tasmota/stat/RESULT = {"POWER1":"ON"}
21:31:25 RSL: tasmota/stat/POWER1 = ON
21:31:25 RSL: tasmota/stat/RESULT = {"ShutterPosition1":0}
21:31:25 SRC: Shutter
21:31:25 SHT: Switched relay: 1 by Shutter
21:31:25 RSL: tasmota/stat/RESULT = {"POWER1":"OFF"}
21:31:25 RSL: tasmota/stat/POWER1 = OFF
21:31:25 SRC: Shutter
21:31:25 SHT: Switched relay: 0 by Shutter
21:31:25 RSL: tasmota/stat/RESULT = {"POWER2":"OFF"}
21:31:25 RSL: tasmota/stat/POWER2 = OFF
21:31:25 SHT: Shutter1 Real 8700, Start 10000, Stop 10000, Dir -1, Delay 0, Rtc 0.65 [s], Freq 1000
21:31:25 RSL: tasmota/stat/SHUTTER1 = 87
21:31:25 RSL: tasmota/tele/RESULT = {"Shutter1":{"Position":87,"Direction":0}}
21:31:26 CFG: Saved to flash at FA, Count 173, Bytes 4096
```

to-scho, 2020-01-10:
I was able to reproduce this behavior. It happens because the shutter is not aware about the interlock change when asked to do a movement afterwards. For open direction it first looks fine but for close direction you got this relais ping-pong. The shutter still thinks to operate in ShutterMode 1 and sets both relais, the interlock reset one, the shutter reacts and stops operation. If you restart the device after interlock change or if you call "shutterrelay1 1" before movement everything is fine.

@stefanbode, is this something we should fix? Here is an easy way to reproduce it

Fail:
backlog interlock off; shutterrelay1 1; shuttersetclose1; interlock 1,2; interlock on; shutteropen1
...wait to open...
shutterclose1

Pass:
backlog interlock off; shutterrelay1 1; shuttersetclose1; interlock 1,2; interlock on; shutterrelay1 1; shutteropen1
...wait to open...
shutterclose1

PR #7494 (2020-01-12): initialize shutters when interlock is changed
## Description:

**Related issue (if applicable):** fixes #7425

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [ ] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).
closed 2020-01-12 by arendst