arendst/Tasmota #7427: send RAW IR messages doesn't work in standard tasmota.bin

mani9876, 2020-01-03:
Hello,

I am using a NodeMCU, with the Tasmota Firmware v8.1.0.
I am using it as an IR Transmitter and Receiver, as the Protocol of some of my Remote Control Keys are "UNKNOWN" I need to send this by raw messages.

If I try to send e.g. `irsend 0,932,838,932,856,1772,892,908,842,906,864,906,846,904,922,910,840,908,900,902,850,910,1774,928,842,812`

with the "tasmota-ir.bin" firmware, everything works fine. If I use the tasmota.bin firmware, I get `MQT: stat/tasmota/RESULT = {"IRSend":"Invalid JSON"}.`

Is the Raw IR Send not implemented in the standard tasmota.bin, or is this a bug?

Also, is there a possibilty to speed up the receiving? 
If I press more than 1 key per second on the remote control, not every key get's recogniced by the ESP8266.

Thanks!

closed 2020-01-03 by close-issue-app[bot]

ascillato, 2020-01-03:
Please see release notes. There is an specific version for IR.

ascillato, 2020-01-03:
DUPLICATED ISSUE

mani9876, 2020-01-03:
Hello ascillato, I know that there is a specific version for IR, but as I understand, this only means, that there are many more protocols integrated. 
If I use the standard firmware, I can activate the RAW IR receive mode, so I thought it should also be capable of use the RAW IR send. But I can't find any information, if this is a bug, or it's normal.