arendst/Tasmota #7428: Domoticz doesn't correctly set color temp of CW/WW bulb

grahamf72, 2020-01-03:
### PROBLEM DESCRIPTION
When attempting to set the color temperature of a CW/WW bulb within domoticz, the bulb turns off but color temp isn't changed. If the bulb is set to RGBWW in domoticz, changing the RGB values  with the color wheel change the color temp, with the R value setting the cold, and G value setting the warm.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Laserco CSH-B22WW10W (Tuya bulb)
- [x ] Tasmota binary firmware version number used: 8.1.0_____
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: OTA_____
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
22:10:09 MQT: stat/tasmota/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
22:10:09 MQT: stat/tasmota/RESULT = {"Module":{"0":"Generic"}}
22:10:09 MQT: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"37":"PWM1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"38":"PWM2"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```
22:11:04 MQT: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:11:04 MQT: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-sensors.bin","RestartReason":"Power on","Uptime":"0T00:13:42","StartupUTC":"2020-01-03T20:57:22","Sleep":50,"CfgHolder":4617,"BootCount":24,"SaveCount":412,"SaveAddress":"F6000"}}
22:11:04 MQT: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0(sensors)","BuildDateTime":"2019-12-25T12:38:43","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"363/699"}}
22:11:04 MQT: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BAD_WOLF",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
22:11:04 MQT: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":22,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":3,"Features":["00000809","8FDAE387","0437A3A4","B7FFBFCD","01DA9BC0","0016FCC7"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,18,19,20,21,22,24,26,29,30","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,34,37,39,40,42,43,51,52,55,56,58"}}
22:11:04 MQT: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-5888","IPAddress":"192.168.1.36","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"BC:DD:C2:B0:37:00","Webserver":2,"WifiConfig":5}}
22:11:04 MQT: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B03700","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:11:04 MQT: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Fri Jan 03 21:11:04 2020","Local":"Fri Jan 03 22:11:04 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:05"}}
22:11:04 MQT: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-01-03T22:11:04"}}
22:11:04 MQT: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-01-03T22:11:04","Uptime":"0T00:13:42","UptimeSec":822,"Heap":22,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":102,"MqttCount":1,"POWER":"ON","Dimmer":16,"Color":"2A00","HSBColor":"0,0,0","Channel":[16,0],"CT":153,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"BAD_WOLF","BSSId":"A0:04:60:1F:29:95","Channel":6,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:12"}}}
```
- [x ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
22:13:19 MQT: Received Topic "domoticz/out", Data Size 409, Data "{
   "Battery" : 255,
   "Color" : {
      "b" : 0,
      "cw" : 0,
      "g" : 0,
      "m" : 2,
      "r" : 0,
      "t" : 255,
      "ww" : 255
   },
   "Level" : 13,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Color Switch",
   "id" : "00082023",
   "idx" : 23,
   "name" : "Living Room 1",
   "nvalue" : 10,
   "stype" : "RGBWW",
   "svalue1" : "13",
   "switchType" : "Dimmer",
   "unit" : 1
}
"
22:13:19 DOM: idx 23, nvalue 10
22:13:19 DOM: Received Topic /Color, Data 0000000021
22:13:19 SRC: MQTT
22:13:19 CMD: Group 0, Index 1, Command "COLOR", Data "0000000021"
22:13:19 SRC: Light
22:13:19 MQT: domoticz/in = {"idx":23,"nvalue":0,"svalue":"0","Battery":100,"RSSI":9}
22:13:19 MQT: stat/tasmota/RESULT = {"POWER":"OFF","Dimmer":0,"Color":"0000","HSBColor":"0,0,0","Channel":[0,0],"CT":153}
22:13:19 CFG: Saved to flash at FA, Count 416, Bytes 4096
22:13:20 MQT: Received Topic "domoticz/out", Data Size 406, Data "{
   "Battery" : 100,
   "Color" : {
      "b" : 0,
      "cw" : 0,
      "g" : 0,
      "m" : 2,
      "r" : 0,
      "t" : 255,
      "ww" : 255
   },
   "Level" : 13,
   "RSSI" : 9,
   "description" : "",
   "dtype" : "Color Switch",
   "id" : "00082023",
   "idx" : 23,
   "name" : "Living Room 1",
   "nvalue" : 0,
   "stype" : "RGBWW",
   "svalue1" : "0",
   "switchType" : "Dimmer",
   "unit" : 1
}
"
22:13:20 DOM: idx 23, nvalue 0
22:13:21 MQT: Received Topic "domoticz/out", Data Size 228, Data "{
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Temp",
   "id" : "82004",
   "idx" : 4,
   "name" : "Inside Temp",
   "nvalue" : 0,
   "stype" : "LaCrosse TX3",
   "svalue1" : "23.81",
   "unit" : 1
}
"
22:13:21 DOM: idx 4, nvalue 0

```

### TO REPRODUCE
Within Domoticz, set a color temperature using the color temperature wheel.

### EXPECTED BEHAVIOUR
The color temperature should change according to the color temp selected within Domoticz.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

I have had a look at the code, and the Domoticz MQTT messages, and have tracked the problem down to the handling of the Domoticz color command in lines 299-308 of xdrv_07_domoticz.ino.  

When Domoticz sends a color change command, the value of "m" within the Color section of the message dictates what information it contains.
From the Domoticz documentation, at https://www.domoticz.com/wiki/Domoticz_API/JSON_URL%27s#Set_an_RGB_CW_WW_or_CW_WW_light_to_a_certain_color_temperature
If "m" is 2, only the color temperature data is provided, and RGB values are 0.
If "m" is 3, only the RGB color data is provided, and cw/ww values are 0.
If "m" is 4, then all RGB and cw/ww values are sent.

The tasmota code ignores the value of "m", and assumes that all RGB / CW / WW values are valid.  It should test the value of m - if it is 2, only adjust the color temp, if 3 only adjust the RGB values, and only adjust all values if m is 4.

The behaviour seems to be then that because tasmota only has 2 PWM pins, it uses the first two color values (R & G) when setting the PWM. This would explain why using the RGB color wheel changes the color temperature based on the R & G values, but using the color-temp wheel it turns the light off, because R&G are 0.

Note - the above logs show that I currently have Domoticz set to an RGBWW light, even though my light is not RGB. I have tested domoticz with the light set as WW, and the behaviour is the same, except without the RGB color wheel I have no way of controlling the color temp from Domoticz.  

From looking at the code, it would appear that the solution would be to test the value of "m" and send the command D_CMND_COLORTEMPERATURE if m is 2.  Unfortunately I haven't been able to work out the paramaters for that command.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-01-04:
I haven't used domoticz for rgb control but could it be your issue is caused by a change in domoticz where a new code is used to distinguish between different color control.

If so the change is dependent on domoticz version and previous versions worked fine.

In any way a solution would need to be found by indeed handling this new m field.

grahamf72, 2020-01-04:
Hi Theo, thanks for your efforts with Tasmota, I only just discovered it, and so far quite impressed.

I’m able to make the needed code change, I’m just not sure about the format and which values to use for the parameters to be sent with the D_CMND_COLORTEMPERATURE command.

arendst, 2020-01-04:
Pls give the latest commit a try and report back.

grahamf72, 2020-01-05:
Wow, didn't expect a fix that quick. Thanks very much.
Tested it, and it works exactly as expected.  I notice you still don't do anything different for the situation where m is 3 (rgb valid, color temp invalid), but not sure if that is a problem.  If I ever get an RGB bulb, and run into dramas I'll log another job.
The only other thing I have noticed is that if color temp is set in the tasmota web interface, it doesn't update in Domoticz, but I don't need that functionality.
Thanks again for the update.

closed 2020-01-05 by grahamf72