arendst/Tasmota #7435: hardware update BlitzWolf BW-SHP-6

sander45, 2020-01-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
The hardware layout of the Blitzwolf- SHP6 energy monitoring socket has been changed. The good news is it can take now 15 Amps in state of 10 Amps. The down side is that the pin layout has been changed too and the gnd pin is not marked. It has a esp8285 chip.
I assume it is the upper left of the 6 pins
[image]
. Did anyone already flashed the updated version? And/or suggestion for the pin layout?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Blitzwolf SHP6
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-01-04:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-01-05 by ascillato2

MaDill, 2020-01-06:
I had the same problem but found a picture via this website: https://www.bastelbunker.de/gosund-sp111-mit-tasmota/

Picture: https://ibb.co/CMgh6KT

sander45, 2020-01-07:
Thanks Madill. That is exact the tip I needed. Unfortunately it is still not working, maybe I blew up a chip while fussing with the wires. Will try it again this weekend.

PrzemoC-PL, 2020-01-10:
I was already flashed this module. Tested with tasmota 6.6 and 8.1 and shows no energy consumption. (counters show "0" and change only "today" and "sum" are updated.)

Do you have this problem Have you calibrated tasmota by command line ?

majherek, 2020-01-10:
Use Gosund SP111 V2 template. 
https://templates.blakadder.com/gosund_SP111_v2.html

sander45, 2020-01-10:
thanks majherek, that was the next tip I needed. It now works perfect.

tdamsma, 2020-01-24:
> Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

I understand this issue was a bit misplaced here, but because it was here and not in chat it helped me too!

marcelloz1966, 2020-02-29:
Hi to all...
This is a noob question, i hope you can help me.
I have a Blitzwolf shp6 with pcb very similar to Gosund SP111 V2 .
How i can enter in flash mode? 
If I don't get it wrong,  I have to connect gnd to gpio0 at boot.
I have do that, according to this photo https://ibb.co/CMgh6KT posted here by MaDill 
But now my unit seems bricked..
Who can help me?
Thank you :)

Jason2866, 2020-02-29:
@marcelloz1966 
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

marcelloz1966, 2020-02-29:
Thank you Jason....:)
Inviato da Yahoo Mail su Android

szaman-89, 2020-03-08:
> I was already flashed this module. Tested with tasmota 6.6 and 8.1 and shows no energy consumption. (counters show "0" and change only "today" and "sum" are updated.)
> 
> Do you have this problem Have you calibrated tasmota by command line ?

Did you solved your problem? I have same situation and each tamplate won't work.

marcelloz1966, 2020-03-08:
Hi! Yes...My blitzwolf pcb is equal to the one that yiu can see in this page:https://templates.blakadder.com/gosund_SP111_v2.html

 Use the template you found inside...or copy this one{"NAME":"SP111 v1.1","GPIO":[56,0,158,0,132,134,0,0,131,17,0,21,0],"FLAG":0,"BASE":45}
Remember to flag the square box  to "Activate" the template and reboot.Try! ;)

Inviato da Yahoo Mail su Android

szaman-89, 2020-03-08:
@marcelloz1966 It worked. Ticked "Activate" checkbox did the trick. Thank you.

marcelloz1966, 2020-03-08:
Your welcome :)Bye

Inviato da Yahoo Mail su Android

crolaser, 2021-02-25:
i flashed and work well but the Voltage meter showing 310 v instead 240 v .any help ??