arendst/Tasmota #7438: Arilux LCO1 - RGB with KNX

goofy79, 2020-01-04:
### PROBLEM DESCRIPTION
I think the KNX Settings for the RGB Module doesn't work ?
I have **Arilux LC01** Modul with the Tasmota Version 8.1.0.
In the KNX Settings, i can choose KNX RX 1 - KNX RX 5
But it doesn't matter what data i send to the Inputs - the Color and the brightness
won't change.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Arilux LC01
- [x] Tasmota binary firmware version number used: 8.0.1
  - [x] Pre-compiled
- [x] Flashing tools used: nodemcu pyflasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

16:35:12 CMD: Backlog Template; Module; GPIO 255
16:35:12 RSL: RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
16:35:12 RSL: RESULT = {"Module":{"37":"Arilux LC01"}}
16:35:12 RSL: RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"159":"ALux IrSel"},"GPIO3":{"0":"None"},"GPIO4":{"147":"ALux IrRcv"},"GPIO5":{"37":"PWM1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"38":"PWM2"},"GPIO13":{"39":"PWM3"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```

- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

16:37:37 CMD: Status 0
16:37:37 RSL: STATUS = {"Status":{"Module":37,"FriendlyName":["Tasmota"],"Topic":"ARILUX_LC01","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:37:37 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"1T02:26:29","StartupUTC":"2020-01-03T13:11:08","Sleep":50,"CfgHolder":4617,"BootCount":6,"SaveCount":171,"SaveAddress":"F9000"}}
16:37:37 RSL: STATUS2 = {"StatusFWR":{"Version":"8.1.0.1(tasmota)","BuildDateTime":"2019.12.31 14:39:21","Boot":7,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"366/699"}}
16:37:37 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!Box Fon WLAN 7170",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0A008001","2805C8000100060000005A00000000000000","00000000","00000000"]}}
16:37:37 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":595,"Free":408,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000407","9FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,11,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
16:37:37 RSL: STATUS5 = {"StatusNET":{"Hostname":"ARILUX_LC01-6912","IPAddress":"172.16.0.32","Gateway":"172.16.0.1","Subnetmask":"255.255.255.0","DNSServer":"172.16.0.1","Mac":"D8:F1:5B:86:FB:00","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:37:37 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sat Jan 04 15:37:37 2020","Local":"Sat Jan 04 16:37:37 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:06"}}
16:37:37 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-01-04T16:37:37"}}
16:37:37 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-01-04T16:37:37","Uptime":"1T02:26:29","UptimeSec":95189,"Heap":22,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"AN","Dimmer":89,"Color":"00E38F","HSBColor":"158,100,89","Channel":[0,89,56],"Scheme":0,"Fade":"AUS","Speed":1,"LedTable":"AN","Wifi":{"AP":1,"SSId":"FRITZ!Box Fon WLAN 7170","BSSId":"08:96:D7:57:F7:08","Channel":1,"RSSI":30,"Signal":-85,"LinkCount":3,"Downtime":"0T00:00:10"}}}
```

### TO REPRODUCE
Set KNX Group Addresses with ETS - than set the Groupaddresses the receive Date an send any data to this Groupaddresses. 
My Groupaddresses are 
3/3/70 -> Output 1 On/Off **(work)**
3/3/71 -> KNX RX 1 _(won't work with any data)_
3/3/72 -> KNX RX 2 _(won't work with any data)_
3/3/73 -> KNX RX 3 _(won't work with any data)_
3/3/74 -> KNX RX 4 _(won't work with any data)_
3/3/75 -> KNX RX 5 _(won't work with any data)_

### EXPECTED BEHAVIOUR
I expected that it i send a value between 0-255 the the KNX RX 1-3 the Color of the RGB Stripe will change. So KNX RX 1 for Red - KNX RX 2 for Green - KNX RX 3 for Blue - KNX RX 4 maybe for brightness and KNX RX 5 for the last control.
So i can change the color with a string linke 
0/0/255 for Blue
255/0/255 for Pink
and so on ...

I think the Values are there
`16:43:56 RSL: RESULT = {"POWER":"AN","Dimmer":89,"Color":"00E38F","HSBColor":"`**158,100,89**`","Channel":[0,89,56]}`
the only think to do is, to set the received values to the right variables, or ?

### SCREENSHOTS
[image]

[image]

[image]

[image]

### ADDITIONAL CONTEXT
nothing at the moment

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-01-04:
KnxRx slots are meant to be used with rules. See wiki and docs at KNX chapter on how to use knx rules triggers. If you send the color separated in channels, you should be able to set rules to use them and set the colors in Tasmota. 

If the command you are sending is the DPT for color, it is not supported by now.

goofy79, 2020-01-05:
Ok - is this planed ? Can you give me a link
to the KNX chapter? Because i don‘t find anything if i use the search function ?

Thank you so mutch.

ascillato2, 2020-01-05:
Hi, yes, it is planned to be added. I have to find some time to do it.

About your rule. Your rule is ok. The issue is the DPT your KNX system is using to send that value that is not what Tasmota expects. Try to change the DPT. If you can't you will have to wait until Color DPT support is added to the Tasmota KNX driver.

goofy79, 2020-01-05:
can you tell me what kind of DPT the Tasmota driver expected ?
Actual i send 5.005 Decimal (0-255)

ascillato2, 2020-01-05:
Yes, sorry, Tasmota expects DPT 9 (2 bytes float) for recepted values

goofy79, 2020-01-05:
puh, that's heavy - to convert a 0-255 RGB value in KNX to a 2 Byte Float.
my Visu send, for example, a 1-Byte value per Color

[image]

JohannesRegner, 2020-01-08:
Hi, i have the same problem! I'm using an Gira X1 KNX Visua Server, and tryed to change the color via GUI, but i can only switch on/off.
I think, the problem is the same as @goofy79  described ...

 if you need any support/testing, I can also help!

githubtzzz, 2020-03-11:
Hi, any updates here?

ascillato, 2020-03-11:
Hi, yes, it is planned to be added. I have to find some time to do it.

githubtzzz, 2020-03-26:
Hi @ascillato, do you think you can find time for this?
Do you need any support?

ascillato2, 2020-04-02:
@githubtzzz Please, give me some days. Thanks.

ascillato2, 2020-04-06:
From #8084, also add Dimming control.

fluppie, 2020-09-01:
+1 I was also looking for this func. I can switch the LED strip on and off, but also want to regulate the brightness with the KNX switches.

JackyKNX, 2020-10-22:
+1 more ;) Is there any chance to get it soon ?

jpirnay, 2020-10-26:
+1, I do have some tasmota driven LED strips that would definitely benefit from this!

edvandreas, 2020-11-05:
+1, I would also like to get this implemented

Jaegerstefan, 2021-01-07:
+1 It would be really helpful to get this working.

eldi0s, 2021-03-08:
Definitely +1

I wonder how DPT9 could make it while DPT5 did not ;-)

closed 2021-04-14 by ascillato2