arendst/Tasmota #7441: Feit Electric OM100/RGBW/CA/AG fcc ID:syw-a21rgbwagt2

meberlein, 2020-01-04:
### PROBLEM DESCRIPTION
Hi All,

I purchased a new Feit Electric bulb from Menards an unfortunately after sacrificing one it appears
that it is bit different from the other Feit bulbs that are so far supported. Through trying different 
templates I can often get parts to work but not whole device. 

### REQUESTED INFORMATION
Any help in figuring out a working template would be greatly appreciated.

### EXPECTED BEHAVIOUR
	- RGB
	- Cool white/warm white selectable
	- multiple patterns

### ADDITIONAL CONTEXT
Interior pictures:
	[All interior pictures](https://photos.app.goo.gl/Syn5m33ccNbK6nqx7)
	[ICs Labeled](https://photos.app.goo.gl/hBLNjp91Cg2JkWEm8)

Status:
	- Flashed fine
		- Did brick two post flash (presumably by setting all channels to 100% during my testing)

Flashed version:
	8.1.0

Templates attempted:
	Many combinations of templates from the following issues:
		https://github.com/arendst/Tasmota/issues/6534

	Currently:
		{"NAME":"OM60/RGBW","GPIO":[0,0,0,0,140,37,0,0,38,142,141,0,0],"FLAG":0,"BASE":18}
		- Behavior with above template (Admin page showing 5 gray/white sliders, 5 toggle buttons):
			Toggle 3 controls cool white(1)/warm white(0)
			Slider 4 controls brightness of white
			Toggle 5 controls green on off
			Slider 6 controls brightness of green

Product page:
[Product Purchase Link](	https://www.menards.com/main/electrical/light-bulbs/led-light-bulbs/feit-electric-100w-equivalent-a19-color-selectable-led-light-bulb/pom100-rgbw-ca-ag/p-1550302361952-c-7482.htm)
Main Controller Board:
	P/N: TYWE3L

LED Controls:
	1x BP2306 http://bpsemi.com.cn/uploads/file/20190723152414_567.pdf
	2x BP8516 Hamamatsu (Can't find data sheet)
	1x BP 5926 https://www.dianyuan.com/upload/community/2019/04/25/1556153072-61489.pdf (Some data in here but unfortunately in Chinese)

Pins out to LED board:
	W+/W-
	C+/C-
	R
	G
	B
	RGB

Rule output:
19:41:13 CMD: Backlog Rule1; Rule2; Rule3
19:41:13 MQT: stat/GarageEast/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
19:41:13 MQT: stat/GarageEast/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
19:41:13 MQT: stat/GarageEast/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

Status0 output:
19:41:35 MQT: stat/GarageEast/STATUS = {"Status":{"Module":0,"FriendlyName":["GarageEast","Tasmota2","Tasmota3","Tasmota4",""],"Topic":"GarageEast","ButtonTopic":"0","Power":19,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

closed 2020-01-04 by close-issue-app[bot]

meberlein, 2020-01-04:
Only supports green toggle on off (no control):
https://templates.blakadder.com/feit_electric-OM60RGBWCAAG.html

meberlein, 2020-01-04:
Green/Blue/Warm White supported:
{"NAME":"AoycocrBR30","GPIO":[0,0,0,0,0,39,0,0,38,0,37,40,0],"FLAG":0,"BASE":18}

meberlein, 2020-01-04:
All colors work, white is dim and ww/cw reversed on slider
{"NAME":"Arlec RGBWW","GPIO":[0,0,0,0,37,40,0,0,38,41,39,0,0],"FLAG":0,"BASE":18}

Manual color:
Color FF00000000 (red)
Color 00FF000000 (green)
Color 0000FF0000 (blue)
Color 000000FF00 (warm white)
Color 000000FFDD Cool White

ascillato, 2020-01-05:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

pk-666, 2020-03-27:
I purchased one of these.
Instead of the stock 5 channel PWM R,G,B,CW,WW the controls for this bulb are R,G,B,Brightness,PWM-CT(0 for WW and 255 for CW)
I tried Module 48, but it only seems to work with 2 channel bulbs.

s-hadinger, 2020-03-27:
Module 48 is supposed to work with 5 channels bulbs too. But still it's very uncommon, there's maybe a bug that nobody noticed up to now

s-hadinger, 2020-03-27:
I checked code and I don't see anything wrong:

```

  if ((LST_COLDWARM == Light.subtype) || (LST_RGBCW == Light.subtype)) {
    // channels for white are always the last two channels
    uint32_t cw1 = Light.subtype - 1;       // address for the ColorTone PWM
    uint32_t cw0 = Light.subtype - 2;       // address for the White Brightness PWM
    uint16_t white_bri10 = cur_col_10[cw0] + cur_col_10[cw1];   // cumulated brightness
    uint16_t white_bri10_1023 = (white_bri10 > 1023) ? 1023 : white_bri10;    // max 1023

    if (PHILIPS == my_module_type) {   // channel 1 is the color tone, mapped to cold channel (0..255)
      // Xiaomi Philips bulbs follow a different scheme:
      cur_col_10[cw1] = light_state.getCT10bits();
      // channel 0=intensity, channel1=temperature
      if (Settings.light_correction) { // gamma correction
        cur_col_10[cw0] = ledGamma10_10(white_bri10_1023);    // 10 bits gamma correction
      } else {
        cur_col_10[cw0] = white_bri10_1023;  // no gamma, extend to 10 bits
      }
```

pk-666, 2020-03-28:
I was trying to configure the blakadder template graphically and it didn't work. (probably some newbie's mistake) 
The template is: 
{"NAME":"OM100/RGBWW","GPIO":[0,0,0,0,37,40,0,0,38,41,39,0,0],"FLAG":0,"BASE":18}
Then I changed the template to:
{"NAME":"OM100/RGBWW","GPIO":[0,0,0,0,37,40,0,0,38,50,39,0,0],"FLAG":0,"BASE":48}
Now all light commands and sliders are working properly!!!

jangeador, 2020-06-30:
@pk-666 I tried the configuration that you posted and it does work, but the warm white and cold white sliders are inverted. I changed the pin from `PWM5i (50)` to `PWM5 (41)` and that fixed the slider. Thanks for your contribution. Here is the one I used:

```
{"NAME":"OM100/RGBWW","GPIO":[0,0,0,0,37,40,0,0,38,41,39,0,0],"FLAG":0,"BASE":48}
```

dylandn, 2021-01-13:
Hello, 
I'm curious how you managed to flash these bulbs, with tuya-convert or serial?
I've just spent a couple hours trying both, no dice.

jangeador, 2021-01-13:
I used tuya-convert. Flashing them was not a problem at all, and I flashed several of them. If you are having problems you might have a different hardware or they might have changed it.