arendst/Tasmota #7443: Resolving wrong IP Address

ObiToo55, 2020-01-05:
Information button on Main Menu
AP1 SSId (RSSI)	Aruba (52%, -74 dBm)
Hostname	Leeslampje
IP Address	192.168.178.69
Gateway	192.168.178.1
Subnet Mask	255.255.255.0
DNS Server	192.168.178.1			<- should be 192.168.178.**5**
MAC Address	60:01:94:CB:FD:6E

So I entered the Console:
[image]

After hitting Enter the result will be:
[image]

And the Information button on Main Menu again:
[image]

I am using Program Version | 8.1.0(tasmota)

Is this a bug, or am I doing something wrong ?
BTW the same will happen trying to enter IPAddress2 (Gateway)

closed 2020-01-05 by close-issue-app[bot]

Jason2866, 2020-01-05:
Your network setup is somewhere wrong. NOT a error in Tasmota.
Tasmota uses what has been told via DHCP

5chufti, 2020-09-03:
that was not the mistake, just omit the "=" when entering an ip
```
08:54:33 CMD: ipaddress4 = 10.254.11.253
08:54:33 RSL: RESULT = {"IPAddress4":"0.254.11.253"}
08:55:28 CMD: ipaddress4 10.254.11.253
08:55:28 RSL: RESULT = {"IPAddress4":"10.254.11.253"}
```
for the initial wrong DNS: it is not clear if they use dhcpd or antered the wrong DNS manually.
would be a nice feature to indicate using dhcp on info page.