arendst/Tasmota #7445: add function to upgrade firmware from ftp server

smartboart, 2020-01-05:
**Have you looked for this feature in other issues and in the docs?**  
yes
**Is your feature request related to a problem? Please describe.**  
no
_A clear and concise description of what the problem is._
no  problem ,, request for new feature
**Describe the solution you'd like**  
I would like to upgrade all my tasmota devices automatically, as soon as a new firmware is available.
My devices are all offline and not connected tothe internet. With  iobroker i check for new updates. if available i download it automatically tomy ftpserver... But upgrade with ota url is not possible if a ftp link is entered.
_A clear and concise description of what you want to happen._
i would like to havethe possibility to upgrade ota via ftp... 

to complete it: If ftp will work i would like to upgrade with an web command automatically ..
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-01-05:
Why don't you set up an HTTP server instead of FTP server? Esp8266 ressources are VERY limited and there is little chance we would add FTP support.

ascillato2, 2020-01-05:
Thanks for sharing your ideas. As the resources are very restricted on an esp8266 chip, not all possible features are going to be added. If you want, you can provide a Pull Request with a new ftp update routine as optional.

At this moment, Tasmota supports OTA upgrades from a http url, from a file (using the Tasmota webui), and from arduinoOTA port. All these 3 features are part of the Arduino core and are optimized for the esp8266. If you want, you should ask in Arduino Core repository if they are going to add a FTP OTA upgrade option and if it is possible to use that in an esp8266 chip. 

Added the label Requested feature (hold over) for future reference.

closed 2020-01-05 by ascillato2