arendst/Tasmota #7447: Baudrate <x> command seems to store the changed value, but after reboot the device reverts to a default value

teixeluis, 2020-01-05:
### PROBLEM DESCRIPTION

Having flashed Tasmota 8.1.0 in a Sonoff Dual R2 module, I tested the Baudrate command, by changing the setting:

```
15:28:24 CMD: Baudrate 9600
15:28:24 SRC: WebConsole from 192.168.1.150
15:28:24 CMD: Group 0, Index 1, Command "BAUDRATE", Data "9600"
15:28:24 APP: Set Serial to 8N1 9600 bit/s
15:28:24 MQT: stat/dual-switch/RESULT = {"Baudrate":9600}
15:28:24 CFG: Saved to flash at F8, Count 172, Bytes 4096
```

And after performing a reboot:

```
15:30:29 CMD: Baudrate
15:30:29 SRC: WebConsole from 192.168.1.150
15:30:29 CMD: Group 0, Index 1, Command "BAUDRATE", Data ""
15:30:29 MQT: stat/dual-switch/RESULT = {"Baudrate":115200}
```

It falls back to the original value.

This can be circumvented by creating a rule that is triggered by the System#Boot event, but I think is not the ideal approach though (the setting seems to be saved to flash after the command).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Dual R2
- [x] Tasmota binary firmware version number used: 8.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

19:49:11 CMD: Backlog Template; Module; GPIO 255
19:49:11 SRC: WebConsole from 192.168.1.150
19:49:11 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
19:49:11 SRC: Backlog
19:49:11 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
19:49:11 MQT: stat/dual-switch/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
19:49:12 SRC: Backlog
19:49:12 CMD: Group 0, Index 1, Command "MODULE", Data ""
19:49:12 MQT: stat/dual-switch/RESULT = {"Module":{"39":"Sonoff Dual R2"}}
19:49:12 SRC: Backlog
19:49:12 CMD: Group 0, Index 1, Command "GPIO", Data "255"
19:49:12 MQT: stat/dual-switch/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"22":"Relay2"},"GPIO9":{"0":"None"},"GPIO10":{"17":"Button1"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

19:50:03 CMD: Status 0
19:50:03 SRC: WebConsole from 192.168.1.150
19:50:03 CMD: Group 0, Index 1, Command "STATUS", Data "0"
19:50:03 MQT: stat/dual-switch/STATUS = {"Status":{"Module":39,"FriendlyName":["dual-switch","Tasmota2"],"Topic":"dual-switch","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
19:50:03 MQT: stat/dual-switch/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T04:19:50","StartupUTC":"2020-01-05T14:30:13","Sleep":50,"CfgHolder":4617,"BootCount":33,"SaveCount":174,"SaveAddress":"F6000"}}
19:50:03 MQT: stat/dual-switch/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"403/699"}}
19:50:03 MQT: stat/dual-switch/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["griffinnet-zh","griffinnet-zh-ext"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","2805C8000100060000005A00000000000000","00000040","00000000"]}}
19:50:03 MQT: stat/dual-switch/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
19:50:03 MQT: stat/dual-switch/STATUS5 = {"StatusNET":{"Hostname":"dual-switch-3114","IPAddress":"192.168.1.33","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:98:0C:2A","Webserver":2,"WifiConfig":4}}
19:50:03 MQT: stat/dual-switch/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.20","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_980C2A","MqttUser":"sonoff-device","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
19:50:03 MQT: stat/dual-switch/STATUS7 = {"StatusTIM":{"UTC":"Sun Jan 05 18:50:03 2020","Local":"Sun Jan 05 19:50:03 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:07"}}
19:50:03 MQT: stat/dual-switch/STATUS10 = {"StatusSNS":{"Time":"2020-01-05T19:50:03"}}
19:50:03 MQT: stat/dual-switch/STATUS11 = {"StatusSTS":{"Time":"2020-01-05T19:50:03","Uptime":"0T04:19:50","UptimeSec":15590,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"griffinnet-zh","BSSId":"E8:94:F6:F5:3C:14","Channel":11,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

19:52:42 CMD: Baudrate
19:52:42 SRC: WebConsole from 192.168.1.150
19:52:42 CMD: Group 0, Index 1, Command "BAUDRATE", Data ""
19:52:42 MQT: stat/dual-switch/RESULT = {"Baudrate":115200}

19:53:10 CMD: Baudrate 9600
19:53:10 SRC: WebConsole from 192.168.1.150
19:53:10 CMD: Group 0, Index 1, Command "BAUDRATE", Data "9600"
19:53:10 APP: Set Serial to 8N1 9600 bit/s
19:53:10 MQT: stat/dual-switch/RESULT = {"Baudrate":9600}
19:53:11 CFG: Saved to flash at F9, Count 179, Bytes 4096

19:53:30 CMD: Restart 1
19:53:30 SRC: WebConsole from 192.168.1.150
19:53:30 CMD: Group 0, Index 1, Command "RESTART", Data "1"
19:53:30 MQT: stat/dual-switch/RESULT = {"Restart":"Restarting"}

<restarted>

19:53:52 CMD: Baudrate
19:53:52 SRC: WebConsole from 192.168.1.150
19:53:52 CMD: Group 0, Index 1, Command "BAUDRATE", Data ""
19:53:52 MQT: stat/dual-switch/RESULT = {"Baudrate":115200}
```

### TO REPRODUCE
Change the baudrate to a value different from 115200, restart the device, check the baudrate setting.

### EXPECTED BEHAVIOUR
The baudrate setting should be persisted after a restart.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-01-05:
Fixed in latest development version.
You can download at http://thehackbox.org/tasmota/tasmota.bin

teixeluis, 2020-01-05:
> Fixed in latest development version.
> You can download at http://thehackbox.org/tasmota/tasmota.bin

Awesome!

closed 2020-01-05 by ascillato2

reimerp, 2020-12-06:
Seems to be  present in 9.1.0 agaIn.

lalo-uy, 2020-12-07:
Confirm on 9.1.0 released.

MissingTwins, 2024-05-04:
Broken again on 
[image]
```
00:00:00.001 HDW: ESP8266EX
00:00:00.054 CFG: Loaded from flash at F9, Count 91
00:00:00.059 QPC: Count 1
00:00:00.071 NRG: Init driver 14
00:00:00.074 Project tasmota - Tasmota Version 13.4.0(tasmota)-2_7_6(2024-02-14T16:13:56)
00:00:00.110 SNS: Hardware Serial
00:00:01.001 WIF: Connecting to AP1 Channel 1 BSSId xxxxxxxxx in mode 11n as gosund-21-1659...
00:00:02.516 WIF: Connected
00:00:02.767 HTP: Web server active on gosund-21-1659 with IP address 192.168.5.228
07:21:24.281 MQT: Attempting connection...
07:21:24.438 MQT: Connected
07:21:24.441 MQT: tele/gosund_21/LWT = Online (retained)
07:21:24.443 MQT: cmnd/gosund_21/POWER = 
07:21:24.448 MQT: tele/gosund_21/INFO1 = {"Info1":{"Module":"Gosund CP1-AM","Version":"13.4.0(tasmota)","FallbackTopic":"cmnd/DVES_7D267B_fb/","GroupTopic":"cmnd/tasmotas/"}}
07:21:24.451 MQT: tele/gosund_21/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"gosund-21-1659","IPAddress":"192.168.5.228"}}
07:21:24.454 MQT: tele/gosund_21/INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":16}}
07:21:24.456 MQT: stat/gosund_21/RESULT = {"POWER":"ON"}
07:21:24.457 MQT: stat/gosund_21/POWER = ON
07:21:26.252 QPC: Reset
07:21:29.216 MQT: tele/gosund_21/STATE = {"Time":"2024-05-04T07:21:29","Uptime":"0T00:00:10","UptimeSec":10,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"HAHAHAHAH","BSSId":"XXXXXXXX","Channel":1,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:04"}}
07:21:29.225 MQT: tele/gosund_21/SENSOR = {"Time":"2024-05-04T07:21:29","ENERGY":{"TotalStartTime":"2024-05-04T05:58:56","Total":0.011,"Yesterday":0.000,"Today":0.011,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
07:21:46.240 CMD: Baudrate
07:21:46.247 MQT: stat/gosund_21/RESULT = {"Baudrate":4800}
07:21:54.950 CMD: Baudrate 9600
07:21:54.953 SER: Set to 8N1 9600 bit/s
07:21:54.960 MQT: stat/gosund_21/RESULT = {"Baudrate":9600}
07:21:56.253 CMD: Baudrate
07:21:56.259 MQT: stat/gosund_21/RESULT = {"Baudrate":9600}
```

sfromis, 2024-05-04:
What you post is not relevant to digging up some old thread about a different issue. Support for energy monitoring chip is by specific drivers, and the `Baudrate` command is only when explicitly configuring Tasmota for serial communication (which does not mean when some driver may "happen" to be using serial).

And when it comes to the driver for BL0942, this driver has specific code supporting options for the baud rate, selected by the configuration dropdown for "BL0942 Rx". If I read the code right, dropdown value 2 is what you want
`// Support 1 (4800), 2 (9600), 3 (19200), 4 (38400)`

Do **not** try to use the `Baudrate` command.

MissingTwins, 2024-05-04:
@sfromis I appreciate your prompt response and clarification.
 I apologize for posting in the incorrect thread.

This is very helpful, I was always wondering, what dose the following drop-down numbers do.
`// Support 1 (4800), 2 (9600), 3 (19200), 4 (38400)`

Thanks, and sorry again.