arendst/Tasmota #7449: Implement google assistant/home local execution api

qwertangel, 2020-01-06:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
Once we flash tasmota on a device, it loses the ability to natively be controlled by google assistant. That is because google assistant doesn't have a good way to send command to tasmota sitting behind NAT. alternatives (opening port, using mqtt bridge etc) are not that easy to setup.

**Describe the solution you'd like**  
Google has released Local execution api. Which uses google home devices sitting inside NAT to make local http calls thus solving original issue. So now assistant can send command to google home device which will then make http/tcp/udp calls in local network. This will be perfect for tasmota. It seems tasmota will need to implement few specific json responses (related to discovery and status). It doesn't seem it will add too much code.

**Additional context**  
https://developers.google.com/assistant/smarthome/concepts/local

Jason2866, 2020-01-06:
Thx for sharing your ideas/findings. A PR is always welcome.

slimcdk, 2020-01-11:
This is due to no access to severs that are able to parse the request.

qwertangel, 2020-01-11:
the way I see it, there is need for an google action for tasmota which people would need to add to their assistant. when they speak to google assistant, that would use that action to relay the command to local google home devices which would control tasmota using local wifi.
I am not sure I understand your comment about 'no access to severs that are able to parse the request'.

slimcdk, 2020-01-11:
For the execution to be local only, the GH has to do the inference of the audio clip it recorded when you spoke to it on its internal hardware. This is a heavy computation. So to overcome this, GH only recognize "Hey Google" and handles the audio clip to Google servers for faster computation. The request is sent directly from Googles servers to the servers of your device vendor, which you have connected your device to. uPnP in your router automatically handles the traffic to your device. 

Tasmota does not have any servers on the internet to connect to, nor can Google send the request directly to your device as it is hidden behind your router. For this to work you'll have to configure domains with SSL pointing to your network and open ports to devices.

qwertangel, 2020-01-11:
yup i understand. that's how traditionally it worked. however,  if you take a look at how local sdk works , it allows a new path. attaching image. here, tasmota would be at bottom right. 
[image]

the upper right path (cloud fallback) would not be required (or can go to dummy link)

markg85, 2020-01-18:
> yup i understand. that's how traditionally it worked. however, if you take a look at how local sdk works , it allows a new path. attaching image. here, tasmota would be at bottom right.
> [image]
> 
> the upper right path (cloud fallback) would not be required (or can go to dummy link)

Ohh, i wish this were true in the exact term you said there.
Sadly it's not *that* awesome.

Google did a bit of a nasty thing here. Yes, you can have a local execution path, but the very first initial SYNC command to "discover" devices has to be done in the cloud. So in other terms, you have to register your devices in the plain old way in the cloud app. You then then tell google (in the cloud app) that some devices support local execution. Kinda like saying to google: "hey, great you contact my cloud service, but you can skip me and do it locally instead". When google knows that it will try to use the local execution path.

Quote (from [here](https://developers.google.com/assistant/smarthome/develop/local)):

> A local execution path is established when Google matches a locally-controllable device to a device returned in the SYNC response from your cloud fulfillment. 

So sadly.. you still need the cloud service. Just a lot less.

ascillato2, 2020-02-04:
Closing as nobody is working on this.

If anyone wants to work on this and make a PoC (proof of concept) that this approach can be programmed in an small ESP8266, please do a PR (Pull Request). If anyone needs help on that, please ask. Thanks.

closed 2020-02-04 by ascillato2