arendst/Tasmota #7457: Hiking DDS238 energy meter bad energy values

RedCat30, 2020-01-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Hello

I have Hiking DDS238-2 ZN/S energy meter too. I have flashed the latest Tasmota-sensors.bin version (7.2.0)
I found this page yesterday. So far for me, energy today, yesterday, total, export active showed zero. Yesterday I tried SetOption71 1, it worked partially. Instead of zeros there are values, only wrong values.
Take a look at the two pictures. One is the inverter, today's output is 1.5 kWh, the other is Tasmota, today production is 0.001 kWh. Yesterday's values were also bad, according to the inverter 7.92 kWh.

[image]
[image]

Voltage, power, current, frequency values are good.
Is it possible that this is a third version of DDS238? Can anyone help me how to get the right values (Today, yesterday, total)?
Thanks!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-01-07:
Please, update your Tasmota to latest

pablozg, 2020-01-07:
And if you want have the totals from the hardware counter, use setoption72 1 too.

RedCat30, 2020-01-07:
I updated to 8.1.0, and I used setcommand71 1 and setcommand72 1.
Today, yesterday, total energy show something, but it is wrong.

[image]

Now the sun is shine, energy production 1500w, but a today and total energy not changes.

RedCat30, 2020-01-07:
1 hour later:

[image]

Something changes, but the values are wrong.

Thx for help :)

arendst, 2020-01-07:
I think you draw the wrong conclusions. The webpage will ONLY show usage. It does not show what you  produce with your solar panels.

Production energy is only shown in the JSON data send at teleperiod intervals.

RedCat30, 2020-01-07:
I hooked up a 1500w consumer (hair dryer). I turned it on for 15 minutes. Current power (1485w) showed good value. But today and total energy are bad. (In this case, yesteday energy cannot be interpreted.)

The problem: Today I used 8,2 kw of energy, why doesn't today's energy show this? Today energy show me 0.001 KWh. Total energy show 0.059 kWh. Here's the bug.

arendst, 2020-01-07:
Enable more logging with command ``weblog 4`` and report the data received from the device.

Play with the load and see if anything changes in the received data.

RedCat30, 2020-01-08:
Hello

Now 1000w cunsumer, weblog 4 command:

09:42:58 CMD: weblog 4
09:42:58 MQT: otthon/stat/BW_napelem_mero/RESULT = {"WebLog":4}
09:42:58 DMP: 01 03 24 00 00 1C 7B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 88 09 77 01 A5 03 F7 00 73 03 E1 13 8A CC 1D
09:42:58 CFG: Saved to flash at F4, Count 35107, Bytes 4096
09:42:59 DMP: 01 03 24 00 00 1C 7B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 88 09 78 01 A5 03 F6 00 75 03 E1 13 89 24 C8
09:42:59 CFG: Saved to flash at FB, Count 35108, Bytes 4096
09:43:00 DMP: 01 03 24 00 00 1C 7B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 88 09 77 01 A5 03 F6 00 74 03 E1 13 89 29 1C
09:43:00 CFG: Saved to flash at FA, Count 35109, Bytes 4096
09:43:01 DMP: 01 03 24 00 00 1C 7B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 88 09 77 01 A5 03 F6 00 76 03 E1 13 8A 10 DD
09:43:01 CFG: Saved to flash at F9, Count 35110, Bytes 4096
09:43:02 DMP: 01 03 24 00 00 1C 7B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 88 09 77 01 A5 03 F5 00 75 03 E1 13 8B A6 1D
09:43:02 CFG: Saved to flash at F8, Count 35111, Bytes 4096
09:43:03 DMP: 01 03 24 00 00 1C 7B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 88 09 77 01 A5 03 F6 00 74 03 E1 13 8A 69 1D
09:43:03 CFG: Saved to flash at F7, Count 35112, Bytes 4096
09:43:04 DMP: 01 03 24 00 00 1C 7B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 88 09 78 01 A5 03 F7 00 70 03 E1 13 89 F8 08
09:43:04 CFG: Saved to flash at F6, Count 35113, Bytes 4096
09:43:05 DMP: 01 03 24 00 00 1C 7B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 88 09 77 01 A5 03 F8 00 70 03 E1 13 8A 77 1D
09:43:05 MQT: otthon/tele/BW_napelem_mero/STATE = {"Time":"2020-01-08T09:43:05","Uptime":"0T02:33:23","UptimeSec":9203,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":2,"SSId":"Otthon","BSSId":"44:D9:E7:2F:32:4E","Channel":11,"RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:06"}}
09:43:05 MQT: otthon/tele/BW_napelem_mero/SENSOR = {"Time":"2020-01-08T09:43:05","ENERGY":{"TotalStartTime":"2020-01-08T09:43:05","Total":0.060,"Yesterday":0.001,"Today":0.001,"ExportActive":12.670,"Period":0,"Power":1016,"ApparentPower":1020,"ReactivePower":112,"Factor":0.99,"Frequency":50,"Voltage":242,"Current":4.210}}
09:43:05 CFG: Saved to flash at F5, Count 35114, Bytes 4096
09:43:06 DMP: 01 03 24 00 00 1C 7B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 88 09 77 01 A6 03 F8 00 70 03 E1 13 8A 63 ED
09:43:06 CFG: Saved to flash at F4, Count 35115, Bytes 4096
09:43:07 DMP: 01 03 24 00 00 1C 7B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 88 09 77 01 A6 03 F8 00 70 03 E1 13 8A 63 ED
09:43:07 CFG: Saved to flash at FB, Count 35116, Bytes 4096
09:43:08 DMP: 01 03 24 00 00 1C 7B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 88 09 76 01 A6 03 F8 00 70 03 E1 13 8B A6 D1
09:43:08 CFG: Saved to flash at FA, Count 35117, Bytes 4096
09:43:09 DMP: 01 03 24 00 00 1C 7B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 88 09 77 01 A6 03 F8 00 72 03 E1 13 8B DB ED
09:43:09 CFG: Saved to flash at F9, Count 35118, Bytes 4096
09:43:10 WIF: Checking connection...
09:43:10 WIF: Connected
09:43:10 DMP: 01 03 24 00 00 1C 7B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 88 09 77 01 A6 03 F8 00 73 03 E1 13 8B E6 2D
09:43:10 CFG: Saved to flash at F8, Count 35119, Bytes 4096
09:43:11 DMP: 01 03 24 00 00 1C 7C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 89 09 76 01 A6 03 F8 00 70 03 E1 13 8A 60 51
09:43:11 CFG: Saved to flash at F7, Count 35120, Bytes 4096
09:43:12 DMP: 01 03 24 00 00 1C 7C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 89 09 73 01 A6 03 F9 00 6E 03 E2 13 8B F8 9F
09:43:12 CFG: Saved to flash at F6, Count 35121, Bytes 4096
09:43:13 DMP: 01 03 24 00 00 1C 7C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 89 09 73 01 A6 03 F9 00 6E 03 E2 13 8B F8 9F
09:43:13 CFG: Saved to flash at F5, Count 35122, Bytes 4096
09:43:14 DMP: 01 03 24 00 00 1C 7C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 89 09 73 01 A6 03 F9 00 6D 03 E2 13 8B BC 9F
09:43:14 CFG: Saved to flash at F4, Count 35123, Bytes 4096
09:43:15 DMP: 01 03 24 00 00 1C 7C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 89 09 74 01 A6 03 F9 00 70 03 E1 13 8B BA E9
09:43:15 MQT: otthon/tele/BW_napelem_mero/STATE = {"Time":"2020-01-08T09:43:15","Uptime":"0T02:33:33","UptimeSec":9213,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":2,"SSId":"Otthon","BSSId":"44:D9:E7:2F:32:4E","Channel":11,"RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:06"}}
09:43:15 MQT: otthon/tele/BW_napelem_mero/SENSOR = {"Time":"2020-01-08T09:43:15","ENERGY":{"TotalStartTime":"2020-01-08T09:43:15","Total":0.060,"Yesterday":0.001,"Today":0.001,"ExportActive":12.670,"Period":0,"Power":1017,"ApparentPower":1021,"ReactivePower":112,"Factor":0.99,"Frequency":50,"Voltage":242,"Current":4.220}}
09:43:15 CFG: Saved to flash at FB, Count 35124, Bytes 4096
09:43:16 DMP: 01 03 24 00 00 1C 7C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 89 09 75 01 A6 03 F9 00 70 03 E1 13 8B BE 15
09:43:16 CFG: Saved to flash at FA, Count 35125, Bytes 4096
09:43:17 DMP: 01 03 24 00 00 1C 7C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 89 09 75 01 A6 03 F9 00 71 03 E1 13 8B 83 D5
09:43:17 CFG: Saved to flash at F9, Count 35126, Bytes 4096
[... cut ...] 4.260}}
09:44:15 CFG: Saved to flash at F7, Count 35184, Bytes 4096
09:44:16 DMP: 01 03 24 00 00 1C 7D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 8A 09 78 01 AA 04 04 00 72 03 E1 13 89 2D 92
09:44:16 CFG: Saved to flash at F6, Count 35185, Bytes 4096
09:44:17 DMP: 01 03 24 00 00 1C 7D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 8A 09 7B 01 AA 04 04 00 72 03 E1 13 89 22 D6
09:44:17 CFG: Saved to flash at F5, Count 35186, Bytes 4096

It is ok and enough?

Thx for help :)

arendst, 2020-01-08:
From the log I see:
- You have a version 2 device so you should set ``SetOption71 1``
- Every second a flash write is performed which will ruin your hardware very soon. This is probably caused by the wrong total value it receives. So execute ``SetOption72 0`` for now.

With the above recommendations resend a new log please.

RedCat30, 2020-01-08:
After 10 mins:

09:57:14 CFG: Saved to flash at F4, Count 35963, Bytes 4096
09:57:15 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 79 01 C7 04 4C 00 6A 03 E3 13 89 10 31
09:57:15 MQT: otthon/tele/BW_napelem_mero/STATE = {"Time":"2020-01-08T09:57:15","Uptime":"0T02:47:33","UptimeSec":10053,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Wifi":{"AP":2,"SSId":"Otthon","BSSId":"44:D9:E7:2F:32:4E","Channel":11,"RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:06"}}
09:57:15 MQT: otthon/tele/BW_napelem_mero/SENSOR = {"Time":"2020-01-08T09:57:15","ENERGY":{"TotalStartTime":"2020-01-08T09:57:15","Total":0.060,"Yesterday":0.001,"Today":0.001,"ExportActive":12.670,"Period":0,"Power":1100,"ApparentPower":1103,"ReactivePower":106,"Factor":1.00,"Frequency":50,"Voltage":243,"Current":4.550}}
09:57:15 CFG: Saved to flash at FB, Count 35964, Bytes 4096
09:57:16 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 79 01 C8 04 4D 00 6A 03 E3 13 89 41 01
09:57:16 CFG: Saved to flash at FA, Count 35965, Bytes 4096
09:57:17 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 79 01 C8 04 4D 00 68 03 E3 13 88 F9 01
09:57:17 CFG: Saved to flash at F9, Count 35966, Bytes 4096
09:57:18 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 79 01 C8 04 4E 00 69 03 E3 13 89 36 01
09:57:18 CFG: Saved to flash at F8, Count 35967, Bytes 4096
09:57:19 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 79 01 C8 04 4E 00 6A 03 E3 13 88 B3 C1
09:57:19 CFG: Saved to flash at F7, Count 35968, Bytes 4096
09:57:20 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 78 01 C8 04 4E 00 69 03 E3 13 88 F3 3D
09:57:20 CFG: Saved to flash at F6, Count 35969, Bytes 4096
09:57:21 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 78 01 C8 04 4E 00 69 03 E3 13 88 F3 3D
09:57:21 CFG: Saved to flash at F5, Count 35970, Bytes 4096
09:57:22 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 79 01 C8 04 4E 00 6C 03 E3 13 89 FA 01
09:57:22 CFG: Saved to flash at F4, Count 35971, Bytes 4096
09:57:23 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 78 01 C8 04 4D 00 71 03 E2 13 89 70 3F
09:57:23 CFG: Saved to flash at FB, Count 35972, Bytes 4096
09:57:24 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 78 01 C8 04 4C 00 6F 03 E2 13 89 C8 FD
09:57:24 CFG: Saved to flash at FA, Count 35973, Bytes 4096
09:57:25 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 78 01 C8 04 4C 00 70 03 E2 13 89 5D 3F
09:57:25 MQT: otthon/tele/BW_napelem_mero/STATE = {"Time":"2020-01-08T09:57:25","Uptime":"0T02:47:43","UptimeSec":10063,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":2,"SSId":"Otthon","BSSId":"44:D9:E7:2F:32:4E","Channel":11,"RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:06"}}
09:57:25 MQT: otthon/tele/BW_napelem_mero/SENSOR = {"Time":"2020-01-08T09:57:25","ENERGY":{"TotalStartTime":"2020-01-08T09:57:25","Total":0.060,"Yesterday":0.001,"Today":0.001,"ExportActive":12.670,"Period":0,"Power":1100,"ApparentPower":1105,"ReactivePower":112,"Factor":0.99,"Frequency":50,"Voltage":242,"Current":4.560}}
09:57:25 CFG: Saved to flash at F9, Count 35974, Bytes 4096
09:57:26 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 79 01 C8 04 4C 00 72 03 E2 13 88 E1 C3
09:57:26 CFG: Saved to flash at F8, Count 35975, Bytes 4096
09:57:27 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 78 01 C8 04 4B 00 71 03 E2 13 88 D7 FF
09:57:27 HTP: Console
09:57:27 CFG: Saved to flash at F7, Count 35976, Bytes 4096
09:57:28 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 79 01 C8 04 4C 00 71 03 E2 13 89 64 03
09:57:28 CFG: Saved to flash at F6, Count 35977, Bytes 4096
09:57:29 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 78 01 C8 04 4D 00 71 03 E2 13 88 B1 FF
09:57:29 CFG: Saved to flash at F5, Count 35978, Bytes 4096
09:57:30 WIF: Checking connection...
09:57:30 WIF: Connected
09:57:30 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 77 01 C9 04 4E 00 70 03 E2 13 87 C2 BF
09:57:30 CFG: Saved to flash at F4, Count 35979, Bytes 4096
09:57:31 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 76 01 C9 04 4D 00 6F 03 E2 13 8B 60 84
09:57:31 CFG: Saved to flash at FB, Count 35980, Bytes 4096
09:57:32 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 72 01 C9 04 4D 00 6D 03 E3 13 88 1D B5
09:57:32 CFG: Saved to flash at FA, Count 35981, Bytes 4096
09:57:33 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 76 01 C8 04 4D 00 6E 03 E3 13 88 41 15
09:57:33 CFG: Saved to flash at F9, Count 35982, Bytes 4096
09:57:34 DMP: 01 03 24 00 00 1C 95 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A2 09 78 01 C8 04 4C 00 6E 03 E3 13 89 A4 FD
09:57:34 CFG: Saved to flash at F8, Count 35983, Bytes 4096
[... cut ...]
09:58:15 MQT: otthon/tele/BW_napelem_mero/SENSOR = {"Time":"2020-01-08T09:58:15","ENERGY":{"TotalStartTime":"2020-01-08T09:58:15","Total":0.061,"Yesterday":0.001,"Today":0.001,"ExportActive":12.670,"Period":0,"Power":1092,"ApparentPower":1097,"ReactivePower":104,"Factor":1.00,"Frequency":50,"Voltage":242,"Current":4.540}}
09:58:15 CFG: Saved to flash at F7, Count 36024, Bytes 4096
09:58:16 DMP: 01 03 24 00 00 1C 97 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 A4 09 71 01 C6 04 45 00 68 03 E3 13 89 07 07
09:58:16 CFG: Saved to flash at F6, Count 36025, Bytes 4096

The energy values: 
Power 1100w OK
Today 0.001 kwh BAD
Yesterday 0.001 kwh BAD
Total 0.060 kwh BAD

arendst, 2020-01-08:
Also execute the following commands to get decimal information: ``Voltres 1``, ``Ampres 3``, `` Wattres 3``, ``Freqres 2``

RedCat30, 2020-01-08:
After setoption71 1, setoption72 0, Voltres 1, Ampres 3, Wattres 3, Freqres 2 commands:

10:09:24 CMD: SetOption72 0
10:09:24 SRC: WebConsole from 10.0.3.1
10:09:24 CMD: Group 0, Index 72, Command "SETOPTION", Data "0"
10:09:24 MQT: otthon/stat/BW_napelem_mero/RESULT = {"SetOption72":"OFF"}
10:09:25 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 77 01 F2 04 B3 00 6D 03 E3 13 87 74 93
10:09:26 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 78 01 F3 04 B4 00 6C 03 E3 13 87 02 17
10:09:27 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 78 01 F3 04 B5 00 6E 03 E3 13 88 2B 13
10:09:28 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 78 01 F3 04 B5 00 6D 03 E3 13 88 6F 13
10:09:29 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 77 01 F2 04 B3 00 6F 03 E3 13 87 0D 53
10:09:30 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 77 01 F2 04 B2 00 6F 03 E3 13 88 5D 97
10:09:31 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 77 01 F1 04 B0 00 6D 03 E3 13 87 53 63
10:09:32 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 77 01 F1 04 AF 00 6C 03 E3 13 86 41 A2
10:09:32 MQT: otthon/tele/BW_napelem_mero/STATE = {"Time":"2020-01-08T10:09:32","Uptime":"0T00:11:02","UptimeSec":662,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":2,"SSId":"Otthon","BSSId":"44:D9:E7:2F:32:4E","Channel":11,"RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:06"}}
10:09:32 MQT: otthon/tele/BW_napelem_mero/SENSOR = {"Time":"2020-01-08T10:09:32","ENERGY":{"TotalStartTime":"2020-01-08T00:00:00","Total":0.061,"Yesterday":0.001,"Today":0.001,"ExportActive":12.670,"Period":0.020,"Power":1199.000,"ApparentPower":1204.231,"ReactivePower":108.000,"Factor":1.00,"Frequency":49.98,"Voltage":242.3,"Current":4.970}}
10:09:33 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 77 01 F1 04 AF 00 6C 03 E3 13 86 41 A2
10:09:34 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 77 01 EE 04 AA 00 6D 03 E3 13 87 64 92
10:09:35 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 77 01 ED 04 A7 00 6E 03 E3 13 87 E8 A2
10:09:36 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 77 01 ED 04 A5 00 6D 03 E3 13 87 8F 62
10:09:37 WIF: Checking connection...
10:09:37 WIF: Connected
10:09:37 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 77 01 EC 04 A4 00 6D 03 E3 13 88 D2 36
10:09:38 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 77 01 EC 04 A3 00 6E 03 E3 13 86 61 32
10:09:39 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 77 01 EB 04 A2 00 6E 03 E3 13 87 96 02
10:09:40 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 77 01 EB 04 A1 00 6F 03 E3 13 86 59 02
10:09:41 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 78 01 EA 04 9F 00 6A 03 E3 13 86 77 45
10:09:42 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 78 01 E9 04 9D 00 6B 03 E3 13 86 7D B5
10:09:42 MQT: otthon/tele/BW_napelem_mero/STATE = {"Time":"2020-01-08T10:09:42","Uptime":"0T00:11:12","UptimeSec":672,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":2,"SSId":"Otthon","BSSId":"44:D9:E7:2F:32:4E","Channel":11,"RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:06"}}
10:09:42 MQT: otthon/tele/BW_napelem_mero/SENSOR = {"Time":"2020-01-08T10:09:42","ENERGY":{"TotalStartTime":"2020-01-08T00:00:00","Total":0.061,"Yesterday":0.001,"Today":0.001,"ExportActive":12.670,"Period":0.000,"Power":1181.000,"ApparentPower":1185.336,"ReactivePower":107.000,"Factor":1.00,"Frequency":49.98,"Voltage":242.4,"Current":4.890}}
10:09:43 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 79 01 E8 04 9C 00 6C 03 E3 13 86 D1 D9
10:09:44 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 7A 01 E9 04 9D 00 6C 03 E3 13 87 02 0D
10:09:45 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 7A 01 E9 04 9D 00 6E 03 E3 13 86 BA 0D
10:09:46 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 79 01 E9 04 9E 00 72 03 E3 13 87 96 4B
10:09:47 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 79 01 EA 04 9F 00 6F 03 E3 13 88 3E 7D
10:09:48 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 79 01 EB 04 A2 00 6F 03 E3 13 86 5E EA
10:09:49 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 78 01 EC 04 A4 00 6F 03 E3 13 87 DB E6
10:09:50 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 78 01 ED 04 A6 00 6F 03 E3 13 88 B5 B2
10:09:51 DMP: 01 03 24 00 00 1C AD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BA 09 78 01 ED 04 A7 00 6D 03 E3 13 87 9C B6
[... cut ...]
10:10:57 DMP: 01 03 24 00 00 1C B0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BD 09 77 01 EF 04 AC 00 6F 03 E3 13 87 01 81
10:10:58 DMP: 01 03 24 00 00 1C B0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BD 09 77 01 EF 04 AB 00 6C 03 E3 13 88 73 45
10:10:59 DMP: 01 03 24 00 00 1C B0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BD 09 75 01 F1 04 AD 00 72 03 E3 13 88 37 5F
10:11:00 DMP: 01 03 24 00 00 1C B0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 F3 00 00 17 BD 09 75 01 F1 04 AD 00 72 03 E3 13 88 37 5F

arendst, 2020-01-08:
You might want to try a change in file xnrg_09_dds2382.ino line 80

from
```
      EnergyUpdateTotal(import_active, false);  // 484.708 kWh
```
to
```
      EnergyUpdateTotal(import_active, true);  // 484.708 kWh
```
and see if energy total is incremented with real values.

RedCat30, 2020-01-08:
I've never tried compiled. Could you compiled it for me? Or where can I find a description? 
Thx! :)

arendst, 2020-01-08:
I'll do it for you. Just wait for the next commit in development binaries

arendst, 2020-01-08:
The updated files pop-up here http://thehackbox.org/tasmota/ within an hour.

RedCat30, 2020-01-08:
Thx a lot! :) I will try it!

This will fix the today and yesterday energy values too?

arendst, 2020-01-08:
yesterday is copy of today. So after a few days it should report as expected.

RedCat30, 2020-01-08:
I flashed this: 

tasmota-sensors.bin | http://thehackbox.org/tasmota/tasmota-sensors.bin | 592k | 20200108 13:01 GMT+2 | 8.1.0.3 | a15528a

I have 2 screenshot:
Consumer: 400w, in this 20 min allways on:

13:22
[image]

13:42
[image]

I think nothing changes :(

RedCat30, 2020-01-08:
> 
> 
> yesterday is copy of today. So after a few days it should report as expected.

Yes, but the "today" value bad too. Take a look my last 2 screenshots.

arendst, 2020-01-08:
Energy Today is changing as expected.

Energy Yesterday won't change as it will reflect it's state at 00:00 by copying over the then Energy.today value

Energy Total will increment too.

With Setoption72 1 this is the expected result as it uses the tasmota counters and NOT the DS238 counters.

Wait a few days and see if counters update. Once they do do SetOption72 0 and watch the real counters. You may ahve to use EnergyReset to pre-eset the counters.

RedCat30, 2020-01-08:
[image]

Energy Today not change.

13:42 0.004 kWh this is 4w
14:22 0.004 kWh this is 4w too. Not changes.

But 400w consumer running 50 mins.

I will wait 1-2 days, and we will see the changes.

arendst, 2020-01-08:
Could you pls provide the output of command ``status 0``

RedCat30, 2020-01-08:
Status 0 command:

14:37:30 CMD: status 0
14:37:30 SRC: WebConsole from 10.0.3.1
14:37:30 CMD: Group 0, Index 1, Command "STATUS", Data "0"
14:37:30 MQT: otthon/stat/BW_napelem_mero/STATUS = {"Status":{"Module":18,"FriendlyName":["Energy Meter Solar"],"Topic":"BW_napelem_mero","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:37:30 MQT: otthon/stat/BW_napelem_mero/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-sensors.bin","RestartReason":"Software/System restart","Uptime":"0T01:16:08","StartupUTC":"2020-01-08T12:21:22","Sleep":50,"CfgHolder":4617,"BootCount":130,"SaveCount":36051,"SaveAddress":"FA000"}}
14:37:30 MQT: otthon/stat/BW_napelem_mero/STATUS2 = {"StatusFWR":{"Version":"8.1.0(sensors)","BuildDateTime":"2019-12-25T12:38:43","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"370/699"}}
14:37:30 MQT: otthon/stat/BW_napelem_mero/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Home","Otthon"],"TelePeriod":10,"Resolution":"559B90C0","SetOption":["0000800B","2805C8000100060000005A00000000000000","00200000","00000000"]}}
14:37:30 MQT: otthon/stat/BW_napelem_mero/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":23,"ProgramFlashSize":1024,"FlashSize":16384,"FlashChipId":"1840EF","FlashMode":2,"Features":["00000809","8FDAE387","0437A3A4","B7FFBFCD","01DA9BC0","0016FCC7"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,18,19,20,21,22,24,26,29,30","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,34,37,39,40,42,43,51,52,55,56,58"}}
14:37:30 MQT: otthon/stat/BW_napelem_mero/STATUS5 = {"StatusNET":{"Hostname":"Napelem_mero","IPAddress":"10.0.0.81","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.1","Mac":"84:0D:8E:A9:F8:D3","Webserver":2,"WifiConfig":4}}
14:37:30 MQT: otthon/stat/BW_napelem_mero/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.51","MqttPort":1883,"MqttClientMask":"BW_napelem_mero","MqttClient":"BW_napelem_mero","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
14:37:30 MQT: otthon/stat/BW_napelem_mero/STATUS7 = {"StatusTIM":{"UTC":"Wed Jan 08 13:37:30 2020","Local":"Wed Jan 08 14:37:30 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:42","Sunset":"17:11"}}
14:37:30 MQT: otthon/stat/BW_napelem_mero/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
14:37:30 MQT: otthon/stat/BW_napelem_mero/STATUS10 = {"StatusSNS":{"Time":"2020-01-08T14:37:30","ENERGY":{"TotalStartTime":"2020-01-08T00:00:00","Total":0.063,"Yesterday":0.001,"Today":0.004,"ExportActive":12.670,"Power":179.000,"ApparentPower":228.138,"ReactivePower":142.000,"Factor":0.78,"Frequency":49.97,"Voltage":242.7,"Current":0.940}}}
14:37:30 MQT: otthon/stat/BW_napelem_mero/STATUS11 = {"StatusSTS":{"Time":"2020-01-08T14:37:30","Uptime":"0T01:16:08","UptimeSec":4568,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":2,"SSId":"Otthon","BSSId":"F0:9F:C2:7A:23:30","Channel":1,"RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:06"}}}

arendst, 2020-01-08:
That confirmes my expectation; you're still running the "old" sensor version 8.1.0 and not the development version with my latest change.

You will have to install the correct binary to be downloaded from the link I've shown before. You will need to update your ``otaurl`` to make this possible.

RedCat30, 2020-01-08:
I'm sorry, it's my fault. I corrected the otalink. I come back tomorrow with new energy values.

RedCat30, 2020-01-10:
Tegnap nem volt időm tesztelni. A mai Alakul a helyzet.
3 órával ezelőtt bekapcsoltam egy 1,5 kw fogyaztót. Nem néztem meg a dds238 tasmota weboldalát.
2 óra után, megnéztem. Az alábbi értékeket mutatja:

[image]

Újabb 30 perc elteltével, ismét megnéztem:

[image]

A szolgáltató álltal mért mai fogyasztásom:

[image]

I notice that dds238 seems to be "asleep" before first opening the Tasmota webpage. After I open webpage, the values seem good. It is possible?

I'll look at the values later.

RedCat30, 2020-01-11:
Hello 

Something wrong. Look at my screenshot:

[image]

Yesterday energy value is bad. The today’s energy value is low. The real value is 6.49 kWh. 

If necessary and help you then i will give you access to my dds238 device. I open a port and you can check it out. Or what can I do?

arendst, 2020-01-11:
If you remove power form the tasmota device you will lose the daily energy values. So make sure your tasmota device is always powered on for at least 2 days to check Energy Today and Energy Yesterday.

ascillato2, 2020-01-15:
@RedCat30

Hi, any news on this?

RedCat30, 2020-01-15:
I tested it for a few days: I think it doesn't work well. Here is today's example:
Today's consumption is 3.16 kW according to the service provider's electricity meter.

[image]

Check out the values of Tasman's dds238:
[image]

It shows totally bad values. The values were not reset. This is shown after 3 days of continuous operation.

ascillato2, 2020-01-16:
Hi,

In your status 0 can be seen "SerialLog":2

As the template was never fulfilled, we can't see which gpios are you using. If you are using TX/RX GPIOs for power measurement, it will interfere with the data. Try `seriallog 0`

RedCat30, 2020-01-17:
Hi

Sorry, this is my first post here. What do I need to fill out?

Here is a picture of the module settings.
[image]

 I try seriallog 0. In the afternoon I will report on the changes.

ascillato2, 2020-01-19:
Hi, any news on this?

RedCat30, 2020-01-20:
Hi!

Unfortunately, as you can see there is no change. Don't know what to try? 

[image]

You see, what small numbers dds238 shows, these values are not real.

ascillato, 2020-01-20:
Your power factor is very low. I don't think that is correct. What do you have connected?

RedCat30, 2020-01-20:
Check out my previous posts. There the pictures show that the power factor seems good, many times 1.00 (lamp).

The power factor is not the problem. Daily, yesterday and total energy values are bad.

gomes89, 2020-02-11:
It seems I have a similar problem with PZEM. Counters are not accurate at all. I changed the Energy Total to hardware counter with SetOption72 and the value is completely different. As for the Energy Today, it keeps resetting, and I noticed that it resets when I have a strange period value, e.g. "Period":42949392. Any ideas?

ascillato, 2020-02-12:
Please, recheck your wiring and connections. We had some older cases where wiring were not doing a good contact and also some other cases where the adaptation of voltages were incorrect (like sending rs482 to the esp or like sending 5v to the esp)

gomes89, 2020-02-12:
> Please, recheck your wiring and connections. We had some older cases where wiring were not doing a good contact and also some other cases where the adaptation of voltages were incorrect (like sending rs482 to the esp or like sending 5v to the esp)

Wiring and connections are fine. The reported values are all correct (I checked them with a multimeter clamp), and hardware counter for Energy Total is also fine (SetOption72). The only issue is with Tasmota counter (Energy Today) that keep resetting throughout the day.

As for voltages, this guide was followed and PZEM is being powered from 5V, but there is a 1K resistor between 5V and the optocoupler as instructed:
https://github.com/arendst/Tasmota/wiki/PZEM004T-Energy-Monitor

closed 2020-03-14 by stale[bot]

thiagoarreguy, 2020-06-05:
> It seems I have a similar problem with PZEM. Counters are not accurate at all. I changed the Energy Total to hardware counter with SetOption72 and the value is completely different. As for the [...]

Hi Gomes, are you from Brazil? can you add me on whatsapp (31)99467-5735