arendst/Tasmota #7467: LE LampUX light with dimmer + color temp, odd config?

stleusc, 2020-01-09:
### PROBLEM DESCRIPTION
hello, new to tasmota and trying to configure a led light that supports dimming and color temp.
Template is set to: {"NAME":"LE LampUx","GPIO":[0,0,0,0,0,37,0,0,38,0,0,0,0],"FLAG":0,"BASE":18}
I set SetOption68 1 in order to be able to play with each PWM.
Here what I found out using the GUI
Slider 1: controls over ON/OFF and brightness, NOT anything related to TEMP
Slider 2: control ONLY the temp

I think I am too stupid to figure out how to make it work with the usual sliders, SetOption68 0 shows the usual sliders but completly wrong behaviour.
How can I tell Tasmota that I have PWM for changing brightness and one for Colortemp. Seems like default is one PWM for warm and one for cold. But not here....

more settings I tried with SetOption68 0
GPIO5 PWM1, controls on/off brightness
GPIO5 RELAY1 and GPIO12 PWM1, Toggle1 is ON/OFF, 'Brighness slider (PWM1)' changes Color Temp

So I need to teach Tasmoto that I have 1 PWM for brightness and one for color temp

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stleusc, 2020-01-09:
OK, here is what I found...
I stumbled across BASE 48 which controls light as expected.
Other issue was light turns off at 50%, LEDTABLE 0 fixed that.

closed 2020-01-09 by stleusc

s-hadinger, 2020-01-09:
There was a bug that would drive brightness too low with module 48. Please retry with the latest development version and Ledtable 1.