arendst/Tasmota #7469: Support Shelly addon / sensor module

vergoossen, 2020-01-09:
**Have you looked for this feature in other issues and in the docs?**  
Yes, couldn't find anything.

**Is your feature request related to a problem? Please describe.**  
https://shop.shelly.cloud/temperature-sensor-addon-for-shelly-1-1pm-wifi-smart-home-automation#312
Tried to assign al gpio's to ds18b20 sensor, but it doesn't work. I think this module needs an additonal driver.

**Describe the solution you'd like**  
Should be nice if Tasmota can support this module, it's great to attach inwall / in switchbox sensors.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

llevering, 2020-01-16:
As the module 'clicks' onto the standard available GPIO pins of the Shelly 1, I would assume the module communicates with serial communication over the RX/TX pins. As no useful other GPIO pin is available for the click on module. Which sadly makes it a bit more work to implement I guess. But would be really nice if it can get support in Tasmota.

Gamester17, 2020-01-16:
+1 however I would like to connect a Si7021 or AM2301 humidity sensor instead as requested in #7530

Gamester17, 2020-01-16:
@ascillato2 as I understand it that while the GPIO ports on the Shelly 1PM is connected to mains AC power, that new GPIO-adapter that Shelly sells as part of its Temperature Sensor Addon is isolated so any sensors connected via that GPIO-adapter on the Shelly 1PM would not be directly connected to mains AC power. 

https://shop.shelly.cloud/temperature-sensor-addon-for-shelly-1-1pm-wifi-smart-home-automation#312

ascillato2, 2020-01-16:
@Gamester17 The Shelly ADD-ON is isolating the sensor from mains. That is why you can connect shelly supported sensors to a shelly device through this ADD-ON. But only through this ADD-ON. And so far, only using stock firmware due this ADD-ON is not supported by Tasmota so far.

You are saying:

> Sonoff Pow or Sonoff Pow R2 with a wired Si7021 or AM2301 sensors would also work be a solution my purpose of....

An that is very dangerous. You must read the docs. NEVER CONNECT SENSORS TO ANY POWER MEASURING DEVICE like a SONOFF POW.. You can electrocute yourself. Those devices are NOT ISOLATED and your sensor will have MAINS AC. DO NOT PUT YOURSELF, YOUR FAMILY AND HOME AT RISK.

arendst, 2020-01-16:
Let someone buy it, crack it open and report pin lay-out.

vergoossen, 2020-01-16:
I have some at home.
Can crack it tonight if it helps :)

vergoossen, 2020-01-16:
[image]
[image]
[image]

There you go :)

vergoossen, 2020-01-16:
[image]

Texts are very hard to read... Let me know if you need to know more / if I can test something.

Gamester17, 2020-01-17:
@vergoossen Nice! Is it possible for you to get some better pictures?

vergoossen, 2020-01-17:
What kind of pictures do you want, what do you want to see?

Gamester17, 2020-01-17:
I guess that I just mean sharper closeups so that you can clearly see all paths & chip text + orientations

vergoossen, 2020-01-17:
[image]
[image]
[image]
[image]

Are these pictures better?

arendst, 2020-01-17:
Way better. No we know the 8-pin chip is a TexasInstruments ISO7221 Dual channel Digital Isolator.

GPIO03 is connected to pin 2 and would become an input.
GPIO00 is connected to pin 3 and would become an output.

Both pins 6 and 7 are connected together and fed through to the second pin on the connector where the DS18B20 is connected.

So it seems in order to talk to the DS18B20 we will need a driver sending data to the DS18B20 via GPIO00 and receiving data from the DS18B20 via GPIO03.

vergoossen, 2020-01-17:
Hmm, that sounds like a lot of work?
And do you need to write a seperate driver for each sensor? For example a DHT22 needs another new driver? Or can it be a "universal" driver which can handle all already supported sensors?

arendst, 2020-01-17:
At first it would be a separate driver for any sensor where it is even questionable if it would work for other sensors considering the two pin to one wire construction.

arendst, 2020-01-17:
Looks to me like a nice hobby project for someone who loves Shelly and sensors ;-)

vergoossen, 2020-01-17:
Haha yes, unfortunately I can't write C, only "web languages". 
Otherwise I would love to help :).

vergoossen, 2020-01-17:
Maybe it's not that simple, but:
When you configure template gpio3 to ds18x20, that should be step 1?
And then tasmota should send data to gpio0? Can I test that with a serialsend in console? Or is it nog that simple?

arendst, 2020-01-17:
No it's not that simple. The single wire protocol used by DS18B20 is timing dependant.

What you could try is 
- make a copy of file xsns_05_ds18x20.ino and rename it as xsns_05_ds18x20_shelly.ino
- change the top define #ifdef USE_DS18x20 into USE_DS18x20SHELLY
- disable define USE_DS18x20 in user_config.h (or user_config_override.h)
- add define USE_DS18x20SHELLY in above file and recompile to check your copy is OK
- then change reads from input port to GPIO03 and writes to output port GPIO00 (that's tricky if you don't know how to address a port/GPIO)

Seeing what I wrote there is much more to it.

I think I make a basic writeup for this sensor for Shelly and let you test.

Hold on.

altelch, 2020-01-17:
The TI chip does galvanic separation for IO on GPIO0 and GPIO3. I don't really understand the pinout of the second connector. There are only 3 Lines connected. Pin 4-6 seems to be floating as I can'd detect a gnd-layer on the backside of the board or any vias connecting this pins to anywhere. The we are left to 1-3.  1 and to are the GPIO-Pins, 3 is GND or VCC. DS18x20 is a One-Wire protocol, we need only one gpio to talk to the device, so the second GPIO is probably used to ground (?) or power the vcc line of the connected sensors. I'd guess Pin 1 is vcc, Pin 2 is data and pin3 is GND.

arendst, 2020-01-17:
Give the latest commit a try. You will have to compile yourself and make the changes I mentionend above in either ``my_user_config.h`` or ``user_config_override.h``

vergoossen, 2020-01-17:
Do I still need to configure something in web interface?
I selected shely1 in modules and in console i get a lot of this messages:
18:04:48 CMD: nknown"}
18:04:48 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
18:04:48 CMD: D: a/RESULT = {"Command":"Unknown"}
18:04:48 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}
18:04:48 CMD: :04:47 CMD: }
18:04:48 CMD: t/tasmota/RESULT = {"Command":"Unknown"}
18:04:48 RSL: stat/tasmota/RESULT = {"Command":"Unknown"}

arendst, 2020-01-17:
Turn seriallog off

vergoossen, 2020-01-17:
Okay sorry its gone now.
But, i selected shelly1 as module, but nothing comes in the webinterface/console.
I also tried to configure template gpio3 to ds18x20 but nothing happens..

arendst, 2020-01-17:
Other than Shelly1 or Shelly1PM there is nothing to configure. The GPIOs 00 and 03 are hardcoded. 

If you see the option for DS18x20 in your gui than you did not disable the old driver (and possibly activate the new driver). The option should not be there.

vergoossen, 2020-01-17:
Okay then I did something wrong.
Ill check tonight.

vergoossen, 2020-01-17:
Now there's no more option DS18x20 in the gui, but i still dont see anything in gui/console about the sensor.

arendst, 2020-01-18:
I ordered one. Once arrived I will dive into.

vergoossen, 2020-01-18:
Nice! I could also send one to you.

Gamester17, 2020-01-20:
In somewhat relevant news I read on Shelly German support-forum that Dimitar Dimitrov (CEO of Allterco Robotics which makes Shelly) recently posted on Shelly's Facebook page with a statement that their next official firmware for Shelly 1 and Shelly 1PM will add support DHT22 Temperature and Humidity Sensor

https://www.shelly-support.eu/forum/index.php?thread/1547-temperatur-luftfeuchtigkeitsensor-dht22/

So my follow-up question to this is if the official Shelly firmware code is really open source not?

meingraham, 2020-01-20:
Shelly firmware is not open source.

ascillato, 2020-01-20:
@Gamester17 Shelly is not open source, what they said is that their API to control shelly devices in a local network is open and fully explained in their documentation. So, there is your confusion. Their firmware is closed source but how to control it externally is open and documented.

Gamester17, 2020-01-20:
@ascillato honest mistake given that they title it as "open source" and not "open API" on their site:

https://shelly.cloud/shelly1-open-source/

All resellers of Shelly that I can find also re-hash that "open source" title when selling the Shelly 1.

So a just a case false advertisement from Allterco Robotics when marketing their Shelly products?

ascillato, 2020-01-20:
> So a just a case false advertisement from Allterco Robotics when marketing their Shelly products?

I don't think so. May be just a bad translation to english or bad choice of words. As soon as you enter in their site, it is explained as my previous comment.

vergoossen, 2020-01-23:
Don't know if it makes sense.
But i thought the addon has an internal pullup resistor.

vergoossen, 2020-02-03:
@arendst did you already receive the addon? :)

arendst, 2020-02-04:
Got it. Need time to dive

arendst, 2020-02-05:
To start you'll need to undefine GPIO00 as LedI in the standard template as it is now being used as an input for the sensor. 

So use this template when the Add-On is connected.
```
{"NAME":"Shelly 1PM U2","GPIO":[0,0,0,0,82,134,0,0,0,0,0,21,0],"FLAG":2,"BASE":18}
```
It doesn't show (additional) temperatures as the protocol still needs to be worked out...

arendst, 2020-02-05:
This is what we're looking for:

[image]

Now finding a way to integrate...

arendst, 2020-02-05:
Latest dev commit and version 8.1.0.6 supports the shelly add-on for DS18x20.

To use it you'll need an updated Shelly 1PM template. This is the correct one:
```
{"NAME":"Shelly 1PM U2","GPIO":[255,255,0,255,82,134,0,0,0,0,0,21,0],"FLAG":2,"BASE":18}
```
Once activated go to the Module Configuration screen and enter the DS18x20 parameters like 

[image]

With the latest version 8.1.0.6 now Shelly 1 also allows the above configuration.

After the restart the additional sensor(s) should be displayed like

[image]

vergoossen, 2020-02-05:
Is the template the same for shelly1?

ascillato, 2020-02-05:
> Is the template the same for shelly1?

No, they are different devices with different capabilities and different connections.

ascillato, 2020-02-05:
TEMPLATES:

```
* SHELLY 1PM
OLD: {"NAME":"Shelly 1PM","GPIO":[56,0,0,0,82,134,0,0,0,0,0,21,0],"FLAG":2,"BASE":18}
NEW: {"NAME":"Shelly 1PM U2","GPIO":[255,255,0,255,82,134,0,0,0,0,0,21,0],"FLAG":2,"BASE":18}

* SHELLY 1
OLD: {"NAME":"Shelly 1","GPIO":[0,0,0,0,21,82,0,0,0,0,0,0,0],"FLAG":0,"BASE":46}
NEW: {"NAME":"Shelly 1 U2","GPIO":[255,255,0,255,21,82,0,0,0,0,0,0,0],"FLAG":0,"BASE":46}
```

vergoossen, 2020-02-05:
Thnx will try asap.

spielkind1, 2020-02-05:
Hi i tried it with:
{"NAME":"Shelly 1 U2","GPIO":[255,255,0,255,21,82,0,0,0,0,0,0,0],"FLAG":0,"BASE":46}
but it dont work

ascillato, 2020-02-05:
@spielkind1

In old versions of Tasmota is not going to work. This has been added today to the codebase. See the date of the comments. Latest precompiled bins are at http://thehackbox.org/tasmota as explained in the docs and in the readme

spielkind1, 2020-02-05:
i found it, but still wont work :(
[image]

ascillato, 2020-02-05:
Please, read Theo's comment on how configure the sensor.

spielkind1, 2020-02-05:
where can i find this comment?

ascillato, 2020-02-05:
Please scroll back a little on this issue

spielkind1, 2020-02-05:
i know what you mean, now i need to restore. sh.... tomorrow ;)

ascillato2, 2020-02-05:
@vergoossen 

Please, tell us if this template works for your SHELLY 1 with SENSOR ADDON with the DS18X20 Temp sensor under latest Tasmota (8.1.0.6). Thanks.

arendst, 2020-02-05:
The updated Shelly 1 template is also embedded in the firmware so an external template is not really needed. The ds18x20 gpio module config is still needed.

arendst, 2020-02-06:
And another teaser, connecting an AM2301 to the Shelly Add-On:

[image]

Working on integration...

arendst, 2020-02-06:
To use a DHT family sensor use the latest commit and the following module config:

[image]

Note: for any of the three available DHT family options you always select ``DHT11o`` for GPIO00.

Gamester17, 2020-02-06:
@arendst Very cool! Do you also happen to own a Si7021 humidity and temperature sensor?

Si7021 sensors are latest sold by Itead for their Sonoff TH series so can be bought nicely packaged

https://www.itead.cc/wiki/Sonoff_Sensor_Si7021

https://sonoff.tech/product/accessories/si7021

Sonoff page incorrectly says it only supports -10 Celcius but it does support -40 same as AM2301

arendst, 2020-02-06:
I don't but as the current driver always supported the SI7021 chances are they will also work on the Shelly Add-On

vergoossen, 2020-02-06:
Addon is not working on 8.1.0.6 on shelly1. Template was also not in, i added myself and configured gpios, restarted device but no ds18x20 temperature shown.

arendst, 2020-02-06:
Pls provide output of commands ``status 0``, ``template`` and ``gpio``.

vergoossen, 2020-02-06:
17:16:36 CMD: backlog status 0;template;gpio
17:16:36 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:16:36 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power on","Uptime":"0T00:00:33","StartupUTC":"2020-02-06T16:16:03","Sleep":50,"CfgHolder":4617,"BootCount":7,"SaveCount":17,"SaveAddress":"FB000"}}
17:16:36 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0.6(tasmota)","BuildDateTime":"2020-02-06T15:55:13","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"337/699"}}
17:16:36 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WATERPAS",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
17:16:36 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":569,"Free":432,"Heap":27,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
17:16:36 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-8022","IPAddress":"192.168.1.113","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"98:F4:AB:F3:9F:56","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:16:36 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F39F56","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
17:16:36 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Thu Feb 06 16:16:36 2020","Local":"Thu Feb 06 17:16:36 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:13","Sunset":"17:54"}}
17:16:36 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-02-06T17:16:36","Switch1":"OFF"}}
17:16:36 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-02-06T17:16:36","Uptime":"0T00:00:33","UptimeSec":33,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WATERPAS","BSSId":"18:E8:29:E7:90:26","Channel":1,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:06"}}}
17:16:36 RSL: stat/tasmota/RESULT = {"NAME":"Shelly 1 U2","GPIO":[255,255,0,255,21,82,0,0,0,0,0,0,0],"FLAG":0,"BASE":46}
17:16:37 RSL: stat/tasmota/RESULT = {"GPIO0":{"192":"DS18x20o"},"GPIO1":{"0":"None"},"GPIO3":{"4":"DS18x20"}}

[41 more replies]

closed 2020-02-06 by arendst