arendst/Tasmota #7470: Disable button for all usages

dgrostoto, 2020-01-10:
Hello,
i have not found in the docs a way to simply disable all button actions (as if the button was a dummy button).
The way that button actions are controlled are rather complex and correspond to various specific need or situations.
May i have to modify the code to get this button fully disabled or is there a way in the http admin web server (in console maybe?)
Any suggestion appreciated.
A little about the context :
my device is a sonoff pow R2 and i'm using it as a reporting consumption device only. (I don't want the relay to be shutted off by the button since the button is freely accessible in my configuration, and i would like to not modify physically the sonoff)

**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Einstein67, 2020-01-10:
You can use: "poweronstate 4" = turn relay(s) ON and disable further relay control

Jason2866, 2020-01-10:
@dgrostoto  you can use template to make a specific version (based on module pow) without a button.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2020-01-10:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-01-10 by ascillato2

dgrostoto, 2020-01-10:
Thanks for your response. Concerning poweronstate=4 i'm not sure that it won't send any MQTT event in case of pressing so i think 'd rather modify source code in gpio template that do not "wire" the push button (presently at GPIO1 for sonoff_pow_r2 in template source code). 

Since chat is the better way to ask for support questions, let me contribute here as an issue :
I think there is a missing option to simply deactivate push button unconditionally (whenever relay is on or off or any other condition).
Actually, my personal use case is to control relay exclusively from MQTT and push button does not interfere. The only thing maybe the button could output may be a MQTT message in order to feedback user that button is deactivated (but with the GPIO disable, I may encounter difficulties...).

Another approach about this, is to detach action from triggers (push button does not control relay by default which was already mentioned in other issues about this project on github) and instead extend/generalize the rules section of tasmota, thus we could wire any action or set option to any trigger

dgrostoto, 2020-01-10:
Let me know, Which other issue is a duplicated of this one ?

ascillato, 2020-01-10:
Hi,

This has been explained in the wiki/docs.

You have several options to ignore a button by just configuring Tasmota. No need to edit the code. Options explained in docs:

poweronstate 4

or

change the gpio template of your device (using the Tasmota WebUI) and setting that gpio as none.

or

make a rule to define the behaviour you want for that button

As this is not a bug of Tasmota, and it is just a configuration/usage, it is better to be addressed in the chat. A chat (serie of questions and answers) is better and faster for these type of config/usage issues. Thanks.

dgrostoto, 2020-01-10:
OK thanks for your response.
For futur reference, only GPIO technique is the complete solution for the need.

ascillato, 2020-01-10:
> For futur reference, only GPIO technique is the complete solution for the need.

Sorry, but the other stated options also performs that. Explanation of each solution is on the Docs. I just wanted to clarify that for other users reading this.