arendst/Tasmota #7471: Shutter resulting position does not match target position

to-scho, 2020-01-10:
### PROBLEM DESCRIPTION
Shutter movement to any target position might result in a position not equal to target position. When this happens I typically observe a difference by 1%. In particular movements to 0% or 100% ending up at 1% or 99% break some of my home assistant automations. Full movements from 0% to 100% or vice versa are ok.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
with calibration:
09:45:00 CMD: backlog shutterrelay1 1; shuttersetclose1; shutterposition1 100
09:45:00 SRC: WebConsole from 192.168.42.2
09:45:00 CMD: Group 0, Index 1, Command "BACKLOG", Data "shutterrelay1 1; shuttersetclose1; shutterposition1 100"
09:45:00 SRC: Backlog
09:45:00 CMD: Group 0, Index 1, Command "SHUTTERRELAY", Data "1"
09:45:00 SHT: Shutter 0 Closefreq: 1113
09:45:00 SHT: Shutter 1 (Relay:1): Init. Pos: 42638 [99 %], Open Vel.: 100 Close Vel.: 111 , Max Way: 43000, Opentime 21.5 [s], Closetime 19.3 [s], CoeffCalc: c0: 214, c1 437, c2: 696, c3: 881, c4: 1000, binmask 3, is inverted 0, shuttermode 1,motordelay 7
09:45:00 MQT: stat/rollo-test/RESULT = {"ShutterRelay1":1}
09:45:00 SRC: Backlog
09:45:00 CMD: Group 0, Index 1, Command "SHUTTERSETCLOSE", Data ""
09:45:00 MQT: stat/rollo-test/RESULT = {"ShutterSetClose1":"Configuration reset"}
09:45:00 SRC: Backlog
09:45:00 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data "100"
09:45:00 SHT: Pos. in: payload 100 (3), payload 100, idx 1, src 9
09:45:00 SHT: lastsource 9:, real 0, target 43000, payload 100
09:45:00 SHT: Wait for Motorstop..
09:45:00 SRC: Shutter
09:45:00 SHT: Switched relay: 0 by Shutter
09:45:00 MQT: stat/rollo-test/RESULT = {"POWER1":"OFF"}
09:45:00 MQT: stat/rollo-test/POWER1 = OFF
09:45:00 SRC: Shutter
09:45:00 SHT: Switched relay: 0 by Shutter
09:45:00 MQT: stat/rollo-test/RESULT = {"POWER1":"OFF"}
09:45:00 MQT: stat/rollo-test/POWER1 = OFF
09:45:00 SRC: Shutter
09:45:00 SHT: Switched relay: 0 by Shutter
09:45:00 MQT: stat/rollo-test/RESULT = {"POWER2":"OFF"}
09:45:00 MQT: stat/rollo-test/POWER2 = OFF
09:45:00 SRC: Shutter
09:45:00 SHT: Switched relay: 1 by Shutter
09:45:00 MQT: stat/rollo-test/RESULT = {"POWER1":"ON"}
09:45:00 MQT: stat/rollo-test/POWER1 = ON
09:45:00 MQT: stat/rollo-test/RESULT = {"ShutterPosition1":100}
[... 170 lines cut ...]

### TO REPRODUCE
backlog shutteropenduration 21.5; shuttercloseduration 19.3; shuttercalibration 214 437 696 881 1000
 backlog shutterrelay1 1; shuttersetclose1; shutterposition1 100
...wait to open, result is 100, ok...
shutterposition 94
...wait for movement, result is 95!!!...
shutterposition 100
...wait for open, result is 99!!!...

Without shuttercalibration I did not manage to easily reproduce the 100%/99% but same happens:
backlog shutteropenduration 21.5; shuttercloseduration 19.3; shuttercalibration 300 500 700 900 1000
backlog shutterrelay1 1; shuttersetclose1; shutterposition1 7
...wait for movement, result is 6!!!...

### EXPECTED BEHAVIOUR
Shutter movement result is same as target position. When numerical accuracy does not allow this at least the 0% and 100% positions should be guaranteed. 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

thucar, 2020-01-11:
Just ran into this as well. I upgraded from 6.7.1 - in that version shutterposition was working correctly.

Rayzilt, 2020-01-12:
There are alot of fixes since version 8.1.0, especially for the shutterposition code.
See: https://github.com/arendst/Tasmota/issues/7351 https://github.com/arendst/Tasmota/issues/7331 https://github.com/arendst/Tasmota/issues/7305

Did you compile from the latest development branch? Otherwise you could try that or wait for a new version of Tasmota.

thucar, 2020-01-12:
Indeed, I can confirm that the Dev branch works correctly.

to-scho, 2020-01-12:
Most recent dev branch was used when getting above logs for this issue.

PR #7502 (2020-01-13): Fix target shutter position inaccuracy
Introduced numerical rounding when calculating between "%" and "time" position values

## Description:

**Related issue (if applicable):** fixes #7471 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [ ] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md# [... cut]
closed 2020-01-13 by arendst