arendst/Tasmota #7473: Wakeup ignores provided value

balli1187, 2020-01-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
First of all: Thanks for an amazing piece of Software! I'm using your firmware in a lot of devices and really love it.

However, I noticed unintended behaviour:
In one device, I’m only using `Dimmer 0` to set it `off` and `Dimmer 100` to set it `on`. I also want to use Wakeup command but it ignores the given value in `Wakeup 100`. As my stored dimmer is 0 cause It has been set off in the evening, nothing happens when calling Wakeup in the morning. If I set a dimmer value afterwards wakeup sequence starts to this new value Dimmer value.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): LD12 (LED-Driver)
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```10:41:24 MQT: SmartHome/192.168.17.227/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
10:41:24 MQT: SmartHome/192.168.17.227/stat/RESULT = {"Module":{"0":"Generic"}}
10:41:25 MQT: SmartHome/192.168.17.227/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"37":"PWM1"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```10:38:53 MQT: SmartHome/192.168.17.227/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["Sz_Wohnwand"],"Topic":"192.168.17.227","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:38:53 MQT: SmartHome/192.168.17.227/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power on","Uptime":"35T16:56:56","StartupUTC":"2019-12-05T16:41:57","Sleep":0,"CfgHolder":4617,"BootCount":14,"SaveCount":75,"SaveAddress":"F9000"}}
10:38:53 MQT: SmartHome/192.168.17.227/stat/STATUS2 = {"StatusFWR":{"Version":"7.1.0(tasmota)","BuildDateTime":"2019.11.29 15:44:57","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
10:38:53 MQT: SmartHome/192.168.17.227/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AS-privat",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
10:38:53 MQT: SmartHome/192.168.17.227/stat/STATUS4 = {"StatusMEM":{"ProgramSize":563,"Free":440,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440EF","FlashMode":3,"Features":["00000407","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
10:38:53 MQT: SmartHome/192.168.17.227/stat/STATUS5 = {"StatusNET":{"Hostname":"Sz_Wohnwand","IPAddress":"192.168.17.227","Gateway":"192.168.17.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.17.1","Mac":"5C:CF:7F:93:EB:E0","Webserver":2,"WifiConfig":4}}
10:38:53 MQT: SmartHome/192.168.17.227/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.17.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_93EBE0","MqttUser":"DVES_USER","MqttCount":56,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
10:38:53 MQT: SmartHome/192.168.17.227/stat/STATUS7 = {"StatusTIM":{"UTC":"Fri Jan 10 09:38:53 2020","Local":"Fri Jan 10 10:38:53 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:41","Sunset":"17:13"}}
10:38:53 MQT: SmartHome/192.168.17.227/stat/STATUS10 = {"StatusSNS":{"Time":"2020-01-10T10:38:53"}}
10:38:53 MQT: SmartHome/192.168.17.227/stat/STATUS11 = {"StatusSTS":{"Time":"2020-01-10T10:38:53","Uptime":"35T16:56:56","UptimeSec":3085016,"Heap":24,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":56,"POWER":"ON","Dimmer":100,"Fade":"ON","Speed":5,"LedTable":"ON","Wifi":{"AP":1,"SSId":"AS-privat","BSSId":"EC:08:6B:D9:66:B1","Channel":6,"RSSI":94,"LinkCount":3,"Downtime":"0T00:01:00"}}}
 
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- set `Dimmer 0`
- call `Wakeup` with any additional value

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
`Wakup 100` starts a slow fade from `Dimmer 0` to `Dimmer 100`

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-01-10:
Feature is working as designed and described in documentation. Wakeup command has NO payload.
`Start wake up sequence from OFF to stored Dimmer value
0..100 = Start wake up sequence from OFF to provided Dimmer value`

balli1187, 2020-01-10:
@Jason2866 
as there has been a feature request for this in [here](https://github.com/arendst/Tasmota/issues/867) i thought there ae two Options:
 - `Wakeup` --> Start wake up sequence from OFF to stored Dimmer value
 - `Wakeup 0...100` --> 0..100 = Start wake up sequence from OFF to provided Dimmer value

If it is not implemented, I want to ask for this feature.

Jason2866, 2020-01-10:
Ahh, you are right! @s-hadinger can you take a look at? Thx!

s-hadinger, 2020-01-10:
Yes. I will look at it

s-hadinger, 2020-01-10:
Fixed. It will be included in the pre-compiled development firmwares in less than 2 hours.

PR #7477 (2020-01-10): Fix ``WakeUp <x>`` ignores provided value (#7473)
## Description:

**Related issue (if applicable):** fixes #7473 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).
closed 2020-01-10 by arendst