arendst/Tasmota #7474: Tasmota crashes every 15 minutes (on multiple devices) after OTA from 6.5 -> 7.2 -> 8.1

jghek, 2020-01-10:
Hi All,

This my first issue here. I read that I am required to fill out a template. I cannot find this template. So I'll do my best to give you the information I think you need. I am a developer myself, so I've got some idea, but I don't know the specifics. Also, thank you for the effort and hard work you put in the community and development of Tasmota. I'm really enjoying it.

Yesterday I've upgraded my devices (mostly Shelly1, two Shelly2.5 and a Sonoff Dual R2) from Tasmota 6.5 to the recent 8.1. I did this by using OTA TasmoAdmin (first 7.2 and then 8.1) and some OTA manually (moving to 7.2 minimal, 7.2, 8.1 minimal and finally 8.1.

All devices now crash every 15 minutes or so. Here's to log from a Shelly 2.5
```
00:00:00 CFG: Loaded from flash at F8, Count 1038
00:00:00 I2C: ADE7953 found at 0x38
00:00:00 Project tasmota Schuur TL verlichting Version 8.1.0(tasmota)-2_6_1
00:00:00 WIF: Connecting to AP1 IoT in mode 11N as Shelly-Schuur...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on Shelly-Schuur with IP address 192.168.1.192
00:00:06 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/lights/groteschuur/verlichting/LWT = Online (retained)
00:00:06 MQT: cmnd/lights/groteschuur/verlichting/POWER = 
00:00:06 MQT: tele/lights/groteschuur/verlichting/INFO1 = {"Module":"Shelly 2.5","Version":"8.1.0(tasmota)","FallbackTopic":"cmnd/Shelly25-Schuur_fb/","GroupTopic":"cmnd/sonoffs/"}
00:00:06 MQT: tele/lights/groteschuur/verlichting/INFO2 = {"WebServerMode":"Admin","Hostname":"Shelly-Schuur","IPAddress":"192.168.1.192"}
00:00:06 MQT: tele/lights/groteschuur/verlichting/INFO3 = {"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4025bca7","4025bc3c","4025bbe3","4025ad44","4025ad6d","402587d8","401036e6","40254b3d","402594cf","40000f58","40258f42","40262693","40261f53","40240743","40000f49","40000f49","40000e19","40001878","40105865","4010586b","4010000d","40260e38","40260de9","40246d4a","402475e1","40246d4a","40246c80","402476e0","40246d4a","40283efc","40283efc"]}}
```

After that i see a couple of lines of STATE and SENSOR results that look perfectly normal. Here's the log of my SONOFF R2 Dual:

```
00:00:00 CFG: Loaded from flash at F9, Count 429
00:00:00 Project tasmota Buitenverlichting Version 8.1.0(tasmota)-2_6_1
00:00:00 WIF: Connecting to AP1 IoT in mode 11N as Sonoff2-Achterdeur...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on Sonoff2-Achterdeur with IP address 192.168.1.204
00:00:06 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/lights/achterdeur/LWT = Online (retained)
00:00:06 MQT: cmnd/lights/achterdeur/POWER = 
00:00:06 MQT: tele/lights/achterdeur/INFO1 = {"Module":"Generic","Version":"8.1.0(tasmota)","FallbackTopic":"cmnd/Sonoff2-Achterdeur_fb/","GroupTopic":"cmnd/sonoffs/"}
00:00:06 MQT: tele/lights/achterdeur/INFO2 = {"WebServerMode":"Admin","Hostname":"Sonoff2-Achterdeur","IPAddress":"192.168.1.204"}
00:00:06 MQT: tele/lights/achterdeur/INFO3 = {"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4025bca7","4025bc3c","4025bbe3","4025ad44","4025ad6d","402587d8","40250874","40254b3d","4024983d","402594cf","40258f42","40262693","40261f53","40267fef","40000f49","40000f49","40000e19","40001878","40105865","4010586b","4010000d","40260e38","40260de9","40246d4a","402475e1","40246d4a","40246c80","402476e0","40246d4a","40283efc","401009b4"]}}
00:00:06 MQT: stat/lights/achterdeur/RESULT = {"POWER1":"OFF"}
00:00:06 MQT: stat/lights/achterdeur/POWER1 = OFF
00:00:06 MQT: stat/lights/achterdeur/RESULT = {"POWER2":"OFF"}
00:00:06 MQT: stat/lights/achterdeur/POWER2 = OFF
00:00:10 MQT: tele/lights/achterdeur/STATE = {"Time":"1970-01-01T00:00:10","Uptime":"0T00:00:12","UptimeSec":12,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"IoT","BSSId":"56:D9:E7:91:5F:96","Channel":11,"RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:06"}}
00:00:10 MQT: tele/lights/achterdeur/SENSOR = {"Time":"1970-01-01T00:00:10","Switch1":"ON","Switch2":"OFF"}
00:05:10 MQT: tele/lights/achterdeur/STATE = {"Time":"1970-01-01T00:05:10","Uptime":"0T00:05:12","UptimeSec":312,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"IoT","BSSId":"56:D9:E7:91:5F:96","Channel":11,"RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:06"}}
00:05:10 MQT: tele/lights/achterdeur/SENSOR = {"Time":"1970-01-01T00:05:10","Switch1":"ON","Switch2":"OFF"}
```

It also restarts due to exception 29. Reading up on internet, I get that exception 29 is associated with bad flashing. So I did some manually, no change. Since all devices do this, I thought that this might be a software issue.

I have an old house and 'smartifying' is not a simple thing due to the electrical installation used. Some of the smart switches are hard to reach and doing a serial flash is not really a thing I want to do.

Since I have no retain flags in MQTT and set Tasmota to 'PowerOnState 3'. The only thing noticable from the outside is the 0.1 sec turn on whenever something is on during device reset. It itches and kills the WAF. So please help.

arendst, 2020-01-10:
Will follow up once my home power is restored.

arendst, 2020-01-10:
Try to provide output of command status 0

Also enable more logging with weblog 3 and provide output.

ascillato2, 2020-01-10:
@jghek Please, do not delete the template and complete it in order to have information so as to properly help you.

TEMPLATE:

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-01-10:
You have no valid time & date. I would fix this first. Maybe not related but Tasmota loose
a lot of its features

ascillato, 2020-01-10:
@Jason2866 Good eye!!!

@jghek Yes, Tasmota needs the correct time. There is a bug in the arduino core that if you don't have a correct slew rate on the NTP server request, it restarts. So your issue is indeed the time source. The bug in the arduino core was not solved so far (it is in the sntp.c file of the arduino core)

If your router or your ISP don't provide time or block the port udp123, you will need to set the time manually.

If you go to the console and type `time 1578667834` you should avoid those exception 29 restarts.

ascillato, 2020-01-10:
if you want to set the time to be as your NOW, you can go to https://www.epochconverter.com/ and copy the epoch number and paste it with `time EPOCH_NUMBER` in your tasmota console.

ascillato, 2020-01-10:
Other more permanent solution is to have a local NTP server (that can be your router if it support that, a RPI, or a RTS chip or a GPS chip attached to any of your Sonoff Dual). If you go to the GPS chip solution, then you need to point the NTP time source of all your devices to your Sonoff Dual. Command NTPSERVER on Tasmotas.

jghek, 2020-01-10:
Thanks guys for all your support! I will try this, fill out the template and let you know the result.

jghek, 2020-01-10:
I did test with different tools and those worked fine on the NTP Server of the Mikrotik. But not on my tasmota devices. I ended up setting op installing chrony (an NTP server hass addon) on my hass RPi. Changed the NTP Server option (42) in the DHCP of my Mikrotik to point to my HASS machine and everything works fine now. 

Would you still like me to provide additional information? Or can we close this?

Many thanks!

Jason2866, 2020-01-10:
Please close. Thx. Users with this issue can find it via search.

closed 2020-01-10 by jghek

ynuyasha, 2020-04-14:
Hi, i have multiple sonoff touch on tasmota 8.1 and i have this problem.
I set manually time, but after sometime time was resetted to 1970.
I don't know why ntp not working in tasmota, inside local network where sonoff are connected ntp is working.
```
ntpdate -q pool.ntp.org
server 162.159.200.1, stratum 3, offset 0.854860, delay 0.03285
server 37.247.53.178, stratum 2, offset 0.855553, delay 0.03596
server 31.14.131.188, stratum 3, offset 0.855112, delay 0.04202
server 162.159.200.123, stratum 3, offset 0.854785, delay 0.03316
14 Apr 17:41:19 ntpdate[408]: step time server 37.247.53.178 offset 0.855553 sec
```

this is console log

```
00:00:00 CFG: Loaded from flash at F5, Count 4661
00:00:00 Project tasmota SalaLight Version 8.1.0.10(b7c6a7b-tasmota)-2_6_1
00:00:00 WIF: Connected
00:00:00 HTP: Web server active on sonoff-sala-6280 with IP address myip
00:00:01 MQT: Attempting connection...
00:00:01 MQT: Connected
00:00:01 MQT: tele/sonoff-sala/LWT = Online (retained)
00:00:01 MQT: cmnd/sonoff-sala/POWER = 
00:00:01 MQT: tele/sonoff-sala/INFO1 = {"Module":"Sonoff Touch","Version":"8.1.0.10(b7c6a7b-tasmota)","FallbackTopic":"cmnd/DVES_97D888_fb/","GroupTopic":"cmnd/sonoffs/"}
00:00:01 MQT: tele/sonoff-sala/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-sala-6280","IPAddress":"myip"}
00:00:01 MQT: tele/sonoff-sala/INFO3 = {"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4025d397","4025d32c","4025d2d3","4025c434","4025c45d","40259ecc","40256231","4024b30d","4025abc3","4025a636","40263d8b","4026364b","402424a7","40000f49","40000f49","40000e19","40001878","40105865","4010586b","4010000d","4026252c","402624dd","40100ca0","401028dd","401024ac","40105747","40104922","40104d93","4010485b","40103720","4000050c"]}}
00:00:01 MQT: stat/sonoff-sala/RESULT = {"POWER1":"OFF"}
00:00:01 MQT: stat/sonoff-sala/POWER1 = OFF (retained)
00:00:01 MQT: stat/sonoff-sala/RESULT = {"POWER1":"OFF"}
00:00:01 MQT: stat/sonoff-sala/POWER1 = OFF (retained)
00:00:05 MQT: tele/sonoff-sala/STATE = {"Time":"1970-01-01T00:00:05","Uptime":"0T00:00:07","UptimeSec":7,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"myssd","BSSId":"mymac","Channel":6,"RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:01"}}
```

ynuyasha, 2020-04-14:
i found that if i use a static ip instead of dhcp ntp start to works

ascillato, 2020-04-14:
> i found that if i use a static ip instead of dhcp ntp start to works

Great, you found the issue! When using DHCP, the ntp servers that are used are the ones that provides your router (no matter if your router is giving incorrect ones), and when using fixed IP, the DHCP settings (including network NTP address) are ignored and the NTP used are the ones configured in Tasmota.
This is the SDK behavior of the arduino core in which Tasmota is built. If you want to use DHCP again, you need to configure in your router a correct NTP server or point to a correct NTP address, or just disable NTP network address.

ynuyasha, 2020-04-14:
for everyone that have a FritzOS i have resolved dhcp and ntp issue following this guide https://en.avm.de/service/fritzbox/fritzbox-7590/knowledge-base/publication/show/336_Configuring-time-synchronization-NTP-for-FRITZ-Box-and-network-devices/