arendst/Tasmota #7479: Novostella Smart Flood Light Color Temperature inverted

brianegge, 2020-01-10:
### PROBLEM DESCRIPTION
Color temperature is inverted on "Novostella Smart Flood Light"

According to Amazon "Buy one Novostella Smart Flood Light, get three lights in fact to cover most of your lighting needs. RGBCW = RGB + Warm White + Daylight White. Tunable white from 2700K to 6500K, 2000lm. "

CT 163 is very warm, CT 500 is very cool. I tried inverting PWM5 but that broke the color temp altogether.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Novostella Smart Flood Light (Magic Home)
- [ ] Tasmota binary firmware version number used: 7.0.3
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: OTA convert
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
23:53:55 RSL: stat/tasmota/RESULT = {"NAME":"Generic","GPIO":[0,0,0,0,37,41,0,0,38,40,39,0,0],"FLAG":0,"BASE":18}
23:53:55 RSL: stat/tasmota/RESULT = {"Module":{"0":"Generic"}}
23:53:55 RSL: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"37":"PWM1"},"GPIO5":{"41":"PWM5"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"38":"PWM2"},"GPIO13":{"40":"PWM4"},"GPIO14":{"39":"PWM3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
23:54:36 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:54:36 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:15:25","StartupUTC":"2020-01-10T22:39:11","Sleep":50,"CfgHolder":4617,"BootCount":12,"SaveCount":1113,"SaveAddress":"FB000"}}
23:54:36 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"7.0.0.3(tasmota)","BuildDateTime":"2019-11-06T23:07:19","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
23:54:36 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["fiji",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
23:54:36 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":446,"Free":556,"Heap":30,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":0,"Features":["00000809","0F400082","04048000","00000005","000000C0","00000080"],"Drivers":"1,2,3,4,9,10,12,16","Sensors":"1,2,3"}}
23:54:36 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"rgb-flood-4","IPAddress":"192.168.254.92","Gateway":"192.168.254.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.254.254","Mac":"2C:F4:32:82:A9:CF","Webserver":2,"WifiConfig":2}}
23:54:36 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_82A9CF","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
23:54:36 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Fri Jan 10 22:54:36 2020","Local":"Fri Jan 10 23:54:36 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:41","Sunset":"17:13"}}
23:54:36 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-01-10T23:54:36"}}
23:54:36 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-01-10T23:54:36","Uptime":"0T00:15:25","UptimeSec":925,"Heap":30,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":310,"MqttCount":0,"POWER":"ON","Dimmer":100,"Color":"000000FF00","HSBColor":"0,0,0","Channel":[0,0,0,100,0],"CT":153,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"fiji","BSSId":"88:1F:A1:42:3D:00","Channel":11,"RSSI":76,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
CT 153 create warm white, CT 500 is cool white

### EXPECTED BEHAVIOUR
CT 153 create cool white, CT 500 is warm white

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

brianegge, 2020-01-13:
I had PWM4 and PWM5 reversed. After mapping GPIO5:PWN4(40) and GPIO13:PWN5(41) it works fine.

closed 2020-01-13 by brianegge

2ZZ, 2020-03-01:
@brianegge can you share your full mapping?

brianegge, 2020-03-05:
Detailed write up here:

https://www.theeggeadventure.com/2019/12/connecting-novostella-20w-smart-led-flood-lights-to-smartthings/

-Brian