arendst/Tasmota #7482: Send retained message to cmnd topic when a switch is pressed or command is given from the webgui.

Adityakush, 2020-01-11:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I have a centralized system in which the MQTT broker is used to store the state of the relays. 
I have used the settings 
setoption0 0
powerretain 1
The Home Automation Software publishes commands with retained flag to the cmnd topic. Tasmota gets this state from the broker upon a restart. Until this part it works fine but if a physical switch is pressed on Tasmota side or a command is given from the webgui/timers/rules etc then it never publishes the new state to the cmnd topic and hence upon a restart Tasmota does not get this updated state as the cmnd topic still holds the last state sent by the Home Automation Software. This is not intended in a system using retained settings as all sources of control should publish the new state to the cmnd topic only then it will be useful and otherwise it is quite problematic.
Right now Tasmota only publishes retained messages to the stat topic but that that only does half of the task. This even produces clashes between the cmnd and stat topic as they can have different values if the last command is given from the webgui or a switch is pressed. 

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
If a switch is pressed or webgui/rules/timers change the state of relays, then the new state should be published to the cmnd topic with a retained flag by Tasmota so that this new state can be used by itself upon a restart.
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
nothing
**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-01-11:
This request of @Adityakush comes from a Discord Chatting. Instead of a decentralized system (as Tasmota is aimed), a centralized approach is needed by @Adityakush 

So, as the MQTT broker is the center of @Adityakush system, the response needed for every relay change is a retained message in `cmnd/topic/POWER` instead of the default `stat/topic/POWER`

In this approach, Tasmota is configured to not save its state internally and to rely only on the mqtt broker. 

As a workaround for now, the following rule produces the behaviour @Adityakush wants:
```lua
rule1 1
rule1 on power1#state=1 do publish2 cmnd/%topic%/power1 ON endon power1#state=0 do publish2 cmnd/%topic%/power OFF endon
 ```

@arendst 
The remaining question is if we can add a precompiler Define key to change stat to cmnd only for relays for this case of centralized systems.

_( Anyway, for other users, it is still recommended to use Tasmota's default approach - a decentralized system - if you don't need any advanced or custom approach )_

Adityakush, 2020-01-11:
Thanks @ascillato2. Just one thing I would like to suggest is to publish update to the cmnd topic in addition to the stat topic and not replace it.

ascillato2, 2020-01-11:
Ok, even easier request.

Just to add a precompiler KEY to publish by mqtt `cmnd/topic/POWER` along with the stat of the relays.

arendst, 2020-01-12:
Pls execute command ``prefix2 cmnd`` and see if it fixes your main issue.

Adityakush, 2020-01-12:
Hi @arendst.
I tried `prefix2 cmnd`. 
As soon as tasmota starts, it publishes to the cmnd topic and overwrites the cmnd topic instead of getting the last command from the cmnd topic. So it does not help.

arendst, 2020-01-12:
And use setoption4?

Adityakush, 2020-01-12:
Tried with `setoption4 1`. But no effect.

Adityakush, 2020-01-13:
Hi !
Just a small correction. @ascillato2  the rule mentioned above does not work as is intended.
 However the request still remains the same 

> to add a precompiler KEY to publish by mqtt cmnd/topic/POWER along with the stat of the relays

ascillato2, 2020-01-14:
Please, copy and paste again the rule. There was a syntax error. I have edited the comment. Thanks.

Adityakush, 2020-01-14:
Yes the rule works now. Thanks for that.

ascillato2, 2020-01-22:
Closing this issue for now as it is not going to be implemented due this centralized approach is different than the aimed with Tasmota (decentralized) and adding an on the fly setoption for changing this behavior may confuse some new users and will raise many issues for having mixed control type.

For now, as it is something advanced, let's leave it with the rules workaround. 

Thanks for sharing your ideas.

closed 2020-01-22 by ascillato2

Adityakush, 2020-01-22:
Hi @ascillato2.
I would like to mention though that there is a small problem with the rules way.
As soon as I press a switch, it changes the state and sends a retained message to the stat topic and the rule also gets implemented and sends a retained message to the cmnd topic. As there is a update in the cmnd topic, tasmota again reads this cmnd topic and for a second time changes the state to ON(though it was already ON) and sends a retained message to the stat topic.
It could have been better if there was a solution for that too. But I guess that requires some work in the code.

ascillato, 2020-01-22:
There is no problem if Tasmota receives a power ON while it is already at ON. You are not going to have any issue with that.

Adityakush, 2020-01-22:
Okay. I can live with that. Thanks a lot @ascillato2.