arendst/Tasmota #7485: Support for Wyze Bulb

Quantum-cross, 2020-01-11:
Hi all, I've decided to crack open a Wyze bulb and try to flash Tasmota onto the ESP-WROOM-02. It wasn't too difficult to get inside the potting material, the relevant connections for flashing are broken out pretty close to the top.

I have tried flashing the pre-built binary via esptool.py and I've also tried building and flashing using arduino IDE.

It seems that when I reboot the device after a flash, I just get a mess of stack traces and fatal exceptions over the serial port.

I'll leave my flashing setup hooked up for now. If anyone has any ideas feel free to post them here. Thanks!

closed 2020-01-11 by close-issue-app[bot]

meingraham, 2020-01-11:
Unless the ESP-WROOM-02 is an ESP8266 under the covers, Tasmota will not run on it

Quantum-cross, 2020-01-12:
the WROOM-02 is just Yet Another ESP8266 module: https://www.espressif.com/en/products/hardware/esp-wroom-02/resources