arendst/Tasmota #7486: Support for Wyze bulb (trouble flashing)

Quantum-cross, 2020-01-11:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
Yes, having trouble flashing software to Wyze bulb. Breaking new ground, not surprised.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Help flashing the software to Wyze bulb

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Not flashing the software to Wyze bulb :p

**Additional context**  
_Add any other context or screenshots about the feature request here._
Hi all, I've decided to crack open a Wyze bulb and try to flash Tasmota onto the ESP-WROOM-02. It wasn't too difficult to get inside the potting material, the relevant connections for flashing are broken out pretty close to the top.

I have tried flashing the pre-built binary via esptool.py and I've also tried building and flashing using arduino IDE.

It seems that when I reboot the device after a flash, I just get a mess of stack traces and fatal exceptions over the serial port.

I'll leave my flashing setup hooked up for now. If anyone has any ideas feel free to post them here. Thanks!

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-01-11:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-01-11 by ascillato2

Quantum-cross, 2020-01-11:
Thanks, I had no idea what I was doing... But I figured it out now and got it working. I've submitted a template to the template repository.

jakre1, 2020-06-05:
Would you mind sharing what you did to put the device into boot/flash mode?   I've got the device open and outputting to terminal.  Can it be done with the native headers on the board, or do I need to rip more of the potting out to get directly at the ESP-WROOM-02 chip pins.

Quantum-cross, 2020-06-05:
Just pull WRE to ground and it enables flashing IIRC.

dpersuhn, 2020-10-26:
@jakre1  Just to follow up on your endeavor, I sacrificed a bulb to determine the header pin mapping by desoldering the ESP-WROOM-02 board entirely and mapping out the pins on the board.  From this, I can verify the following mapping from the header:

GND -> Pins 6(IO15),9(GND),13(GND), and 18(GND)
LOG -> Pin 7 (IO2 / UART1, TX - DEBUG)
WRT -> Pin 8 (IO0 / UART Download - Pull Down / Flash Boot - Float - Pull Up)
RX -> Pin 11 (UART0 RXD)
TX -> Pin 12 (UART0 TXD)
3v3 -> Pin 1 (3v3)

Given this, I wanted to ask you about your experience flashing.  It would appear that the jumper to IO15 isn't necessary, only pulling WRE low should be needed.  I wanted to ask if you had tried it without the jumper and if you would be willing to provide any other information about your connection and what you may have seen on the console.  I've had no success getting console output from the bulb on either UART's TX or being able to flash it with WRE pulled low.  I'm not sure if that is due to some change Wyze made in newer firmware or not.