arendst/Tasmota #7488: Counter increments two

robin13, 2020-01-11:
### PROBLEM DESCRIPTION

I have been trying to get a tcrt5000 to work as a reader for the rotating disk on the household power meter.  This has returns a high signal while most of the disk is rotating, and every time the red marked spot passes under the sensor it goes low.
Problem: for every pulse that the sensor sends, I get the counter incrementing by two.
Setup:
D4 / GPIO2 is configured as a `Counter1`
Counter relevant commands set:
```
Backlog CounterDebounce 100; CounterType 0; Counter1 0
```
I was wondering if this might be an issue with the TCRT5000 module, so I set up a simple push-button on a breadboard, and I can get the same effect with this as well.
BUT
With the manual button I can click significantly faster than the dial on the power meter passes (but less than the 100ms debounce time) - if I click very fast, I get the correct single increment of the counter, but if I click slowly, holding the button for (I guess) more than half a second I get two counts.
It feels to me as if counter works, as long as the pulses are very short, but not if they are a bit slower...

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x} Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [x] Tasmota binary firmware version number used: 8.1.0.3
  - [x] Pre-compiled
  - [x] Self-compiled (tried both)
    - [x] IDE / Compiler used: `pio run`
- [x] Flashing tools used: `esptool.py`
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
22:14:51 MQT: stat/tasmota-test-01/RESULT = {"NAME":"D1 Mini","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
22:14:51 MQT: stat/tasmota-test-01/RESULT = {"Module":{"0":"D1 Mini"}}
22:14:52 MQT: stat/tasmota-test-01/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"42":"Counter1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"37":"PWM1"},"GPIO16":{"0":"None"}}
```

- [x] Provide the output of this command: `Status 0`:
```
22:15:22 MQT: stat/tasmota-test-01/STATUS = {"Status":{"Module":0,"FriendlyName":["tasmota-test-01"],"Topic":"tasmota-test-01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:15:22 MQT: stat/tasmota-test-01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:48","StartupUTC":"2020-01-11T21:14:34","Sleep":50,"CfgHolder":4617,"BootCount":14,"SaveCount":163,"SaveAddress":"F9000"}}
22:15:22 MQT: stat/tasmota-test-01/STATUS2 = {"StatusFWR":{"Version":"8.1.0.3(tasmota)","BuildDateTime":"2020-01-10T16:42:41","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"360/699"}}
22:15:22 MQT: stat/tasmota-test-01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["cundr_iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
22:15:22 MQT: stat/tasmota-test-01/STATUS4 = {"StatusMEM":{"ProgramSize":568,"Free":432,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A1","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
22:15:23 MQT: stat/tasmota-test-01/STATUS5 = {"StatusNET":{"Hostname":"tasmota-test-01-5541","IPAddress":"10.10.10.152","Gateway":"10.10.10.1","Subnetmask":"255.255.255.0","DNSServer":"10.10.10.1","Mac":"A4:CF:12:D5:D5:A5","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:15:23 MQT: stat/tasmota-test-01/STATUS6 = {"StatusMQT":{"MqttHost":"hassio.local","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D5D5A5","MqttUser":"hassio","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:15:23 MQT: stat/tasmota-test-01/STATUS7 = {"StatusTIM":{"UTC":"Sat Jan 11 21:15:23 2020","Local":"Sat Jan 11 22:15:23 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:41","Sunset":"17:14"}}
22:15:23 MQT: stat/tasmota-test-01/STATUS10 = {"StatusSNS":{"Time":"2020-01-11T22:15:23","COUNTER":{"C1":20}}}
22:15:23 MQT: stat/tasmota-test-01/STATUS11 = {"StatusSTS":{"Time":"2020-01-11T22:15:23","Uptime":"0T00:00:49","UptimeSec":49,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":31,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"cundr_iot","BSSId":"02:9F:C2:F7:C1:63","Channel":1,"RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```

### TO REPRODUCE
As described above

### EXPECTED BEHAVIOUR
I would expect the counter to increment only once per pulse regardless of the length of the pulse

to-scho, 2020-01-15:
Seems that counter debounce time for less than 200ms does not work reliably. Have you tried larger values.

ascillato2, 2020-01-22:
Hi,

Any news on this? Have you tried a higher counterdebounce? Have you tried the gpio as counter1n?

robin13, 2020-01-22:
I have tried setting `CounterDebounce` to 200, but no change.
I have also tried `Counter1n` but this also has no effect.

to-scho, 2020-01-27:
I had same 2x increment with on board push-button of a sonoff basic. At my place it was gone with CounterDebounce 200. 

If you are willing to debug further:

-You need to build your own binary after altering some code
-In file tasmota/xsns_01_counter.ino add the following single line after line 55 of current development branch:  AddLog_P2(LOG_LEVEL_INFO, PSTR("COUNTER: Update of counter %d @ %d us, last update @ %d us, debounce time is %d us"), index, time, Counter.timer[index], debounce_time);
-Get log from console when 2x inc. happens
-Log should state system timings in us when counter change. Any counter increment should only happen when debounce time in log message is greater than CounterDebounce*1000.
-From that I found that CounterDebounce 200 (in ms) solve my issue.

Here is how updated tasmota/xsns_01_counter.ino looks like around line 55:
void CounterUpdate(uint8_t index)
{
  uint32_t time = micros();
  uint32_t debounce_time = time - Counter.timer[index];
  if (debounce_time > Settings.pulse_counter_debounce * 1000) {
    AddLog_P2(LOG_LEVEL_INFO, PSTR("COUNTER: Update of counter %d @ %d us, last update @ %d us, debounce time is %d us"), index, time, Counter.timer[index], debounce_time);
    Counter.timer[index] = time;
    if (bitRead(Settings.pulse_counter_type, index)) {
...

robin13, 2020-02-12:
Thank you for the debugging help @to-scho 

I tried what you said, and this is the log output resulting:

```
10:54:08 CMD: Counter1 0
10:54:08 MQT: stat/tasmota-test-01/RESULT = {"Counter1":0}
10:54:14 COUNTER: Update of counter 0 @ 38263316 us, last update @ 0 us, debounce time is 38263316 us
10:54:16 COUNTER: Update of counter 0 @ 39895150 us, last update @ 38263316 us, debounce time is 1631834 us
10:54:53 COUNTER: Update of counter 0 @ 76808967 us, last update @ 39895150 us, debounce time is 36913817 us
10:54:53 COUNTER: Update of counter 0 @ 77203754 us, last update @ 76808967 us, debounce time is 394787 us
10:55:16 CMD: CounterDebounce
10:55:16 MQT: stat/tasmota-test-01/RESULT = {"CounterDebounce":100}
10:56:44 COUNTER: Update of counter 0 @ 187480303 us, last update @ 77203754 us, debounce time is 110276549 us
10:56:45 COUNTER: Update of counter 0 @ 189101788 us, last update @ 187480303 us, debounce time is 1621485 us
10:57:41 CMD: CounterDebounce 200
10:57:41 MQT: stat/tasmota-test-01/RESULT = {"CounterDebounce":200}
10:57:50 COUNTER: Update of counter 0 @ 253368739 us, last update @ 189101788 us, debounce time is 64266951 us
10:57:50 COUNTER: Update of counter 0 @ 254286604 us, last update @ 253368739 us, debounce time is 917865 us
10:58:09 CMD: Counter1
10:58:09 MQT: stat/tasmota-test-01/RESULT = {"Counter1":8}
```

As you can see, for each passing of the stripe there are two counts.  I tried with `CounterDebounce` set to 100 and 200 with no difference.

robin13, 2020-02-12:
I've done some more investigation - made these changes to `xsns_01_counter.ino`:
```
void CounterUpdate(uint8_t index)
{
    uint32_t time = micros();
    uint32_t debounce_time = time - Counter.timer[index];
    if (debounce_time > Settings.pulse_counter_debounce * 1000) {
        AddLog_P2(LOG_LEVEL_INFO, PSTR("COUNTER: Update of counter %d @ %d us, last update @ %d us, debounce time is %d us"), index + 1, time, Counter.timer[index], debounce_time);
        Counter.timer[index] = time;
        if (bitRead(Settings.pulse_counter_type, index)) {
            RtcSettings.pulse_counter[index] = debounce_time;
        } else {
            RtcSettings.pulse_counter[index]++;
        }
    }
    AddLog_P2(LOG_LEVEL_INFO, PSTR("COUNTER %d @ %d us, : %d"), index + 1, time, RtcSettings.pulse_counter[index] );
}
```

And from this I'm seeing that there are a _lot_ of triggers to `CounterUpdate`:

```
11:48:32 COUNTER 1 @ 994830147 us, : 1
11:48:32 COUNTER 1 @ 994834169 us, : 1
11:48:32 COUNTER: Update of counter 1 @ 995122268 us, last update @ 994815314 us, debounce time is 306954 us
11:48:32 COUNTER 1 @ 995122268 us, : 2
11:48:32 COUNTER 1 @ 995126952 us, : 2
11:48:32 COUNTER 1 @ 995130062 us, : 2
11:48:32 COUNTER 1 @ 995140078 us, : 2
11:48:36 COUNTER: Update of counter 1 @ 998260029 us, last update @ 995122268 us, debounce time is 3137761 us
11:48:36 COUNTER 1 @ 998260029 us, : 3
11:48:36 COUNTER 1 @ 998270039 us, : 3
11:48:36 COUNTER 1 @ 998280400 us, : 3
11:48:36 COUNTER: Update of counter 1 @ 998580402 us, last update @ 998260029 us, debounce time is 320373 us
11:48:36 COUNTER 1 @ 998580402 us, : 4
11:48:36 COUNTER 1 @ 998584910 us, : 4
11:48:36 COUNTER 1 @ 998590050 us, : 4
11:48:36 COUNTER 1 @ 998600040 us, : 4
11:48:39 COUNTER: Update of counter 1 @ 1002210044 us, last update @ 998580402 us, debounce time is 3629642 us
11:48:39 COUNTER 1 @ 1002210044 us, : 5
11:48:39 COUNTER 1 @ 1002220031 us, : 5
11:48:39 COUNTER 1 @ 1002230062 us, : 5
11:48:39 COUNTER 1 @ 1002238069 us, : 5
11:48:40 COUNTER: Update of counter 1 @ 1002596283 us, last update @ 1002210044 us, debounce time is 386239 us
11:48:40 COUNTER 1 @ 1002596283 us, : 6
11:48:40 COUNTER 1 @ 1002600758 us, : 6
11:48:40 COUNTER 1 @ 1002610046 us, : 6
11:48:40 COUNTER 1 @ 1002620046 us, : 6
11:48:40 COUNTER 1 @ 1002629895 us, : 6
11:48:40 COUNTER 1 @ 1002631403 us, : 6
11:48:44 COUNTER: Update of counter 1 @ 1006260047 us, last update @ 1002596283 us, debounce time is 3663764 us
11:48:44 COUNTER 1 @ 1006260047 us, : 7
11:48:44 COUNTER 1 @ 1006270055 us, : 7
11:48:44 COUNTER 1 @ 1006277432 us, : 7
11:48:44 COUNTER: Update of counter 1 @ 1006576877 us, last update @ 1006260047 us, debounce time is 316830 us
11:48:44 COUNTER 1 @ 1006576877 us, : 8
11:48:44 COUNTER 1 @ 1006581614 us, : 8
11:48:44 COUNTER 1 @ 1006590048 us, : 8
11:48:44 COUNTER 1 @ 1006599321 us, : 8
11:48:44 COUNTER 1 @ 1006601158 us, : 8
```

So it seems the issue is not in the code, but rather in getting a clean signal...

I suspect it is doing something like this:

[image]

where T1 and T2 are the first and second triggers respectively.  The challenge here is that the time between the signal going low/high can be either quite fast (disk spinning fast) or slow (very little energy used), so I cannot see any good value for `CounterDebounce` which would work for all situations... :-/

I'll close this issue as I think it is not related to the tasmota code, but if you have any other ideas how to work around this I'd be very grateful!

closed 2020-02-12 by robin13

arendst, 2020-02-12:
Thx for your investigation. 

I saw these spikes also when I had these "old" energy meters. Extending debounce time did solve it partly with a chance of missing a quick rotation. 

Time to get a "Smart energy meter" ;-)

fluppie, 2021-02-09:
Could a simple capacitor "fix" this debounce issue? I see the same "problem" on a old 1994 Schlumberger gas meter and the TCR5000. I'm running Tasmota 9.2.0 release.

fluppie, 2021-02-09:
With CounterDebounce 500  / CounterDebounceLow 50 / CounterDebounceHigh 50  it's working as expected :-)!

DST73, 2025-01-23:
I have to use these values. With the values above the counter counts about 4 or 5 times to fast

```
=>savedata 1
=>Teleperiod 30
=>CounterDebounce 12000
=>CounterDebounceLow 500
=>CounterDebounceHigh 500
```

but over the time it counts to low.

PR #8021 (2020-03-31): Enhanced counter debouncing
New commands CounterDebounceLow and CounterDebounceHigh to allow individual debounce times for low and high pulse widths to discard non valid falling edges. These are checked before legacy CounterDebounce checks distance between to valid falling edges.
Useful to have robust counter results when using e.g. an TCRT5000 optical sensor to count electromechanical "Ferraris" electricity meters.

## Description:

**Related issue (if applicable):** fixes #7488, fixes #516

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant fil [... cut]