arendst/Tasmota #7489: 2 dpid infos sent in the same tuya message...

MadDoct, 2020-01-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When receiving a tuyareceived message after setting a certain opening on a curtain motor, the message contains info about dpid1 followed by info for dpid2 in the same sequence

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): tuya curtain motor
- [x ] Tasmota binary firmware version number used: 8.1
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
00:12:30 RSL: stat/tasmota/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"107":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"108":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"52":"Led1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
stop the opening or closing of the curtains in any point

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
when receiving a message like:
55AA0307000D0104000101020200040000006489
tasmota should extract information about both dpid1 and 2
As you can see CmndData 01040001010202000400000064, has 2 "sections":
the first one being about dpid1 - 0104000101 and the second one about dpid2 - 0202000400000064

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

shantur, 2020-01-14:
This is the first implementation of Tuya multi-dpid protocol reporting that I have seen. I know the protocol but handling isn't implemented in Tasmota yet. Can you please update  the issue with more logs.
`weblog 4` and then collect logs for 10 seconds and paste all in the issue please. 
I will see when I get time to implement it

MadDoct, 2020-01-15:
Just did it. The log includes several open and close commands...

shantur, 2020-01-15:
@MadDoct : Can you try the attached binary?

[firmware.bin.zip](https://github.com/arendst/Tasmota/files/4065676/firmware.bin.zip)

MadDoct, 2020-01-15:
Not good...

22:37:35 MQT: tele/CortinaSalaJantar/INFO1 = {"Module":"Tuya MCU","Version":"8.1.0.3(tasmota)","FallbackTopic":"cmnd/DVES_22EEA4_fb/","GroupTopic":"cmnd/tasmotas/"}
22:37:35 MQT: tele/CortinaSalaJantar/INFO2 = {"WebServerMode":"Admin","Hostname":"CortinaSalaJantar-3748","IPAddress":"192.168.2.223"}
22:37:35 MQT: tele/CortinaSalaJantar/INFO3 = {"RestartReason":"Software/System restart"}
22:37:35 MQT: stat/CortinaSalaJantar/RESULT = {"POWER":"OFF"}
22:37:35 MQT: stat/CortinaSalaJantar/POWER = OFF (retained)
22:37:35 QPC: Reset
22:37:35 TYA: Set WiFi LED 4 (0)
22:37:35 TYA: Send "55aa000300010407"
22:37:37 HTP: Main Menu
22:37:37 APP: Boot Count 49
22:37:37 CFG: Saved to flash at FA, Count 305, Bytes 4096
22:37:38 TYA: Send "55aa00000000ff"
22:37:38 MQT: tele/CortinaSalaJantar/RESULT = {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
22:37:38 TYA: Heartbeat
22:37:39 MQT: tele/CortinaSalaJantar/STATE = {"Time":"2020-01-15T22:37:39","Uptime":"0T00:00:12","UptimeSec":12,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":1,"POWER":"OFF","Dimmer":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"SnowMania","BSSId":"00:11:32:4D:24:63","Channel":6,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:06"}}
22:37:41 HTP: Console
22:37:49 TYA: Send "55aa00000000ff"
22:37:49 MQT: tele/CortinaSalaJantar/RESULT = {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
22:37:49 TYA: Heartbeat
22:37:51 CMD: tuyamcu
22:37:51 SRC: WebConsole from 192.168.2.104
22:37:51 CMD: Group 0, Index 1, Command "TUYAMCU", Data ""
22:37:51 MQT: stat/CortinaSalaJantar/RESULT = {"TuyaMCU":[{"fnId":11,"dpId":1},{"fnId":21,"dpId":2}]}
22:37:54 WIF: Checking connection...
22:37:54 WIF: Connected
22:38:00 TYA: Send "55aa00000000ff"
22:38:00 MQT: tele/CortinaSalaJantar/RESULT = {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
22:38:00 TYA: Heartbeat
22:38:11 MQT: Received Topic "cmnd/CortinaSalaJantar/dimmer", Data Size 2, Data "39"
22:38:11 SRC: MQTT
22:38:11 CMD: Group 0, Index 1, Command "DIMMER", Data "39"
22:38:11 SRC: Light
22:38:11 TYA: Send dim value=39 (id=2)
22:38:11 TYA: Send "55aa0006000802020004000000273c"
22:38:11 TYA: Send "55aa0006000501010001010e"
22:38:11 MQT: stat/CortinaSalaJantar/RESULT = {"POWER":"ON","Dimmer":39}
22:38:11 TYA: Send dim value=39 (id=2)
22:38:11 TYA: Send "55aa0006000802020004000000273c"
22:38:11 MQT: tele/CortinaSalaJantar/RESULT = {"TuyaReceived":{"Data":"55AA03070008020200040000002740","Cmnd":7,"CmndData":"0202000400000027","3":{"DpId":3,"DpIdType":7,"DpIdData":"0202000400000027"}}}
22:38:11 TYA: fnId=0 is set for dpId=3
22:38:11 MQT: tele/CortinaSalaJantar/RESULT = {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101"}}
22:38:11 MQT: tele/CortinaSalaJantar/RESULT = {"TuyaReceived":{"Data":"55AA03070008020200040000002740","Cmnd":7,"CmndData":"0202000400000027","3":{"DpId":3,"DpIdType":7,"DpIdData":"0202000400000027"}}}
22:38:11 TYA: fnId=0 is set for dpId=3
22:38:11 TYA: Send "55aa00000000ff"
22:38:11 MQT: tele/CortinaSalaJantar/RESULT = {"TuyaReceived":{"Data":"55AA0307000D0104000101020200040000000025","Cmnd":7,"CmndData":"01040001010202000400000000","170":{"DpId":170,"DpIdType":3,"DpIdData":"0D0104000101020200040000000025010001010A","Type3Data":"
22:38:11 TYA: fnId=0 is set for dpId=170
22:38:11 TYA: fnId=0 is set for dpId=13
22:38:11 MQT: tele/CortinaSalaJantar/RESULT = {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
22:38:11 TYA: Heartbeat
22:38:11 CFG: Saved to flash at F9, Count 306, Bytes 4096
22:38:12 MQT: tele/CortinaSalaJantar/RESULT = {"TuyaReceived":{"Data":"55AA0307000D0104000102020200040000000026","Cmnd":7,"CmndData":"01040001020202000400000000","170":{"DpId":170,"DpIdType":3,"DpIdData":"0D0104000102020200040000000026010001010A","Type3Data":"
22:38:12 TYA: fnId=0 is set for dpId=170
22:38:12 TYA: fnId=0 is set for dpId=13
22:38:14 WIF: Checking connection...
22:38:14 WIF: Connected
22:38:18 MQT: tele/CortinaSalaJantar/RESULT = {"TuyaReceived":{"Data":"55AA0307000D010400010102020004000000274C","Cmnd":7,"CmndData":"01040001010202000400000027","170":{"DpId":170,"DpIdType":3,"DpIdData":"0D010400010102020004000000274C010001010A","Type3Data":"
22:38:18 TYA: fnId=0 is set for dpId=170
22:38:18 TYA: fnId=0 is set for dpId=13
22:38:22 TYA: Send "55aa00000000ff"
22:38:22 MQT: tele/CortinaSalaJantar/RESULT = {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
22:38:22 TYA: Heartbeat
22:38:33 TYA: Send "55aa00000000ff"
22:38:33 MQT: tele/CortinaSalaJantar/RESULT = {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
22:38:33 TYA: Heartbeat
22:38:34 WIF: Checking connection...
22:38:34 WIF: Connected

It stopped identifying the correct dpids...

shantur, 2020-01-15:
@MadDoct Found the issue. Can you please try this one

[firmware.bin.zip](https://github.com/arendst/Tasmota/files/4068236/firmware.bin.zip)

commit 016b204c (2020-01-16): Merge pull request #7531 from shantur/development

Implement new Tuya multi dpid protocol. Fixes #7489
commit 920501fe (2020-01-16): Implement new Tuya multi dpid protocol. Fixes #7489
closed 2020-01-16 by arendst