arendst/Tasmota #7491: Per EU directive 80/181/EEC (the "metric directive"), the correct symbol is lower-case "var"

muhkuhgh, 2020-01-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Per EU directive 80/181/EEC (the "metric directive"), the correct symbol is lower-case "var"
DIN 40110 also says so. Do you have any good reason to use "VAr"?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-01-12:
Many good reasons:

> (') Special names for the unit of power: die name volt-ampere (symbol 'VA') when it is used to express the apparent
power of alternating electric current, and var (symbol 'var') when it is used to express reactive electric power. The
'var' is not included in General Conference on Weights and Measures (CGPM) resolutions.

> The unit "var" is allowed by the International System of Units (SI) even though the unit var is representative of a form of power.[2] SI allows one to specify units to indicate common sense physical considerations. Per EU directive 80/181/EEC (the "metric directive"), the correct symbol is lower-case "var",[3] although the spellings "Var" and "VAr" are commonly seen, and "VAR" is widely used throughout the power industry.

So although the 1980 directive mentions "var" there is no reason not to use "VAr".

ascillato2, 2020-01-14:
Thanks for sharing this information. 

Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-01-14 by ascillato2