arendst/Tasmota #7496: Tasmota crashes on receiving ssdp:discover when MQTT/rsyslog logging set to 3+

os400, 2020-01-12:
### PROBLEM DESCRIPTION
_Tasmota devices restarting when receiving an 'ssdp:discover' request directed at 'ssdp:all' when MQTT or Syslog logging set to 3 or 4._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____ESP8266
- [ ] Tasmota binary firmware version number used: _____8.1.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
15:46:13 CMD: Backlog Template; Module; GPIO 255
15:46:13 MQT: stat/tasmota/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
15:46:13 MQT: stat/tasmota/RESULT = {"Module":{"18":"Generic"}}
15:46:13 MQT: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"22":"Relay2"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
 15:47:14 CMD: status 0
15:47:14 MQT: stat/tasmota/STATUS = {"Status":{"Module":18,"FriendlyName":["bbt","Tasmota2"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:47:14 MQT: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Exception","Uptime":"0T00:25:34","StartupUTC":"2020-01-12T14:21:40","Sleep":50,"CfgHolder":4617,"BootCount":18,"SaveCount":47,"SaveAddress":"F8000"}}
15:47:14 MQT: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"364/699"}}
15:47:14 MQT: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"192.168.2.113","LogPort":514,"SSId":["BT",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
15:47:14 MQT: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
15:47:14 MQT: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-0909","IPAddress":"192.168.2.207","Gateway":"192.168.2.10","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.10","Mac":"EC:FA:BC:C0:A3:8D","Webserver":2,"WifiConfig":4}}
15:47:14 MQT: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.113","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C0A38D","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
15:47:14 MQT: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Sun Jan 12 14:47:14 2020","Local":"Sun Jan 12 15:47:14 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:40","Sunset":"17:16"}}
15:47:14 MQT: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-01-12T15:47:14"}}
15:47:14 MQT: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-01-12T15:47:14","Uptime":"0T00:25:34","UptimeSec":1534,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"BT","BSSId":"FC:F5:28:E4:51:84","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:07"}}}
15:47:14 MQT: stat/tasmota/STATUS12 = {"StatusSTK":{"Exception":9,"Reason":"Exception","EPC":["40105f18","00000000","00000000"],"EXCVADDR":"00000003","DEPC":"00000000","CallChain":["40106149","402331d0","402415ec","402169dd","40250906","40216b3d","401016d3","402465b0","40244740","40218059","4023df20","40244740","402275a7","4010170a","40106005","4000050c","4024ca3a","40000f68","40000f58","4010170a","40244700","40250660","401016d3","40262328","402430b6","402683a0","40243ef2","402683ad","402683f2","40267fef","40000f49"]}}
15:51:53 MQT: tele/tasmota/STATE = {"Time":"2020-01-12T15:51:53","Uptime":"0T00:30:13","UptimeSec":1813,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"OFF",

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
Jan 12 14:14:03 tasmota-0909 ESP-UDP: Packet (92)
Jan 12 14:14:12 tasmota-0909 ESP-WIF: Connected
Jan 12 14:14:12 tasmota-0909 ESP-HTP: Web server active on tasmota-0909 with IP address 192.168.2.207
Jan 12 14:14:12 tasmota-0909 ESP-UPP: Multicast (re)joined
Jan 12 14:14:13 tasmota-0909 ESP-UPP: Multicast disabled
Jan 12 14:14:13 tasmota-0909 ESP-MQT: Attempting connection...
Jan 12 14:14:13 tasmota-0909 ESP-MQT: Connected
Jan 12 14:14:13 tasmota-0909 ESP-MQT: tele/tasmota/LWT = Online (retained)
Jan 12 14:14:13 tasmota-0909 ESP-MQT: cmnd/tasmota/POWER =
Jan 12 14:14:13 tasmota-0909 ESP-MQT: Subscribe to cmnd/tasmota/#
Jan 12 14:14:13 tasmota-0909 ESP-MQT: Subscribe to cmnd/tasmotas/#
Jan 12 14:14:13 tasmota-0909 ESP-MQT: Subscribe to cmnd/DVES_C0A38D_fb/#
Jan 12 14:14:13 tasmota-0909 ESP-MQT: tele/tasmota/INFO1 = {"Module":"Generic","Version":"8.1.0(tasmota)","FallbackTopic":"cmnd/DVES_C0A38D_fb/","GroupTopic":"cmnd/tasmotas/"}
Jan 12 14:14:13 tasmota-0909 ESP-MQT: tele/tasmota/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota-0909","IPAddress":"192.168.2.207"}
Jan 12 14:14:13 tasmota-0909 ESP-MQT: tele/tasmota/INFO3 = {"RestartReason":{"Exception":9,"Reason":"Exception","EPC":["40105f18","00000000","00000000"],"EXCVADDR":"00000003","DEPC":"00000000","CallChain":["40106149","402331d0","402415ec","402169dd","40250906","40216b3d","401016d3","402465b0","40244740","40218059","4023df20","40244740","402275a7","4024a423","40106005","4000050c","40000f68","40000f58","40243ee0","4024983d","40244700","4024b031","4024b0bf","4027f34c","402430b6","402683a0","40243ef2","402683ad","402683f2","40240743","40000f49"]}}
```

### TO REPRODUCE
My "BT Smarthub" recreates this every half hour... I've copied the gist of the message and can recreate a request (in DOS format) and send it with netcat:

$ cat sendssdp.txt
M-SEARCH * HTTP/1.1
HOST:239.255.255.250:1900
ST:ssdp:all
MAN:"ssdp:discover"
MX:3

$ cat sendssdp.txt | nc -w 3 -u 239.255.255.250 1900

### EXPECTED BEHAVIOUR
_UPP: Hue 3 response packets sent to 192.168.2.n:p_

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
First discovered with MQTT logging level 3 on a BSD29 tuya compatible plug that caused the light to flicker every half an hour. Found same problem on ESP8266 'v3'. Found the ssdp packet causing the problem, reset logging to 2 and discovered the problem had gone....

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-01-12:
I cannot reproduce on 8.1.0.3. This is what I get:

```
22:12:22 UDP: Packet (84)
22:12:24 UPP: Hue 3 response packets sent to 172.20.10.3:50687
```

Can you type `wc -c sendssdp.txt` to make sure it's 84 bytes?

os400, 2020-01-12:
$ wc -c sendssdp.txt
88 sendssdp.txt
$ hexdump -C sendssdp.txt
00000000  4d 2d 53 45 41 52 43 48  20 2a 20 48 54 54 50 2f  |M-SEARCH * HTTP/|
00000010  31 2e 31 0d 0a 48 4f 53  54 3a 32 33 39 2e 32 35  |1.1..HOST:239.25|
00000020  35 2e 32 35 35 2e 32 35  30 3a 31 39 30 30 0d 0a  |5.255.250:1900..|
00000030  53 54 3a 73 73 64 70 3a  61 6c 6c 0d 0a 4d 61 6e  |ST:ssdp:all..Man|
00000040  3a 22 73 73 64 70 3a 64  69 73 63 6f 76 65 72 22  |:"ssdp:discover"|
00000050  0d 0a 4d 58 3a 33 0d 0a                           |..MX:3..|
00000058
And the log...
23:18:14 UDP: Packet (88)
23:18:15 UPP: Hue 3 response packets sent to 192.168.2.113:55458

And the BSD29 with the original restarting, probably since friday... tuyaconvert'ed with the bin supplied by them, then OTA upgraded:

Program Version	8.1.0(tasmota)
Build Date & Time	2019-12-25T12:33:25
Core/SDK Version	2_6_1/2.2.2-dev(38a443e)
Uptime	0T04:50:06
Flash write Count	284 at 0xFB000
Boot Count	189
Restart Reason	Exception

Interesting you haven't got the same results because once I tried out an ESP8266 I assumed they would be easily recreatable.

os400, 2020-01-12:
And, immediately after, I changed the syslog level to 4

```
Jan 12 23:26:12 tasmota-7193 ESP-LOG: SerialLog 3, WebLog 4, MqttLog 2, SysLog 4, LogHost 192.168.2.113, LogPort 514, TelePeriod 300
...
Jan 12 23:26:25 tasmota-7193 ESP-UDP: Packet (88)
...reboots...
Jan 12 23:26:35 tasmota-7193 ESP-WIF: Connected
...
Jan 12 23:26:36 tasmota-7193 ESP-MQT: tele/tasmota/INFO3 = {"RestartReason":{"Exception":9,"Reason":"Exception","EPC":["40105fbd","00000000","00000000"],"EXCVADDR":"ffffffff","DEPC":"00000000","CallChain":["40261c70","4026c745","40262583","402624ec","40261e0a","40000f49","40000f49","40000e19","40001878","40105865","4010586b","4010000d","40260e38","40260de9","40246d4a","402475e1","402630af","402630fe","401009b4","401024ac","402621bb","4025575b","40105daf","40249145","40246d4a","402493d2","402497e8","401009b4","401024ac","4010575f","40103043"]}}
...
Jan 12 23:26:37 tasmota-7193 ESP-APP: Boot Count 191
```

os400, 2020-01-12:
And back to level 2, but weblog still on 4:
```
23:32:34 LOG: SerialLog 3, WebLog 4, MqttLog 2, SysLog 2, LogHost 192.168.2.113, LogPort 514, TelePeriod 300
23:32:34 HTP: Configuration
23:32:35 CFG: Saved to flash at F6, Count 289, Bytes 4096
23:32:35 WIF: Checking connection...
23:32:35 WIF: Connected
23:32:37 HTP: Main Menu
23:32:39 HTP: Console
23:32:47 UDP: Packet (88)
23:32:49 UPP: Hue 3 response packets sent to 192.168.2.113:45021
23:32:55 WIF: Checking connection...
23:32:55 WIF: Connected
```

s-hadinger, 2020-01-12:
Ah ah. So the issue is on syslog, that's why. I thought the exception was caused by the log generator but it isn't. Unfortunately I don't use syslog.

os400, 2020-01-13:
One thing I find odd is that syslog goes out on port 1900 (to 514), but either way, it crashes with MQTT logging set at 4 as well...

arendst, 2020-01-13:
It seems that exception 9 occurs when syslog (using UDP) or mqqtlog (using TCP) is enabled within a Ticker interrupt ruotine.

The same issue occured in the RTC routine where I had to move the AddLog function out of the interrupt routine and perform it based on a flag set within the interrupt routine.

We have to rethink routines HueRespondToMSearch and WemoRespondToMSearch providing AddLog information within their ticker interrupt routine.

os400, 2020-01-13:
Thanks, yes I can see it now... the only suggestion I can think of is replace ntp_synced_msg with a character field containing the log message itself and the 'PerformEverySecond' routine posts the message if it's not empty... then it'd be easy to do the same in the emulation routines...?

arendst, 2020-01-14:
That would do if these messages were only appearing once a second. If you monitored your uPnP traffic you would be flabbergasted by the amount of rubbish your local devices (Sonos, TP-Link router, ChromeCast) send over your network just to find each other. I had to add tickers for handling them as they otherwise would flood Tasmota and break the system.

Anyway, as a workaround do not use log levels above 2 or disable syslog/mqttlog.

I think I just drop these messages for syslog and mqttlog. Remember they are debug messages and you're still able to see them in the serial and weblog. No need for constant flooding the syslog either.

arendst, 2020-01-14:
As the messages can only pop-up after 1500 to 2200 mSecs it makes sense indeed to prep the messages and provide them once the interrupt service routine has finished.

This is the way it is implemented now. My virtual tests did not exception anymore.

os400, 2020-01-14:
That's great... to confirm 

```
Program Version	8.1.0.3(tasmota)
Build Date & Time	2020-01-14T12:43:56
```

```
13:01:52 MQT: stat/tasmota/RESULT = {"MqttLog":4}
```
MQTT logging after sending fake ssdp:discover
```
20-01-14 13:01:05 stat/tasmota/LOGGING 13:01:05 UDP: Packet (88)
20-01-14 13:01:07 stat/tasmota/LOGGING 13:01:07 UPP: Hue 3 response packets sent to 192.168.2.113:38835
20-01-14 13:01:13 stat/tasmota/LOGGING 13:01:13 WIF: Checking connection...
20-01-14 13:01:13 stat/tasmota/LOGGING 13:01:13 WIF: Connected
```
And as it's only a few minutes wait until a real one comes along...
```
20-01-14 13:14:01 stat/tasmota/LOGGING 13:14:01 UDP: Packet (92)
20-01-14 13:14:03 stat/tasmota/LOGGING 13:14:03 UDP: Packet (174)
20-01-14 13:14:04 stat/tasmota/LOGGING 13:14:03 UPP: Hue 3 response packets sent to 192.168.2.10:55944
```

Thanks Theo!

closed 2020-01-14 by os400

miky-boy, 2020-05-17:
Hi Guys,

I installed sucessfull the Tasmota firmware on my ZMAi‐90 Energy Module, explained on the https://www.creationfactory.co/2019/12/reverse-engineering-zmai-90-din-rail.html

I activated the MQTT and now i detected this same error exceptions every 15 minutes below:

03:58:08 MQT: tele/Tasmota/INFO1 = {"Module":"ZMAi‐90","Version":"8.3.0.1(63d15b6-tasmota)","FallbackTopic":"cmnd/DVES_DE36E9_fb/","GroupTopic":"cmnd/tasmotas/"}
03:58:08 MQT: tele/Tasmota/INFO2 = {"WebServerMode":"Admin","Hostname":"Tasmota","IPAddress":"192.168.2.65"}
03:58:08 MQT: tele/Tasmota/INFO3 = {"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000"}}
03:58:08 MQT: stat/Tasmota/RESULT = {"POWER":"ON"}

After the error the measures from the serial interface stoped working and i need to activate with "Baudrate 9600". After this the equipment send measures again along 15minutes and over and over..

I detected this error on firmware 8.3.0 and i upgraded after to 8.3.0.1, and the problem remains.

Any idea?
Thanks and regards.

miky-boy, 2020-05-19:
Hi Guys,

After some research and tests i solved the problem with the following:

#Creating a new rule:
Rule2 ON System#Boot DO baudrate 9600 ENDON

#Next, enable the rule by typing:
Rule2 on

Thanks and regards