arendst/Tasmota #7500: Sonoff dual physical switch cannot control the switch state

plutosherry, 2020-01-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [X ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _Sonoff Dual_
- [X] Tasmota binary firmware version number used: _6.6_
  - [no] Pre-compiled
  - [no] Self-compiled
    - [no] IDE / Compiler used: _____
- [X] Flashing tools used: _esphome_
- [no] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [no] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here: no rules

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Sonoff dual physical switch cannot control the switch state......PLZ Help......!
In addition to using the switch control in webui, But the physical button can't control sonoff dual.
When i press or double-click, its can't change the status of the switch..
I ran into this problem using tasmota 6.6
Besides that tasmota6. X can connect to WiFi normally, tasmota7. X or 8. X cannot connect to 2.4gwifi. I think it's probably the core version that causes the WiFi connection problem
Does core version affect the use of physical buttons?
This problem is only encountered on sonoff dual, but it is normal on sonoff basic

Jason2866, 2020-01-13:
There are no bugfixes for v.6.6. Old version without support.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

plutosherry, 2020-01-13:
I have solved the problem caused by the template setting. But shouldn't this template be the default model for sonoff dual? Maybe it's a bug. In a word, I just manually set the GPIO0 for button1.and its worked!!!

thanks alots!

closed 2020-01-14 by ascillato2