arendst/Tasmota #7505: Multichannel Gas Sensor - can't read the data

tikky, 2020-01-13:
### PROBLEM DESCRIPTION
I am trying to connect Grove - Multichannel Gas Sensor which seems to be supported by Tasmota
on this list: Grove - Multichannel Gas Sensor.

When I am connecting my sensor to Wemos D1 mini with Tasmota I am using
configuration:
D2 GPIO4 - I2C SDA(6)
D1 GPIO5 - I2C SCL(5)

After reboot I can't see readings, however in console i2cscan can properly find this senosr on I2C bus on default address 0x04:
00:08:01 CMD: i2cscan
00:08:01 RSL: stat/tasmota/RESULT = {"I2CScan":"Device(s) found at 0x04"}

So, how can I read data if sensor is properly found?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: 8.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: pytool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
 not using rules
```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
00:01:01 CMD: Status 0
00:01:01 RSL: stat/tasmota/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:01:01 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:02","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":32,"SaveCount":45,"SaveAddress":"FA000"}}
00:01:01 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"303/699"}}
00:01:01 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SIGNATI",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
00:01:01 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":2,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
00:01:01 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-6188","IPAddress":"192.168.6.240","Gateway":"192.168.6.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.6.1","Mac":"2C:F4:32:78:18:2C","Webserver":2,"WifiConfig":4}}
00:01:01 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_78182C","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
00:01:01 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:01:01 1970","Local":"Thu Jan 01 00:01:01 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
00:01:01 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:01:01"}}
00:01:01 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:01:01","Uptime":"0T00:01:02","UptimeSec":62,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"SIGNATI","BSSId":"00:0C:42:3A:D1:1C","Channel":5,"RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
N/A
```

### TO REPRODUCE
Connect sensor and configure GPIO to use I2C sesnors.

### EXPECTED BEHAVIOUR
I would expect to see the readings from sensor

### SCREENSHOTS
not needed

### ADDITIONAL CONTEXT
Have tried other I2C sensor like BME280 and this sensor works as expected.

ascillato, 2020-01-13:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Jason2866, 2020-01-13:
Not a bug. You have somewhere a error in your setup. MGS is working
[image]

closed 2020-01-14 by ascillato2