arendst/Tasmota #7506: Accept JSON objects from Serial and forward them to MQTT bridge

kioan, 2020-01-13:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
A device is transmitting data as JSON strings over serial port. It would be useful if there was an option to configure Tasmota to handle the serial received data as JSON strings.

**Describe the solution you'd like**  
Let's assume that the device is transmittingthe following JSON string over serial 
`{"timestamp":"1578948149","status":"ok","coord":{"x":"60","y":"154"}}`

The above in more readable format:
```
{
  "timestamp": "1578948149",
  "status": "ok",
  "coord": {
    "x": "60",
    "y": "154"
  }
}
```

With the current implementation, when this string is received by the Tasmota, it will be forwarded to the MQTT as
`tele/myDevice/RESULT = {"SerialReceived":"{"timestamp":"1578948149","status":"ok","coord":{"x":"60","y":"154"}}"}`
which of course isn't a valid JSON object because the whole serial string is enclosed in double quotes and is sent as the value of the SerialReceived field.

Ideally we would like the serial string to be handled as JSON object so that it is forwarded to the MQTT topic as
`{"SerialReceived":{"timestamp":"1578948149","status":"ok","coord":{"x":"60","y":"154"}}}`

The above in more readable format:
```
{
  "SerialReceived": {
    "timestamp": "1578948149",
    "status": "ok",
    "coord": {
      "x": "60",
      "y": "154"
    }
  }
}
```
It would be nice if there was an option to configure the Tasmota serial receiver to handle the received strings either as string (which is the current implementation) or as a JSON object.

**Describe alternatives you've considered**  
If the received string is received with all the double quotes escaped as `\"`, then it is handled correctly. The problem is that we can not force the transmitting device to send escaped strings as it is programmed to generate JSON objects.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-01-14:
Hi,

Which device is the one sending data in json format? What is the usecase that will require this work and flash space?

kioan, 2020-01-14:
Hello,
in my case there are some old card readers that originally where sending this data over TCP/IP but because the were located hundreds of meters away they were connected using IP-to-RS485 converters. At the near end of those RS485 link an ESP would push those data to the MQTT server.
(Of course if that system wasn't already installed, it would be designed completely different)

I now this is a very specific scenario, but due to the relative simplicity (I suppose) of a change to the serial bridge (it justs needs not to enclose the serial string in quotes), I thought I could propose it here and maybe it would be useful for other users as well.

It could be a quick way to bridge other custom devices to an MQTT server using an ESP running Tasmota and allow easy handling of those data using NodeRed etc.

arendst, 2020-01-14:
I'll see what I can do. A simple check for received data starting with a "{" might be enough to not use quoted text.

arendst, 2020-01-14:
Give it a try.

kioan, 2020-01-14:
It works perfectly!

As you can see bellow, the JSON string I mentioned yesterday as an example, now can be easily parsed as an Object:
[image]

Thank you very much!

closed 2020-01-14 by kioan

dukenewcombe, 2020-02-02:
Will this be included in future releases?

ascillato, 2020-02-02:
As explained in previous comments. It is already added to the development branch. You can use it if you update to latest Tasmota. Download the latest from thehackbox.org/tasmota