arendst/Tasmota #7507: The console output should use all the available vertical space of the browser tab/window

rominator1983, 2020-01-13:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

When looking at the console output of the web gui the user has to scroll vertically more often than neccessary especially when using weblog 4 (many output lines).

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

The console output should use all the available vertical space of the browser tab/window.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

I *might* be able to come up with a pull request if someone points me to the right direction in the code that is responsible for rendering of the HTML provided there is an agreement on how the HTML should be (CSS-Table-Layout?) and if this feature should be implemented at all.

**(Please, remember to close the issue when the problem has been addressed)**
I do hope that I will.

ascillato, 2020-01-14:
Please, provide browser name and screenshot. I can't reproduce your issue

MaceMoneta, 2020-01-14:
Google Chrome
Version 79.0.3945.86 (Official Build) (64-bit)

[image]

arendst, 2020-01-14:
The provided commit adds user control over console vetical size (right bottom handle).

As the console textarea is part of a fieldset AND we have to service small mobile phone screens full textarea control using window size is not feasable; on a mobile there would be only a few lines visible.

The provided is a compromise allowing you to make the console textarea as large as you like.

arendst, 2020-01-14:
Had to revert the change as the scrolling stopped working. Investigating.

arendst, 2020-01-14:
This should work as explained before.

ascillato2, 2020-01-14:
Closing as this request has been added. Thanks.

closed 2020-01-14 by ascillato2

rominator1983, 2020-01-14:
Wow that was quick. I do understand what you say about mobile devices and I think the solution (from your description) is a good compromise.

I will try to test the solution on one of my devices in the next days. I have not yet flashed a tasmota development build.
Until now I only compiled my custom firmware based on the release tags.

rominator1983, 2020-01-21:
just checked it out on develop and it serves my purpose. thanks again