arendst/Tasmota #7512: Cannot create Lambda functions with Node.js 8.10 runtime

irvanfarasatha, 2020-01-14:
### PROBLEM DESCRIPTION
As announced [here](https://aws.amazon.com/blogs/compute/node-js-12-x-runtime-now-available-in-aws-lambda/) and documented [here](https://docs.aws.amazon.com/lambda/latest/dg/runtime-support-policy.html), lambdas with Node.js 8.10 runtime cannot be created after January 6, 2020 (or updated after February 3, 2020).

I found out that due to current update which requires nodejs12.x rather than nodejs8.10, AWS is unable to CreateCSR. This condition leads to the situation which AWS integration is fully inapplicable. 

If anyone have any common issue or already have solution give me some information.

### REQUESTED INFORMATION

- [1] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [1] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [1] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [1] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [1] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [1] Read and Follow every step to integrate AWS IoT with Tasmota carefully

### SCREENSHOTS
[image]

closed 2020-01-14 by close-issue-app[bot]

s-hadinger, 2020-01-14:
I will provide a new set of scripts shortly to automate the whole process

irvanfarasatha, 2020-01-14:
@s-hadinger sounds great. Look forward for the update :)

dwtowner, 2020-03-26:
I hit this problem too. Here is an updated template that can be used:

https://gist.github.com/dwtowner/59dd0624d952307b95e99217c4077bb0

To use it:

- download the contents of the gist.
- Run the stack creation link:  (e.g., https://console.aws.amazon.com/cloudformation/home?region=eu-west-1#/stacks/new?stackName=IotThingGenerator&templateURL=https://tasmota-aws-iot.s3-eu-west-1.amazonaws.com/IotThingGenerator.yaml)
- On the first page that appears, scroll down and choose `Upload Template', and point the browser at your local copy of gist downloaded from above.

The stack should then be created successfully.