arendst/Tasmota #7513: IRsend problem with self-compiled firmware {"IRSend":"Invalid JSON"}

Elbflorenz, 2020-01-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When I flash the tasmota-ir.bin I can send IR codes by e.g. 

IRSend 0,111,90,...

But if I compile and flash my own firmware I get the above error

{"IRSend":"Invalid JSON"}

I have #define FIRMWARE_IR_CUSTOM  enabled in my_user_config.h

Any ideas what the reason might be?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): IR blaster
- [ ] Tasmota binary firmware version number used: 7.1
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Arduino
- [ ] Flashing tools used: Arduino
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
15:30:33 MQT: stat/tasmota/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
15:30:34 MQT: stat/tasmota/RESULT = {"Module":{"0":"Generic"}}
15:30:34 RSL: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"52":"Led1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"51":"IRrecv"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"8":"IRsend"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
15:31:28 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:31:28 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:20:31","StartupUTC":"2020-01-14T14:10:57","Sleep":50,"CfgHolder":4617,"BootCount":7,"SaveCount":17,"SaveAddress":"F9000"}}
15:31:28 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"7.1.0(tasmota)","BuildDateTime":"2020-01-14T23:08:11","Boot":7,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)","Hardware":"ESP8266EX"}}
15:31:28 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Gruenesgewoelbe",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000100","00000000"]}}
15:31:28 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":602,"Free":400,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","0F5AE3B6","003683A1","220217C5","010008C0","00007881"],"Drivers":"1,2,3,4,5,6,7,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,22,26,34"}}
15:31:28 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-4627","IPAddress":"192.168.1.162","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"D8:BF:C0:50:92:13","Webserver":2,"WifiConfig":4}}
15:31:28 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.20","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_509213","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":2500,"KEEPALIVE":30}}
15:31:28 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Tue Jan 14 14:31:28 2020","Local":"Tue Jan 14 15:31:28 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:39","Sunset":"17:18"}}
15:31:28 MQT: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-01-14T15:31:28"}}
15:31:28 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-01-14T15:31:28","Uptime":"0T00:20:31","UptimeSec":1231,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Gruenesgewoelbe","BSSId":"D4:6E:0E:52:5E:D2","Channel":4,"RSSI":90,"LinkCount":1,"Downtime":"0T00:00:27"}}}
15:31:30 RSL: tele/tasmota/STATE = {"Time":"2020-01-14T15:31:30","Uptime":"0T00:20:33","UptimeSec":1233,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Gruenesgewoelbe","BSSId":"D4:6E:0E:52:5E:D2","Channel":4,"RSSI":86,"LinkCount":1,"Downtime":"0T00:00:27"}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Why is IRSend not reacting the same way?

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-01-14:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

s-hadinger, 2020-01-14:
You need to switch to `tasmota-ircustom` in platform.io to enable compile time options. This is required by IRremoteESP8266, and cannot rely on #defines.

closed 2020-01-14 by ascillato2