arendst/Tasmota #7515: PowerDelta 1xx is missing changes that (even) WebUI gets

muhkuhgh, 2020-01-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

It seems that the PowerDelta function uses its own but inferior Energy.power_history or gets triggered less often. I haven't searched the whole Code yet.

But this should be enough to clarify:
I put on a dynamic load.
I observe the values in the WebUI.. I see e. g. 
starting at 0
7 W
15 W
13 W
3 W
0

I have PowerDelta set so 102 before. So it should fire on every 2 Watt change. But in the console log I will only find

15
3

Shouldn't PowerDelta be at least as good as the WebUI?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ x ] Tasmota binary firmware version number used: 8.1
  - [ x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-01-14:
Which device are you using?

Please, can you complete the template. Please share status 0 output and the console output when this behavior happens. Remember to set weblog to 4 for that.

muhkuhgh, 2020-01-14:
> 
> 
> Which device are you using?
> 
> Please, can you complete the template. Please share status 0 output and the console output when this behavior happens. Remember to set weblog to 4 for that.

Sorry, I will expand it when I come home. For the time being you may have a look at my edit regarding the to/from 0 Problem.
Gosund SP112 and Avatar AWP07L.

muhkuhgh, 2020-01-15:
For the to/from 0 not getting caught it should be enough to move the "min_power > 0" condition in these lines here to only affect the <101 condition part...  to fix the fix. That's like 1 Minute of coding, but I don't know how to create a prober pull request.. https://github.com/arendst/Tasmota/blob/8aa91096566c5a3d885c693c9f28cee387ec7b1b/tasmota/xdrv_03_energy.ino#L296

arendst, 2020-01-16:
See https://github.com/arendst/Tasmota/issues/7509#issuecomment-575211932 for some background information.

Your reference to delta is wrong. The goal of PowerDelta is measuring power difference not absolute power. So delta must contain a value representing the power from three seconds ago related to the current power. If it is equal there is no PowerDelta.

Your low power measurements are also at the bottom of what your cheap hardware can measure. Expect no valid data below 5W.

To measure power changes the code compares a sample from three seconds ago to the current value and acts accordingly. If you change loads within this time do not expect intermediate results.

ascillato2, 2020-01-16:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

muhkuhgh, 2020-01-16:
> See [#7509 (comment)](https://github.com/arendst/Tasmota/issues/7509#issuecomment-575211932) for some background information.
> 
> Your reference to delta is wrong. The goal of PowerDelta is measuring power difference not absolute power. So delta must contain a value representing the power from three seconds ago related to the current power. If it is equal there is no PowerDelta.
> 
=> What did you lead to assume I interpreted it any different?
With PowerDelta set to 102 it should send me an update for every change greater 2 Watt it notices... 
But if the load was 10 Watt for 10 seconds then 0 for 10 Seconds then 20 Watt for 10 Seconds and then goes back to 0 it should notice 10,0,20,0. But all I get is 10 and 20. Because with the FIX for not dividing through zero (min_power>0) it no longer works coming from or going down to zero! Allthough for the absolute powerdelta computation that condition is not needed, as we only need the division for the relative percentual powerdelta.. Where am I wrong here? I really don't get it (yet?)

> Your low power measurements are also at the bottom of what your cheap hardware can measure. Expect no valid data below 5W.
=> I am also using this for a master thesis for one of my students I am currently in charge of.. this circuitry is designed around low loads..

> To measure power changes the code compares a sample from three seconds ago to the current value and acts accordingly. If you change loads within this time do not expect intermediate results.
=> How does it come then, that my Gosund SP112 flashed with the latest stock 8.1 tasmota is reporting up to 4 values per second (only using PowerDelta 102 command) and get's the peak of the fridge turning on every time perfectly?

> Decent results for a 20W load are:
> 
> ```
> 16:56:43 MQT: stat/shp5/RESULT = {"POWER1":"ON"}
> 16:56:43 MQT: stat/shp5/POWER1 = ON
> 16:56:45 MQT: tele/shp5/SENSOR = {"Time":"2020-01-16T16:56:45","ENERGY":{"TotalStartTime":"2019-06-03T15:18:50","Total":15.122,"Yesterday":0.076,"Today":0.058,"Power":19.16,"ApparentPower":22.13,"ReactivePower":11.10,"Factor":0.87,"Voltage":224.7,"Current":0.098}}
> 16:56:58 MQT: stat/shp5/RESULT = {"POWER1":"OFF"}
> 16:56:58 MQT: stat/shp5/POWER1 = OFF
> 16:57:00 MQT: tele/shp5/SENSOR = {"Time":"2020-01-16T16:57:00","ENERGY":{"TotalStartTime":"2019-06-03T15:18:50","Total":15.122,"Yesterday":0.076,"Today":0.058,"Power":4.57,"ApparentPower":11.83,"ReactivePower":10.90,"Factor":0.39,"Voltage":227.0,"Current":0.052}}
> ```

arendst, 2020-01-16:
Ah regarding 0 I see the issue. Will look into it.

My devices almost never return to 0 so I didn't see it. Now it makes sense.

muhkuhgh, 2020-01-16:
> Ah regarding 0 I see the issue. Will look into it.
> 
> My devices almost never return to 0 so I didn't see it. Now it makes sense.

Thank you so much - I was starting do despair ;-)

> For the to/from 0 not getting caught it should probably be enough to move the "min_power > 0" condition in these lines here to only affect the <101 condition part... to fix the fix. That's like 1 Minute of coding, but I don't know how to create a prober pull request.. like just enclosuring the '(delta * 100) / min_power)' part with a 'if(min_power>0) else false' part..

> https://github.com/arendst/Tasmota/blob/8aa91096566c5a3d885c693c9f28cee387ec7b1b/tasmota/xdrv_03_energy.ino#L296

ascillato, 2020-01-17:
@muhkuhgh 

Hi, This issue has been fixed in the actual codebase. Please, can you test latest changes?

closed 2020-01-18 by ascillato2