arendst/Tasmota #7519: Switch Double/Tripple toggle

jensuffhaus, 2020-01-15:
**Have you looked for this feature in other issues and in the docs?**  YES

**Is your feature request related to a problem? Please describe.**  
I use Sonoff Mini's with Tasmota to enhance all my existing light switches and swapped them for momentarty push buttons. All lights work as they did before using the local input of the mini but I now get the added HOLD feature for scenes/automation. On some switches however I could physically not get a momentary button and am forced to use a latching switch. This means I loose the HOLD feature but would still like to have it at that switch.

**Describe the solution you'd like**  
Implement a long hold feature for latching switches where instead of holding long, you toggle quickly, on/off. The local relay is only toggled if the switch does not toggle more than once within a give delay. there could be any number of actions and the MQTT message could include the actual number of toggles as the payload. on/off/on/off/on would have a payload of 5.

**Describe alternatives you've considered**  
I could not find any alternatives that did not involve changing to a momentary button and that would require physical destruction of the device making the change pointless.

**Additional context**  
This is for bedside lamps that I have enahnced with a sonoff mini. the original toggle switch of the lamp works as it did via the mini but the lamp can now also be controlled by the system. I would like the multi toggle feature so I can engage the "Good night" scene once I am in bed using the toggle switch of the lamp.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-01-15:
Hold it is already implemented. If you want to use it in a latching switch, you just need to configure it properly with switchmode command.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-01-15 by ascillato2

to-scho, 2020-01-15:
If I understood right, this is not supported up to now by any SwitchMode or button feature. Feature requested is similar to some LED dimmers that you can operate by a standard wall mount stationary (non momentary) switch. When you change position fast within a certain interval some extra action is done. On/Off State of the dimmer is only switched when there is no 2nd switch action within a certain interval.

I created PR #7522 that introduce three new switch modes which should at least give a double toggle according to @jensuffhaus needs.

jensuffhaus, 2020-01-15:
@to-scho Thank you. You are spot on. Will wait for the PR to be included in the release.