arendst/Tasmota #7520: IR payload with "Hash" restarts the sonoff-basic

adrianofoschi, 2020-01-15:
### PROBLEM DESCRIPTION
I'm using sonoff basic with ir transmitter diode to send IR signals to my tv, climate etc.
I'm using a receiver diode to capture the IR signals from the remote across tasmota console.
The procedure works perfectly with my tv but I have problems with the climate.

### REQUESTED INFORMATION
```
  Console output here:
10:25:25 MQT: tele/presa_libera/RESULT = {"Time":"2020-01-15T10:25:25","IrReceived":{"Protocol":"UNKNOWN","Bits":28,"Hash":"0x08800347"}}

```
### TO REPRODUCE
I'm using homeassistant mqtt.publish service to send the IR signal, and it works perfectly with my tv that has "Data" keyword into payload. 
If I send the IR via transmitter the sonoff-basic restarts and doesn't send the signal.
If I replace "Hash" keyword with "Data" it sends the code but naturally the climate doesn't turn on.

### EXPECTED BEHAVIOUR
The sonoff should send the code and the climate turn on.

Jason2866, 2020-01-15:
You have to use the Tasmota-IR version

adrianofoschi, 2020-01-15:
I updated the issue with more details about the IR codes that I received from original remote or android app.

Jason2866, 2020-01-15:
Please close. Solution is already explained and described (in Discord).

closed 2020-01-15 by ascillato2