arendst/Tasmota #7525: Zigbee2Tasmota aqara magic cube events

moneybag, 2020-01-15:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  

the cube has this sensors results
18:24:33 MQT: tele/zigbeegateway/SENSOR = {"ZigbeeReceived":{"0x218E":{"PresentValue":259,"LinkQuality":107}}}

**Describe the solution you'd like**  

i would like this results

action | rotate_right | 2020-01-15 17:50:25
-- | -- | --
angle | 4.22 | 2020-01-15 17:50:25
battery | ok | 2020-01-15 17:50:25
battery_level | 80 | 2020-01-15 17:50:25
from_side | 5 | 2020-01-15 14:13:42
linkquality | 39 | 2020-01-15 17:50:25
side | 2 | 2020-01-15 17:50:25
to_side | 1 | 2020-01-15 14:13:42
transmission-state | incoming publish received | 2020-01-15 17:50:25
voltage | 2965 | 2020-01-15 17:50:25

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-01-15:
It seems you are not using the latest development version.
The cube is implemented https://github.com/arendst/Tasmota/pull/7353

moneybag, 2020-01-15:
i am using
Tasmota Version | 8.1.0(tasmota)
-- | --
2020.01.12 18:59:52
2_6_1/2.2.2-dev(38a443e)

thx
robin

closed 2020-01-15 by moneybag