arendst/Tasmota #7528: Shutter-openduration/closeduration measured via Shelly2.5 energiemeter.

kugelkopf123, 2020-01-16:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
No

**Describe the solution you'd like**  
As a owner of some Shelly2.5 devices that I use for my shutters, it would be nice if the calibration of the shutteropenduration & shuttercloseduration time is measured via the energiemeter. 
Ie: The shutter is fully open. You write into the console: "getShutterTime1" and the shutter closes. At the same time a seperate Timer starts or record the time when the shutter starts moving and the timer stops as the energiemeter isn't recognize energie anymore. The timer value gets safed as shuttercloseduration. After five seconds the Shutter fully opens and the same happens for shutteropenduration. Perhaps this is an Option. I don't know how fast the energiemeter recognize  powerchanges. 

**Describe alternatives you've considered**  
-

**Additional context**  
-

**(Please, remember to close the issue when the problem has been addressed)**  
Ok

ascillato, 2020-01-16:
See https://github.com/arendst/Tasmota/issues/7509#issuecomment-575211932

As the power measurement takes time, it won't be enough fast to do a proper time calibration. Instead, you should use the calibration procedure explained in the docs.

Thanks for sharing your ideas.

closed 2020-01-16 by ascillato2

joba-1, 2020-01-20:
For reference:

my shelly 2.5 original firmware does automatic calibration like this. It somehow immediately detects endstops have cut power. So it is not a hardware limitation.