arendst/Tasmota #7530: Wired humidity sensor compatible with the Temperature Sensor Addon for Shelly 1/1PM?

Gamester17, 2020-01-16:
**Have you looked for this feature in other issues and in the docs?**  
Yes.

**Is your feature request related to a problem? Please describe.**  
Does anyone know if there is a wired Humidity sensor like example Si7021 or AM2301 that is compatible that comes with the GPIO-adapter comes with the new Temperature Sensor Addon for Shelly 1/1PM?

https://shop.shelly.cloud/temperature-sensor-addon-for-shelly-1-1pm-wifi-smart-home-automation#312

Can I flash the Shelly 1/1PM with the latest Tosmota and then just connect a Si7021 or AM2301 sensor to a Shelly 1/1PM using their GPIO-adapter from that new Temperature Sensor Addon instead of a DS18B20 based temperature only sensor that Shelly sells for their addon?

I was hoping that Shelly themselves had plans to sell a nicely packaged Si7021 or AM2301-based wired humidity sensor would be compatible with their new GPIO-adapter for Shelly 1/1PM that comes in that Temperature Sensor Addon.

The reason that I ask is that I have a (dumb) dehumidifier and been wanting to make smart and was wondering if could manage to do so with only a single all-in-one Shelly product that would support both electricity consumption measurement like the Shelly 1PM as well as be able to via its GPIO-ports to directly connect a wired humidity sensor like the Si7021 or AM2301 sensors similar to the ones that are sold by Itead for their Sonoff TH series. For my purpose of making a smart dehumidifier, even a DHT11 or DHT22 sensor with humidity support would be better than with a sensor that only supports reading temperature.

https://www.itead.cc/wiki/Sonoff_Sensor_Si7021

Note! This is a feature request related #7469 and pre-sale question since I have not yet bought one.

PS: Sure I could use a Shelly 1PM and a separate wireless Humidity sensor and automate it that as a solution, but I think that having wired humidity sensor like Si7021 or AM2301 directly wired to Shelly 1PM and Shelly 1 would make for nicer integrated products.

**Describe the solution you'd like**  
Tosmota to support Si7021 or AM2301 sensors via the GPIO-adapter that comes with the Temperature Sensor Addon for Shelly 1/1PM

**Describe alternatives you've considered**  
Sonoff Pow or Sonoff Pow R2 with a wired Si7021 or AM2301 sensors would also work be a solution my purpose of making a smart dehumidifier but I can not find any information about connecting a wired sensor to Sonoff Pow or Sonoff Pow R2.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-01-16:
Duplicated issue

closed 2020-01-16 by ascillato2

ascillato2, 2020-01-16:
> Can I flash the Shelly 1/1PM with the latest Tosmota and then just connect a Si7021 or AM2301 sensor to a Shelly 1/1PM using their GPIO-adapter from that new Temperature Sensor Addon instead of a DS18B20 based temperature only sensor that Shelly sells for their addon?

The shelly ADD ON support was just requested. See the already opened issue. There is no support yet for that. So for now, you can't connect sensors to shelly devices using Tasmota.

> Sonoff Pow or Sonoff Pow R2 with a wired Si7021 or AM2301 sensors would also work be a solution...

Sorry, but you must read the docs. NEVER CONNECT SENSORS TO A POWER MEASURING DEVICE. You can electrocute yourself. Those devices are NOT ISOLATED and your sensor will have MAINS AC.

DO NOT PUT YOURSELF, YOUR FAMILY AND HOME AT RISK.