arendst/Tasmota #7532: Can OTA upgrade refuse upgrade if crystal freq is not match?

hamwong, 2020-01-16:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
if upgrade not match FW, device will brick till serial upload correct fw

**Describe the solution you'd like**  
currently ota with fw will refuse proceed under few circumstance lik fw 8.x >> 6.x,
can we add one more condition when crystal freq is not match also refused? as once upload wrong fw, device will brick immediately and some of the devices is difficult to perform serial connection like Huafan ss, 

**Describe alternatives you've considered**  
Nil
**Additional context**  
Nil

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-01-16:
- If you manage to find the compiled crystal freq in the OTA'ed file you're my man.
- If you can do the same in the near future gzipped OTA file you're my man again.

So in short I think it's not doable.

hamwong, 2020-01-16:
Hi @arendst 
I have no idea how do you make fw won't able to downgrade from 8 to 6, but can we add a line in my_user_config.h to remark crystal freq, and esp and identify by it?

also there is a off topic question, how may I define setoption56 and 57 default on when compile the fw? also previous version like 6.x, i can change default module by #define USER_MODULE , but it looks no longer work in 8.x, can you kindly advise me?

arendst, 2020-01-16:
Crystal freq is a compile time option and adding it to user_config.h is no guarantee that the value in user_config.h is also used during compilation.

But as I noticed before, once gzipped binaries are common there is no way to check the data. so it's up to you to not mess with the crystal frequency (as it is with any other breaking fature).

#define USER_MODULE has never been there. You should use #define MODULE.

There is no easy way to set options other than the current available ones.

hamwong, 2020-01-16:
thanks for this great work and detail explanation @arendst

closed 2020-01-16 by hamwong