arendst/Tasmota #7533: Lag MQTT Server

andreamastro, 2020-01-16:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Hi, i have a problem with my Sonoff Dual R2. I configured it with Home Assistant MQTT connection and everytime Home Assistant is not available it try to reconnect every 10 seconds but for about 5 second a cannot turn on/off the lights with buttons or from user interface . There is a way to fix this problem? Thanks
Firmware version: 8.1.0
Core version: 2_6_1/2.2.2-dev(38a443e)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff Dual R2
- [ ] Tasmota binary firmware version number used: 8.1.0
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
20:29:14 MQT: stat/tasmota-lucisala/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
20:29:14 MQT: stat/tasmota-lucisala/RESULT = {"Module":{"39":"Sonoff Dual R2"}}
20:29:14 MQT: stat/tasmota-lucisala/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"22":"Relay2"},"GPIO9":{"0":"None"},"GPIO10":{"17":"Button1"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
20:30:13 MQT: stat/tasmota-lucisala/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
20:30:13 MQT: stat/tasmota-lucisala/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
20:30:13 MQT: stat/tasmota-lucisala/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
20:30:44 MQT: stat/tasmota-lucisala/STATUS = {"Status":{"Module":39,"FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota-lucisala","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:30:44 MQT: stat/tasmota-lucisala/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:59","StartupUTC":"2020-01-16T19:24:45","Sleep":50,"CfgHolder":4617,"BootCount":4,"SaveCount":22,"SaveAddress":"F6000"}}
20:30:44 MQT: stat/tasmota-lucisala/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":7,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"336/699"}}
20:30:44 MQT: stat/tasmota-lucisala/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MASTRO-608996",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
20:30:44 MQT: stat/tasmota-lucisala/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
20:30:44 MQT: stat/tasmota-lucisala/STATUS5 = {"StatusNET":{"Hostname":"tasmota-lucisala-1809","IPAddress":"192.168.1.130","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.214","Mac":"DC:4F:22:97:67:11","Webserver":2,"WifiConfig":4}}
20:30:44 MQT: stat/tasmota-lucisala/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.214","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_976711","MqttUser":"mqttuser","MqttCount":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
20:30:44 MQT: stat/tasmota-lucisala/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 16 19:30:44 2020","Local":"Thu Jan 16 20:30:44 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:38","Sunset":"17:21"}}
20:30:44 MQT: stat/tasmota-lucisala/STATUS10 = {"StatusSNS":{"Time":"2020-01-16T20:30:44"}}
20:30:44 MQT: stat/tasmota-lucisala/STATUS11 = {"StatusSTS":{"Time":"2020-01-16T20:30:44","Uptime":"0T00:05:59","UptimeSec":359,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"MASTRO-608996","BSSId":"5C:E2:8C:7E:2E:F6","Channel":3,"RSSI":100,"Signal":-42,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
20:26:02 MQT: Attempting connection...
20:26:07 MQT: Connect failed to 192.168.1.214:1883, rc -2. Retry in 10 sec

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Configure MQTT connection and turn off MQTT server

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-01-16:
Please, can you try the precompiled bins from thehackbox.org/tasmota ? Seems that your selfcompiling version is using an old pubsubclient library version.

andreamastro, 2020-01-16:
I have tried self-compiled and pre-compiled version. There is no difference

ascillato2, 2020-01-17:
This is an old bug in the core that was already solved. Reference: https://github.com/arendst/Tasmota/issues/1992 (see lasts comments and PRs).

So, we have to track it down again to find where is that hung introduced in the clientcontext routines of the core.

Have you tried compiled using latest core and stage core?

andreamastro, 2020-01-17:
As i wrote in problem description i used core version 2_6_1/2.2.2-dev(38a443e). How can i compile with stage core?

ascillato, 2020-01-17:
latest core is 2.6.3 and also you can change the SDK used (that was 2.2.2 - 38a443e in your case). To change those, you have to select them in platformio_override.ini file (example in platformio_override_sample.ini) To select stage is also available on platformio_override.ini

andreamastro, 2020-01-17:
I have tried with core 2.6.2 and stage with SDK 2.2.2-dev(bb83b9b) and the problem is still present. I found that the problem does not occur when I stop Mosquitto broker in Home Assistant. It occurs when I completely turn off my Raspberry where i have installed home assistant

Rayzilt, 2020-01-18:
I can confirm that when the MQTT broker IP is not available, the firmware hangs for a brief moment when it tries to reconnect to the broker, this does not hapen when the IP is available but the broker is stopped. Hang: Any action performed, like clicking Information button or switching the relay.

I also tried with core 2.6.3/2.3.2 by changing the following and recompile

```
[core_active]
platform                  = ${core_2_6_3.platform}
build_flags               = ${core_2_6_3.build_flags}

[core_2_6_3]
; *** Esp8266 core for Arduino version 2.6.3
platform                  = espressif8266@2.3.2
```
Tasmota information:
```
Program Version	        8.1.0.4(tasmota)
Build Date & Time	2020-01-18T12:18:38
Core/SDK Version	2_6_3/2.2.2-dev(38a443e)
```
Please let me know if further assistance is needed for testing

andreamastro, 2020-01-19:
I have tried with core 2.6.3/2.3.2 and the problem is still present

andreamastro, 2020-01-23:
Is there a different way to fix this problem?

richard-scott, 2020-01-24:
I have had similar (currently on Tasmota v8.1.0.3).

My fix was to disable WMM on my router, and ensure the router could support the 802.11G WiFi protocol. If I configured the router to use only 802.11N (excluding G) then my phones and laptops connected, but not my esp devices. I had the same issues with esp8266 devices running ESPHome too.

wongnam, 2020-01-26:
I also encountered same issue on my sonoff basic and sonoff s22 after i flashed them from ver 7.1.1.1 to 8.1.0.4

I already fixed them by following step as my workaround:
1- on web UI->Backup configuration
2- Reset configuration
3- and "must" power down them and power up again
4- Restore configuration
then my sonoff devices go back to normal operation without any lag of MQTT communication.
I hope these info can help to solve your devices issue.

Jason2866, 2020-02-07:
@ascillato  i do not have a lag when mqtt broker is not online.

arendst, 2020-02-07:
@Jason2866 pls try te attached replacements for the PubSubClient files in the lib folder

[PubSubClient.zip](https://github.com/arendst/Tasmota/files/4171184/PubSubClient.zip)

And let me know if anything changed for the good.

andreamastro, 2020-02-07:
Nothing has changed. As I said I found that the problem does not occur when I stop Mosquitto broker in Home Assistant. It occurs when I completely turn off my Raspberry where i have installed home assistant. I think it happens because Home Assistant IP is unreachable

Jason2866, 2020-02-07:
@andreamastro  This is not a mqtt lag problem. Your lags have other reasons. Please close this issue.

ascillato2, 2020-02-10:
Hi,

After several tests, we could experiment this issue, but only using old arduino cores, this bug is already fixed in latests cores.

Please, update to latest Tasmota. Thanks.

closed 2020-02-10 by ascillato2

ascillato2, 2020-02-10:
Precompiled Tasmota with latest core can be found at http://thehackbox.org/tasmota/pre-2.7/

andreamastro, 2020-02-10:
I also need a version for blinds and shutter. Where can I get it?

ascillato, 2020-02-10:
You need to self compile it

andreamastro, 2020-02-10:
Ok, thanks!

andreamastro, 2020-02-10:
Which code lines i have to add to compile with latest core version?

ascillato, 2020-02-10:
> Which code lines i have to add to compile with latest core version?

Please, rename the **platformio_override_sample.ini** file to **platformio_override.ini** in order to use latest Arduino Core. That is all.

Doing that, you will activate the latest core as the override file states:

```ini
[core_active]
; Select one core set for platform and build_flags
;platform                  = ${core_2_6_1.platform}
;build_flags               = ${core_2_6_1.build_flags}

;platform                  = ${core_2_6_3.platform}
;build_flags               = ${core_2_6_3.build_flags}

platform                  = ${tasmota_core_stage.platform}
platform_packages         = ${tasmota_core_stage.platform_packages}
build_flags               = ${tasmota_core_stage.build_flags}

;platform                  = ${core_stage.platform}
;platform_packages         = ${core_stage.platform_packages}
;build_flags               = ${core_stage.build_flags}
```

If you need further assistance on that, please address this to the Tasmota support chat. Thanks.