arendst/Tasmota #7534: RF Bridge stops sending RF codes after ~2 weeks. Red RF LED stuck on

friendodevil, 2020-01-16:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
RF Bridge stops sending RF codes after a ~2 weeks. Red RF LED stuck on. Only power-on-reset fixes/recovers

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff RF Bridge
- [x] Tasmota binary firmware version number used: 7.1.2
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmota fw upgrade
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
21:10:59 CMD: Backlog Template; Module; GPIO 255
21:10:59 MQT: stat/sonoff_MSSR1/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
21:10:59 MQT: stat/sonoff_MSSR1/RESULT = {"Module":{"25":"Sonoff Bridge"}}
21:11:00 MQT: stat/sonoff_MSSR1/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"148":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"149":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
21:11:03 MQT: stat/sonoff_MSSR1/STATUS = {"Status":{"Module":25,"FriendlyName":["Tasmota"],"Topic":"sonoff_MSSR1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:11:03 MQT: stat/sonoff_MSSR1/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power on","Uptime":"0T00:43:12","StartupUTC":"2020-01-16T19:27:51","Sleep":50,"CfgHolder":4617,"BootCount":10,"SaveCount":19,"SaveAddress":"F9000"}}
21:11:03 MQT: stat/sonoff_MSSR1/STATUS2 = {"StatusFWR":{"Version":"7.1.2(tasmota)","BuildDateTime":"2019-12-06T11:22:22","Boot":7,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285"}}
21:11:03 MQT: stat/sonoff_MSSR1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Terrapin Station",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
21:11:03 MQT: stat/sonoff_MSSR1/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
21:11:03 MQT: stat/sonoff_MSSR1/STATUS5 = {"StatusNET":{"Hostname":"sonoff_MSSR1-2001","IPAddress":"10.0.1.115","Gateway":"10.0.1.138","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.138","Mac":"DC:4F:22:FA:47:D1","Webserver":2,"WifiConfig":4}}
21:11:03 MQT: stat/sonoff_MSSR1/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.150","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FA47D1","MqttUser":"mqttusr","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
21:11:03 MQT: stat/sonoff_MSSR1/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 16 20:11:03 2020","Local":"Thu Jan 16 21:11:03 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:38","Sunset":"17:21"}}
21:11:04 MQT: stat/sonoff_MSSR1/STATUS10 = {"StatusSNS":{"Time":"2020-01-16T21:11:03"}}
21:11:04 MQT: stat/sonoff_MSSR1/STATUS11 = {"StatusSTS":{"Time":"2020-01-16T21:11:04","Uptime":"0T00:43:13","UptimeSec":2593,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Terrapin Station","BSSId":"2C:FD:A1:00:30:C0","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
21:12:49 MQT: stat/sonoff_MSSR1/STATUS = {"Status":{"Module":25,"FriendlyName":["Tasmota"],"Topic":"sonoff_MSSR1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:12:49 MQT: stat/sonoff_MSSR1/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power on","Uptime":"0T00:44:58","StartupUTC":"2020-01-16T19:27:51","Sleep":50,"CfgHolder":4617,"BootCount":10,"SaveCount":19,"SaveAddress":"F9000"}}
21:12:49 MQT: stat/sonoff_MSSR1/STATUS2 = {"StatusFWR":{"Version":"7.1.2(tasmota)","BuildDateTime":"2019-12-06T11:22:22","Boot":7,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285"}}
21:12:49 MQT: stat/sonoff_MSSR1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Terrapin Station",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
21:12:49 MQT: stat/sonoff_MSSR1/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
21:12:49 MQT: stat/sonoff_MSSR1/STATUS5 = {"StatusNET":{"Hostname":"sonoff_MSSR1-2001","IPAddress":"10.0.1.115","Gateway":"10.0.1.138","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.138","Mac":"DC:4F:22:FA:47:D1","Webserver":2,"WifiConfig":4}}
21:12:49 MQT: stat/sonoff_MSSR1/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.150","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FA47D1","MqttUser":"mqttusr","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
21:12:49 MQT: stat/sonoff_MSSR1/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 16 20:12:49 2020","Local":"Thu Jan 16 21:12:49 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:38","Sunset":"17:21"}}
21:12:49 MQT: stat/sonoff_MSSR1/STATUS10 = {"StatusSNS":{"Time":"2020-01-16T21:12:49"}}
21:12:49 MQT: stat/sonoff_MSSR1/STATUS11 = {"StatusSTS":{"Time":"2020-01-16T21:12:49","Uptime":"0T00:44:58","UptimeSec":2698,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Terrapin Station","BSSId":"2C:FD:A1:00:30:C0","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
21:13:30 MQT: stat/sonoff_MSSR1/RESULT = {"WebLog":4}
21:13:30 CFG: Saved to flash at F8, Count 20, Bytes 4096
21:13:35 HTP: Command
21:13:35 SRC: WebCommand from 10.0.1.150
21:13:35 CMD: Group 0, Index 1, Command "STATUS", Data "0"

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
This is the 3rd time this happens to me, it happens every ~2 weeks

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The RF bridge is programmed with Portisch RF firmware. I use the RF Bridge to send RF code to my garage door for open/close operation from home assistance which works fine until the RF Bridge stops sending the command every ~2wks. Send code below

```
  - service: mqtt.publish
    data_template:
      topic: 'cmnd/sonoff_MSSR1/backlog'
      qos: 1
      retain: false
      payload: > 
          rfraw AA B0 15 03 08 0190 029E 30CA 01001100110101010101010102 55; rfraw 0
```

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-01-16:
You need to update your Portisch RF firmware with latest from Tasmota Repository. There was a faulty portish version. The latest version is Ok.

ascillato2, 2020-01-17:
Please, tell us if updating solves your issue. Thanks.

closed 2020-01-17 by ascillato2

friendodevil, 2020-01-17:
i'm already using the latest Portisch firmware (0x04)

ascillato, 2020-01-17:
Please, try this one :

https://github.com/arendst/Tasmota/blob/development/tools/fw_efm8bb1/RF-Bridge-EFM8BB1-20181127.hex

friendodevil, 2020-01-17:
thanks Adrian, that's the one I have loaded

Jason2866, 2020-01-19:
Your hardware is faulty

friendodevil, 2020-01-19:
@Jason2866, how did you conclude hw is faulty?

Jason2866, 2020-01-19:
Because i run 2 RF Bridges without issues.

friendodevil, 2020-01-19:
what's your furthest device from the RF bridge?

Jason2866, 2020-01-19:
20 meters

arendst, 2020-01-20:
Try another power supply

friendodevil, 2020-01-20:
Thanks Theo, I'll advise back if its more stable

friendodevil, 2020-01-24:
running well for a few days now with a replacement 5V USB power supply