arendst/Tasmota #7537: DHCPDISCOVER every 5 minutes

brownleeb, 2020-01-16:
### PROBLEM DESCRIPTION
I am still investigating, but figured I would post while I worked on it.

I am using Tasmota-lite (v8.1) on a Gosund WP3. When I clear settings (RESET 5), the device gets a DHCP address, and seems to follow the server lease time (24h in my case). However, after the first lease renewal approx 12h later, it requests a lease approx every 5 minutes. Rebooting does not go back to the server config lease time, it immediately starts with the 5 minute renewals...

One weird thing may be that it send DHCPREQUEST and immediately a DHCPDISCOVER Then the handshake proceeds. It looks like it might be requesting a lease renewal (without going through the whole DORA process) but it doesnt wait long enough for the server to answer the initial request. The server ACKs the request, but the device has already sent the DISCOVER.

I have verified with wireshark that the lease time being sent to the device is correct. The device just seems to ignore it after the first renewal...

I am using dnsmasq (v2.80). My Tasmoda devices are the only one exhibiting this behavior. All over devices wait the standard "one half of lease time" to start requesting again. I have verified with wireshark that the devices are being given the proper initial lease time. Waiting on one to
I suspect that not a lot of people are checking their DHCP logs (probably on a networking device) so not sure how many even know this is happening...

### REQUESTED INFORMATION
I dont see any config values in my_user_config.h or user_config_override.h...

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-01-17:
Hi,

That is managed by the arduino core. Please, address this issue in the arduino esp8266 core repository. Thanks.

closed 2020-01-17 by ascillato2

brownleeb, 2020-01-17:
I doubt it will matter for this issue, but the current version is 2.6.3.
Tasmoda docs say it is currently built with 2.6.1.

Jason2866, 2020-01-19:
Tasmota can be build with core 2.6.3 too.