arendst/Tasmota #7543: H801 RGB lights and Alexa: failure with black and white

daekka, 2020-01-18:
### PROBLEM DESCRIPTION
When I tell Alexa to set the lights to WHITE the lights go out 
And when i tell Alexa to set the lights to BLACK the lights go white

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): H801
- [ ] Tasmota binary firmware version number used: 8.1.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ x] Flashing tools used: Tasmotizer
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
When I tell Alexa to set the lights to WHITE the lights go out and concole displays the following information:
`07:38:34 MQT: stat/H801_1/RESULT = {"POWER": "ON", "Dimmer":100, "Color": "0000009F60", "HSBColor": "0,0,0", "Channel":[0,0,0,62,38], "CT":284}`

When I tell Alexa to set the lights to BLACK the lights turn white and console display the following information:
`07:38:23 MQT: stat/H801_1/RESULT = {"POWER": "ON", "Dimmer":100, "Color": "FFFFFF0000", "HSBColor": "0,0,100", "Channel":[100,100,100,0,0], "CT":284}`

The other colours work perfectly.

  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ z] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
08:01:59 CMD: status 0
08:01:59 MQT: stat/H801_1/STATUS = {"Status":{"Module":20,"FriendlyName":["Salon"],"Topic":"H801_1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:01:59 MQT: stat/H801_1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T08:02:01","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":8,"SaveCount":79,"SaveAddress":"F5000"}}
08:01:59 MQT: stat/H801_1/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"355/699"}}
08:01:59 MQT: stat/H801_1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["W.O.P.R.",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
08:01:59 MQT: stat/H801_1/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
08:01:59 MQT: stat/H801_1/STATUS5 = {"StatusNET":{"Hostname":"H801","IPAddress":"192.168.1.206","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:41:CC:A7","Webserver":2,"WifiConfig":4}}
08:01:59 MQT: stat/H801_1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.200","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_41CCA7","MqttUser":"main","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
08:01:59 MQT: stat/H801_1/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 08:01:59 1970","Local":"Thu Jan 01 08:01:59 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
08:01:59 MQT: stat/H801_1/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T08:01:59"}}
08:01:59 MQT: stat/H801_1/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T08:01:59","Uptime":"0T08:02:01","UptimeSec":28921,"Heap":22,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Color":"0004FF0000","HSBColor":"239,100,100","Channel":[0,2,100,0,0],"CT":284,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"W.O.P.R.","BSSId":"38:2C:4A:A4:61:40","Channel":4,"RSSI":100,"Signal":-29,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-01-18:
Have you are link where the commands `white` and `black` for Alexa are mentioned?
Afaik this commands does not exists

daekka, 2020-01-18:
I have emulated the H801 as a Hue Brigde.
And when I say 
"Alexa, put the room in black or white", the problem occurs.

Jason2866, 2020-01-18:
I dont think this is a correct command for Alexa.
Thinking this i asked you for a link where it is officialy said from Amazon that this commands are there

daekka, 2020-01-18:
This is the skill...
[image]

s-hadinger, 2020-01-18:
Tasmota Hue emulation does not use the Philips skill.

I never tried to ask black as a color. I'm surprised it works. 

Please do `weblog 4` and report the logs at the moment you send the Alexa commands.

daekka, 2020-01-19:
Alexa, set the Salon light to White:
`
07:04:16 CMD: weblog 4
07:04:16 MQT: stat/H801_1/RESULT = {"WebLog":4}
07:04:16 CFG: Saved to flash at F9, Count 123, Bytes 4096
07:04:31 WIF: Checking connection...
07:04:31 WIF: Connected
07:04:32 HTP: Hue API (/41cca7/lights/68995697/state)
07:04:32 HTP: Hue POST args ({"ct":284})
07:04:32 SRC: Light
07:04:32 MQT: stat/H801_1/RESULT = {"POWER":"ON","Dimmer":88,"Color":"0000008C54","HSBColor":"0,0,0","Channel":[0,0,0,55,33],"CT":284}
07:04:32 HTP: Hue Result ([{"success":{"/lights/1/state/ct":284}}])
07:04:32 HTP: Hue API (/41cca7/lights/68995697)
07:04:32 /lights path=/lights/68995697
07:04:32 HTP: Hue User-Agent: Dalvik/2.1.0 (Linux; U; Android 7.1.2; AEOCK Build/NS6431), gen=2
07:04:32 HTP: Hue Result ({"state":{"on":true,"bri":224,"colormode":"ct","xy":[0.31273,0.32902],"hue":0,"sat":0,"ct":284,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Salon","modelid":"LCT007","uniqueid":"60:01:94:41:cc:a7:00:11-1","swversion":"5.50.1.19085"})
07:04:33 CFG: Saved to flash at F8, Count 124, Bytes 4096
07:04:50 UDP: Packet (175)
07:04:51 WIF: Checking connection...
07:04:51 WIF: Connected
07:04:51 UDP: Packet (175)
07:05:10 MQT: tele/H801_1/STATE = {"Time":"1970-01-02T07:05:10","Uptime":"1T07:05:12","UptimeSec":111912,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":88,"Color":"0000008C54","HSBColor":"0,0,0","Channel":[0,0,0,55,33],"CT":284,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"W.O.P.R.","BSSId":"38:2C:4A:A4:61:40","Channel":4,"RSSI":100,"Signal":-29,"LinkCount":1,"Downtime":"0T00:00:06"}}
07:05:11 WIF: Checking connection...
07:05:11 WIF: Connected
`

Alexa, set the Salon light to BLACK
`
07:05:24 HTP: Hue API (/41cca7/lights/68995697/state)
07:05:24 HTP: Hue POST args ({"hue":0,"sat":0})
07:05:24 MQT: stat/H801_1/RESULT = {"POWER":"ON","Dimmer":88,"Color":"E0E0E00000","HSBColor":"0,0,88","Channel":[88,88,88,0,0],"CT":284}
07:05:24 HTP: Hue Result ([{"success":{"/lights/1/state/hue":0}},{"success":{"/lights/1/state/sat":0}}])
07:05:24 HTP: Hue API (/41cca7/lights/68995697)
07:05:24 /lights path=/lights/68995697
07:05:24 HTP: Hue User-Agent: Dalvik/2.1.0 (Linux; U; Android 7.1.2; AEOCK Build/NS6431), gen=2
07:05:24 HTP: Hue Result ({"state":{"on":true,"bri":224,"colormode":"hs","xy":[0.31273,0.32902],"hue":0,"sat":0,"ct":284,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Salon","modelid":"LCT007","uniqueid":"60:01:94:41:cc:a7:00:11-1","swversion":"5.50.1.19085"})
07:05:25 CFG: Saved to flash at F7, Count 125, Bytes 4096
07:05:32 WIF: Checking connection...
07:05:32 WIF: Connected
07:05:36 CMD: weblog 0
07:05:36 SRC: WebConsole from 192.168.1.97
07:05:36 CMD: Group 0, Index 1, Command "WEBLOG", Data "0"
`

s-hadinger, 2020-01-19:
Ok, I see now. Your template contains 5 channels, this is why when you ask Alexa for white, it switches off channels 1-3 and fires channels 4-5 for white (cold/warm). When you set to black, it goes back to RGB.

If you have only RGB leds, change the template to set only 3 PWM GPIOs, not 5. Once done, remove the device Alexa through the Alexa app, and trigger a new Alexa discovery.

closed 2020-01-22 by ascillato2