arendst/Tasmota #7548: HA discovery template creates invalid entries when metric name contains a period

cnf, 2020-01-18:
### PROBLEM DESCRIPTION
On certain peripherals (in my case SDS011), the metric name contains a period. With the home-assistant auto discovery, this ends up with invalid json as the template: `,"val_tpl":"{{value_json['SDS0X1'].PM2.5}}"` (can't have a dot in the value name, because that is the separator)

The problem is, I think here: https://github.com/arendst/Tasmota/blob/bacc9e4899c9eb2527d6e146b7d6496b22091418/tasmota/xdrv_12_home_assistant.ino#L142

### Proposed fix
Having this set to `",\"val_tpl\":\"{{value_json['%s']['%s']}}\"";`  instead seems to fix the problem.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): adafruit ESP8266
- [x] Tasmota binary firmware version number used: 8.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
16:58:35 MQT: tester/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
16:58:36 MQT: tester/stat/RESULT = {"Module":{"0":"Generic"}}
16:58:36 MQT: tester/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"101":"SDS0X1 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"70":"SDS0X1 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
No rules used
```
- [x] Provide the output of this command: `Status 0`:
```
16:59:26 MQT: tester/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["tester"],"Topic":"tester","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:59:26 MQT: tester/stat/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T03:56:32","StartupUTC":"2020-01-18T12:02:54","Sleep":50,"CfgHolder":4617,"BootCount":13,"SaveCount":28,"SaveAddress":"F8000"}}
16:59:26 MQT: tester/stat/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"350/699"}}
16:59:26 MQT: tester/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IDIoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00000000"]}}
16:59:26 MQT: tester/stat/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":24,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
16:59:26 MQT: tester/stat/STATUS5 = {"StatusNET":{"Hostname":"tester-7615","IPAddress":"192.168.x.x","Gateway":"192.168.x.x","Subnetmask":"255.255.255.0","DNSServer":"192.168.7.1","Mac":"x:x:x:x:x:x","Webserver":2,"WifiConfig":4}}
16:59:26 MQT: tester/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.x.x","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B29DBF","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
16:59:26 MQT: tester/stat/STATUS7 = {"StatusTIM":{"UTC":"Sat Jan 18 15:59:26 2020","Local":"Sat Jan 18 16:59:26 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:36","Sunset":"17:24"}}
16:59:26 MQT: tester/stat/STATUS10 = {"StatusSNS":{"Time":"2020-01-18T16:59:26","SDS0X1":{"PM2.5":1.5,"PM10":5.1}}}
16:59:26 MQT: tester/stat/STATUS11 = {"StatusSTS":{"Time":"2020-01-18T16:59:26","Uptime":"0T03:56:32","UptimeSec":14192,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xxxx","BSSId":"xxxx","Channel":6,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
 Not relevant
```

### TO REPRODUCE
Use a peripheral like SDS011 with `SetOption19 1`

### EXPECTED BEHAVIOUR
valid JSON

### SCREENSHOTS
not applicable

### ADDITIONAL CONTEXT
MQTT Discovery topic: `homeassistant/sensor/B29DBF_SDS0X1_PM2.5/config`

```json
{"name":"tester SDS0X1 PM2.5","stat_t":"~SENSOR","avty_t":"~LWT","frc_upd":true,"pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"B29DBF_SDS0X1_PM2.5","device":{"identifiers":["B29DBF"],"connections":[["mac","x:x:x:x:x:x"]]},"~":"tester/tele/","unit_of_meas":" ","val_tpl":"{{value_json['SDS0X1'].PM2.5}}"}
```

As a sidenote, there is also a bug in home-assistant that stops discovery: https://github.com/home-assistant/home-assistant/issues/30940

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-01-18:
Hi, thanks for reporting this bug. The proposed fix brakes the compatibility with the rest of the sensors. The bug is not in the autodiscovery. The bug is in the driver of the sensor. The name of the sensor has a dot and that is invalid for sensor's name.
Will fix it. Thanks.

cnf, 2020-01-18:
cool, thanks!

ascillato, 2020-01-18:
Proposed PR #7549

cnf, 2020-01-18:
Nice @ascillato ! 

Will that also affect the discovery topic? Right now that is `homeassistant/sensor/B29DBF_SDS0X1_PM2.5/config` but I could not figure out where that is set.

ascillato2, 2020-01-18:
You will need to do setoption19 0, then update the firmware, and then do setoption19 1

ascillato2, 2020-01-18:
Yes. The change in the driver will affect all topics. That is why you need first to delete the discovery settings from your broker (by setoption19 0)

cnf, 2020-01-18:
Right, I was suspecting this would be the case, but I could not figure out where in the code this was done. Thanks again \o

Jason2866, 2020-01-18:
@ascillato  The PR breaks backwards compability. A `.` is a valid character in JSON.
If something is not working it has to be fixed in HA AD

cnf, 2020-01-18:
@Jason2866 except tasmota is sending out invalid json...

ascillato, 2020-01-18:
@Jason2866 the problem is how home assistant parses a json. The dot makes issues in home assistant side. So we can avoid that just changing the dot in json names to an underscore

ascillato2, 2020-01-18:
I'm closing the PR. I need to test another approach.

cnf, 2020-01-18:
Would my original proposal not let you keep the dot, and send valid templates for names that contain one? It should not change how the parsing is done for valid entries, as both are functionally identical for valid entries.

Jason2866, 2020-01-18:
@ascillato  imho the approach from @cnf  is the way to go. HA has problems with, so let us
try to change (only) the part where problems occour.

ascillato, 2020-01-18:
Yes, need testing

effelle, 2020-01-20:
@ascillato the SDS is the only sensor that has this issue? 
Perhaps adding a `|replace('.', '_')` on `val_tpl` will solve.

Jason2866, 2020-01-20:
@effelle  The PMS 5003/7003 driver has dots in the Json too.

sfromis, 2020-01-20:
IMO, `{{value_json['%s']['%s']}}`is the least problematic solution, as I do not see how this (according to the rules for JavaScript references) would break existing usage. The current implementation requires the syntax of a sensor name to be valid in JavaScript (which fails as the first char after the dot is not a letter). For valid names, the change to `['%s']` is functionally equivalent to `.%s`

Strictly speaking, I do not know if HomeAssistant could be expecting sensor names to form valid JavaScript variable names in *other* places, but with careful usage of JavaScript, this does not have to be an issue.

effelle, 2020-01-21:
> For valid names, the change to ['%s'] is functionally equivalent to .%s

Correct.
Perhaps I can change a bit the value template with another def but I would like to avoid special cases.
`value_template` is a special field and it has his own commands, sadly not well documented. I'll have a look.

effelle, 2020-01-25:
@cnf would you gently test this experimental build and report if the problem is solved?
Please focus just on how SDS is rendered and ignore any other issue. ;-)
[tasmota_experimental.zip](https://github.com/arendst/Tasmota/files/4112997/tasmota_experimental.zip)

Thanks.

cnf, 2020-01-27:
@effelle OTA upgrade with that fails, is this expected?

effelle, 2020-01-27:
Yes, you need either to install a minimal and then the final .bin or write it trough serial.

sgrzys, 2020-01-28:
I uploaded your tasmota_experimental, but HA still doesn't discover the PM2.5. 
I used SetOption19 0 and 1. The value_template looks OK now:  "val_tpl":"{{value_json['SDS0X1']['PM2.5']}}" but seems that it doesn't help to register new entity.
```
15:42:06 MQT: homeassistant/sensor/2F8183_SDS0X1_PM2.5/config = {"name":"Wemos PM sensor SDS0X1 PM2.5","stat_t":"~SENSOR","avty_t":"~LWT","frc_upd":true,"pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"2F8183_SDS0X1_PM2.5","device":{"identifiers":["2F8183"],"connections":[["mac","2C:3A:E8:2F:81:83"]]},"~":"dom_PM2_2F8183/tele/","val_tpl":"{{value_json['SDS0X1']['PM2.5']}}"} (retained)
15:42:06 MQT: homeassistant/sensor/2F8183_SDS0X1_PM10/config = {"name":"Wemos PM sensor SDS0X1 PM10","stat_t":"~SENSOR","avty_t":"~LWT","frc_upd":true,"pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"2F8183_SDS0X1_PM10","device":{"identifiers":["2F8183"],"connections":[["mac","2C:3A:E8:2F:81:83"]]},"~":"dom_PM2_2F8183/tele/","unit_of_meas":" ","val_tpl":"{{value_json['SDS0X1'].PM10}}"} (retained)
15:42:06 MQT: homeassistant/sensor/2F8183_status/config = {"name":"Wemos PM sensor status","stat_t":"~HASS_STATE","avty_t":"~LWT","frc_upd":true,"pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"2F8183_status","device":{"identifiers":["2F8183"],"connections":[["mac","2C:3A:E8:2F:81:83"]],"name":"Wemos PM sensor","model":"Generic","sw_version":"8.1.0.4(tasmota)","manufacturer":"Tasmota"},"~":"dom_PM2_2F8183/tele/"} (retained)
15:42:13 MQT: dom_PM2_2F8183/tele/STATE = {"Time":"2020-01-28T15:42:13","Uptime":"0T01:45:06","UptimeSec":6306,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"ais-dom","BSSId":"F4:6B:EF:B6:37:72","Channel":11,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:06"}}
15:42:13 MQT: dom_PM2_2F8183/tele/HASS_STATE = {"Version":"8.1.0.4(tasmota)","BuildDateTime":"2020-01-25T19:50:48","Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Module":"Generic","RestartReason":"Power on","Uptime":"0T01:45:06","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:06","MqttCount":1,"BootCount":34,"SaveCount":64,"IPAddress":"192.168.2.27","RSSI":"78","LoadAvg":19}
15:42:13 MQT: dom_PM2_2F8183/stat/RESULT = {"Version":"8.1.0.4(tasmota)","BuildDateTime":"2020-01-25T19:50:48","Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Module":"Generic","RestartReason":"Power on","Uptime":"0T01:45:06","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:06","MqttCount":1,"BootCount":34,"SaveCount":64,"IPAddress":"192.168.2.27","RSSI":"78","LoadAvg":19}
15:42:18 MQT: dom_PM2_2F8183/tele/HASS_STATE = {"Version":"8.1.0.4(tasmota)","BuildDateTime":"2020-01-25T19:50:48","Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Module":"Generic","RestartReason":"Power on","Uptime":"0T01:45:11","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:06","MqttCount":1,"BootCount":34,"SaveCount":64,"IPAddress":"192.168.2.27","RSSI":"76","LoadAvg":19}
15:42:20 MQT: dom_PM2_2F8183/tele/STATE = {"Time":"2020-01-28T15:42:20","Uptime":"0T01:45:13","UptimeSec":6313,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"ais-dom","BSSId":"F4:6B:EF:B6:37:72","Channel":11,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:06"}}
15:42:20 MQT: dom_PM2_2F8183/tele/SENSOR = {"Time":"2020-01-28T15:42:20","SDS0X1":{"PM2.5":7.5,"PM10":14.7}}
```

effelle, 2020-01-28:
@sgrzys , I need to see your HA mqtt logs (perhaps you need to change the log level for that) because the json test report the code as valid:
```
{ 
   "name":"Wemos PM sensor SDS0X1 PM2.5",
   "stat_t":"~SENSOR",
   "avty_t":"~LWT",
   "frc_upd":true,
   "pl_avail":"Online",
   "pl_not_avail":"Offline",
   "uniq_id":"2F8183_SDS0X1_PM2.5",
   "device":{ 
      "identifiers":[ 
         "2F8183"
      ],
      "connections":[ 
         [ 
            "mac",
            "XX:XX:XX:XX:XX:XX"
         ]
      ]
   },
   "~":"dom_PM2_2F8183/tele/",
   "val_tpl":"{{value_json['SDS0X1']['PM2.5']}}"
}
```
And query the sensor directly under HA works too:
```
{% set value_json = {"Time":"2020-01-28T15:42:20","SDS0X1":{"PM2.5":7.5,"PM10":14.7}}
%}
{{value_json['SDS0X1']['PM2.5']}}

7.5 
```
`2F8183_SDS0X1_PM2.5` is present under discovered list? Did you have any reading or the value is unknown / empty? 

> PS: I will be nice to add the unit by default:
unit_of_measurement: "µg/m³"

Not possible without addding a special case for it. As for now the discovery is using the fallback `HASS_DISCOVER_SENSOR_ANY` to generate the code. Perhaps in the future.

sgrzys, 2020-01-28:
Yes, the json is correct - I can take the values from manual sensor configuration with:
`value_template: "{{ value_json['SDS0X1']['PM2.5'] }}" `

But the discovery list has only PM10 (never discovered PM2.5):
[image]

not sure about this logs from mosquito - this is just after use setoption19 1:
```
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r0, m0, 'dom_PM2_2F8183/stat/RESULT', ... (20 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/light/2F8183_LI_1/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/light/2F8183_LI_1/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/switch/2F8183_RL_1/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/switch/2F8183_RL_1/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/light/2F8183_LI_2/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/light/2F8183_LI_2/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/switch/2F8183_RL_2/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/switch/2F8183_RL_2/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/light/2F8183_LI_3/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/light/2F8183_LI_3/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/switch/2F8183_RL_3/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/switch/2F8183_RL_3/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/light/2F8183_LI_4/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/light/2F8183_LI_4/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/switch/2F8183_RL_4/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/switch/2F8183_RL_4/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/light/2F8183_LI_5/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/light/2F8183_LI_5/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/switch/2F8183_RL_5/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/switch/2F8183_RL_5/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/light/2F8183_LI_6/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/light/2F8183_LI_6/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/switch/2F8183_RL_6/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/switch/2F8183_RL_6/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/light/2F8183_LI_7/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/light/2F8183_LI_7/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/switch/2F8183_RL_7/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/switch/2F8183_RL_7/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/light/2F8183_LI_8/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/light/2F8183_LI_8/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/switch/2F8183_RL_8/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/switch/2F8183_RL_8/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/binary_sensor/2F8183_BTN_1/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/binary_sensor/2F8183_BTN_1/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/binary_sensor/2F8183_BTN_2/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/binary_sensor/2F8183_BTN_2/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/binary_sensor/2F8183_BTN_3/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/binary_sensor/2F8183_BTN_3/config', ... (0 bytes))
1580226085: Received PUBLISH from DOM_2F8183 (d0, q0, r1, m0, 'homeassistant/binary_sensor/2F8183_BTN_4/config', ... (0 bytes))
1580226085: Sending PUBLISH to 7FzHdHvIEU21lWJTNZ9oOq (d0, q0, r0, m0, 'homeassistant/binary_sensor/2F8183_BTN_4/config', ... (0 bytes))
[... cut ...]
1580226086: Received SUBSCRIBE from 7FzHdHvIEU21lWJTNZ9oOq
1580226086: 	dom_PM2_2F8183/tele/HASS_STATE (QoS 0)
1580226086: 7FzHdHvIEU21lWJTNZ9oOq 0 dom_PM2_2F8183/tele/HASS_STATE
1580226086: Sending SUBACK to 7FzHdHvIEU21lWJTNZ9oOq
1580226086: Received SUBSCRIBE from 7FzHdHvIEU21lWJTNZ9oOq
1580226086: 	dom_PM2_2F8183/tele/HASS_STATE (QoS 0)
1580226086: 7FzHdHvIEU21lWJTNZ9oOq 0 dom_PM2_2F8183/tele/HASS_STATE
1580226086: Sending SUBACK to 7FzHdHvIEU21lWJTNZ9oOq
1580226092: Received PINGREQ from DOM_2F8183
1580226092: Sending PINGRESP to DOM_2F8183
```

effelle, 2020-01-28:
You need to check the logs under HA not under mosquitto, because as you confirmed the json is valid.
Look on logs for something like `Unable to parse JSON Wemos PM sensor SDS0X1 PM2.5:`.
If you like you can try directly this other build, since I'm working on sensors right now.
[Tasmota-experimental.zip](https://github.com/arendst/Tasmota/files/4123517/Tasmota-experimental.zip)

sgrzys, 2020-01-28:
Logs form HA with debug level - no errors:
```
1|ais      | 2020-01-28 22:51:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on homeassistant/sensor/2F8183_SDS0X1_PM2.5/config: b'{"name":"Wemos PM sensor SDS0X1 PM2.5","stat_t":"~SENSOR","avty_t":"~LWT","frc_upd":true,"pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"2F8183_SDS0X1_PM2.5","device":{"identifiers":["2F8183"],"connections":[["mac","2C:3A:E8:2F:81:83"]]},"~":"dom_PM2_2F8183/tele/","val_tpl":"{{value_json[\'SDS0X1\'][\'PM2.5\']}}"}'
1|ais      | 2020-01-28 22:51:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on homeassistant/sensor/2F8183_SDS0X1_PM10/config: b'{"name":"Wemos PM sensor SDS0X1 PM10","stat_t":"~SENSOR","avty_t":"~LWT","frc_upd":true,"pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"2F8183_SDS0X1_PM10","device":{"identifiers":["2F8183"],"connections":[["mac","2C:3A:E8:2F:81:83"]]},"~":"dom_PM2_2F8183/tele/","unit_of_meas":" ","val_tpl":"{{value_json[\'SDS0X1\'].PM10}}"}'
1|ais      | 2020-01-28 22:51:32 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on homeassistant/sensor/2F8183_status/config: b'{"name":"Wemos PM sensor status","stat_t":"~HASS_STATE","avty_t":"~LWT","frc_upd":true,"pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json[\'RSSI\']}}","ic":"mdi:information-outline","uniq_id":"2F8183_status","device":{"identifiers":["2F8183"],"connections":[["mac","2C:3A:E8:2F:81:83"]],"name":"Wemos PM sensor","model":"Generic","sw_version":"8.1.0.4(tasmota)","manufacturer":"Tasmota"},"~":"dom_PM2_2F8183/tele/"}'
1|ais      | 2020-01-28 22:51:32 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: sensor 2F8183_SDS0X1_PM10, sending update
1|ais      | 2020-01-28 22:51:32 INFO (MainThread) [homeassistant.components.mqtt.discovery] Component has already been discovered: sensor 2F8183_status, sending update
1|ais      | 2020-01-28 22:51:32 INFO (MainThread) [homeassistant.components.mqtt] Got update for entity with hash: ('sensor', '2F8183_SDS0X1_PM10') '{'name': 'Wemos PM sensor SDS0X1 PM10', 'state_topic': 'dom_PM2_2F8183/tele/SENSOR', 'availability_topic': 'dom_PM2_2F8183/tele/LWT', 'force_update': True, 'payload_available': 'Online', 'payload_not_available': 'Offline', 'unique_id': '2F8183_SDS0X1_PM10', 'device': {'identifiers': ['2F8183'], 'connections': [['mac', '2C:3A:E8:2F:81:83']]}, 'unit_of_measurement': ' ', 'value_template': "{{value_json['SDS0X1'].PM10}}", 'platform': 'mqtt', 'discovery_hash': ('sensor', '2F8183_SDS0X1_PM10')}'
1|ais      | 2020-01-28 22:51:32 INFO (MainThread) [homeassistant.components.mqtt] Updating component: sensor.wemos_pm_sensor_sds0x1_pm10
1|ais      | 2020-01-28 22:51:32 INFO (MainThread) [homeassistant.components.mqtt] Got update for entity with hash: ('sensor', '2F8183_status') '{'name': 'Wemos PM sensor status', 'state_topic': 'dom_PM2_2F8183/tele/HASS_STATE', 'availability_topic': 'dom_PM2_2F8183/tele/LWT', 'force_update': True, 'payload_available': 'Online', 'payload_not_available': 'Offline', 'json_attributes_topic': 'dom_PM2_2F8183/tele/HASS_STATE', 'unit_of_measurement': ' ', 'value_template': "{{value_json['RSSI']}}", 'icon': 'mdi:information-outline', 'unique_id': '2F8183_status', 'device': {'identifiers': ['2F8183'], 'connections': [['mac', '2C:3A:E8:2F:81:83']], 'name': 'Wemos PM sensor', 'model': 'Generic', 'sw_version': '8.1.0.4(tasmota)', 'manufacturer': 'Tasmota'}, 'platform': 'mqtt', 'discovery_hash': ('sensor', '2F8183_status')}'
1|ais      | 2020-01-28 22:51:32 INFO (MainThread) [homeassistant.components.mqtt] Updating component: sensor.wemos_pm_sensor_status

```

effelle, 2020-01-28:
Nothing better than an issue that don't trows error to  close a working day. -_- 
Please try the latest build I've send too, those log are from the older experimental.

sgrzys, 2020-01-28:
OK, logs from the new one:
```
1|ais      | 2020-01-28 23:18:00 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on homeassistant/sensor/2F8183_SDS0X1_PM2.5/config: b'{"name":"Wemos PM sensor SDS0X1 PM2.5","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"2F8183_SDS0X1_PM2.5","device":{"identifiers":["2F8183"],"connections":[["mac","2C:3A:E8:2F:81:83"]]},"~":"dom_PM2_2F8183/tele/","unit_of_meas":" ","val_tpl":"{{value_json[\'SDS0X1\'][\'PM2.5\']}}"}'
1|ais      | 2020-01-28 23:18:00 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on homeassistant/sensor/2F8183_SDS0X1_PM10/config: b'{"name":"Wemos PM sensor SDS0X1 PM10","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"2F8183_SDS0X1_PM10","device":{"identifiers":["2F8183"],"connections":[["mac","2C:3A:E8:2F:81:83"]]},"~":"dom_PM2_2F8183/tele/","unit_of_meas":" ","val_tpl":"{{value_json[\'SDS0X1\'][\'PM10\']}}"}'
1|ais      | 2020-01-28 23:18:00 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on homeassistant/sensor/2F8183_status/config: b'{"name":"Wemos PM sensor status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json[\'RSSI\']}}","ic":"mdi:information-outline","uniq_id":"2F8183_status","device":{"identifiers":["2F8183"],"connections":[["mac","2C:3A:E8:2F:81:83"]],"name":"Wemos PM sensor","model":"Generic","sw_version":"8.1.0.4(tasmota)","manufacturer":"Tasmota"},"~":"dom_PM2_2F8183/tele/"}'
1|ais      | 2020-01-28 23:18:00 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 2F8183_SDS0X1_PM10
1|ais      | 2020-01-28 23:18:00 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 2F8183_status
1|ais      | 2020-01-28 23:18:00 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to dom_PM2_2F8183/tele/LWT
1|ais      | 2020-01-28 23:18:00 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to dom_PM2_2F8183/tele/LWT
1|ais      | 2020-01-28 23:18:00 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to dom_PM2_2F8183/tele/SENSOR
1|ais      | 2020-01-28 23:18:00 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dom_PM2_2F8183/tele/LWT (retained): b'Online'
1|ais      | 2020-01-28 23:18:00 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to dom_PM2_2F8183/tele/HASS_STATE
1|ais      | 2020-01-28 23:18:00 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on dom_PM2_2F8183/tele/LWT (retained): b'Online'
1|ais      | 2020-01-28 23:18:00 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to dom_PM2_2F8183/tele/HASS_STATE

```

effelle, 2020-01-28:
That's funny, if the problem was the subsensor within parenthesis (but I'm sure it is not) also the PM10 should not be generated. 
I'm afraid something else is blocking mqtt discovery and I fear I can't help much more. I'll dig a bit more but that's all for now.

cnf, 2020-01-28:
Please also note https://github.com/home-assistant/home-assistant/issues/30940 which is an home-assistant bug.

Both this, and the HA bug need to be fixed for discovery to work.

I have been away, but I'll try the new image tomorrow.

effelle, 2020-01-29:
Ok, I'll check the proposed code on that issue tomorrow and eventually I'll push a commit.

cnf, 2020-01-29:
@effelle can confirm your last firmware works for my setup (patched HA code).

effelle, 2020-01-29:
Thanks to confirm Frank.
Now, have it patched on HA side is another story. I've already submitted PR there and is a looooooooong way. :P 
I know is not an issue for Tasmota, but I have no idea how other brands software/firmware will receive the new directive. 
I'll upload the new code ASAP for Tasmota, this issue will be closed but I'll keep on eye on HA issue too.

effelle, 2020-01-29:
@ascillato do we have a TAG better suiting the actual situation?

ascillato, 2020-01-29:
Like which one?

effelle, 2020-01-29:
I have no idea Adrian.
The issue is virtually solved on our side, but will not work until HA changes the python code.

cnf, 2020-01-29:
you can temporary edit the files yourself, my report mentioned above has instructions on what to do.

effelle, 2020-01-29:
I've already did that but again when we speak about MQTT discovery we speak about a service, not a Tasmota - focused thing. Works for us, but need testing for other software devs.

cnf, 2020-01-30:
Does on hold mean it won't get merged until the HA people fix their side? It would be useful if the tasmota side works, at least.

effelle, 2020-01-30:
Will be merged on next update ASAP.

closed 2020-02-07 by arendst

effelle, 2020-02-07:
@cnf Frank, 
you may want to compile the latest dev. No hacks required under HA. :-)
[image]

PR #7692 (2020-02-07): Ha Discovery: remove non alpha characters from HA sensor topic name
## Description:
Solve Discovery issue when non alpha characters are used on topic name.
 
**Related issue (if applicable):** fixes #7548

## Checklist:
  - [X] The pull request is done against the latest dev branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR.
  - [X] The code change is tested and works on core 2.6.1
  - [X] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [X] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CO [... cut]