arendst/Tasmota #7552: Rule "ON System#Boot" no longer working in 8.1

mphilipp, 2020-01-18:
### PROBLEM DESCRIPTION
It seems that the rule trigger System#Boot is no longer executed since I upgraded my Shelly to Tasmota 8.1. On another device that is still on Tasmota 6.6. the same rule is working as expected.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly 1
- [x] Tasmota binary firmware version number used: 8.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":465,"Rules":"ON System#Boot DO Backlog Var1 0; Var2 0 ENDON"}
```
- [ ] Provide the output of this command: `Status 0`:
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
00:00:00 CFG: Loaded from flash at FB, Count 33
00:00:00 QPC: Flag 0E
00:00:00 SRC: Restart
00:00:00 Project tasmota Garage Version 8.1.0(tasmota)-2_6_1
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 XXXX in mode 11N as shelly-garage...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on shelly-garage with IP address 192.168.178.50
22:19:38 NTP: Drift 0, (UTC) Sat Jan 18 21:19:38 2020, (DST) Sun Mar 29 02:00:00 2020, (STD) Sun Oct 25 03:00:00 2020
22:19:38 QPC: Reset
22:19:40 APP: Boot Count 9
22:19:40 CFG: Saved to flash at FA, Count 34, Bytes 4096
22:19:57 WIF: Checking connection...
22:19:57 WIF: Connected
22:19:59 CMD: var1
22:19:59 SRC: WebConsole from 192.168.178.46
22:19:59 CMD: Group 0, Index 1, Command "VAR", Data ""
22:19:59 RSL: stat/tasmota/RESULT = {"Var1":""}
```

### TO REPRODUCE

```
Rule1 ON System#Boot DO Backlog Var1 0; Var2 0 ENDON
Rule1 1
Restart 1
Var1
```

### EXPECTED BEHAVIOUR
On Tasmota 6.6 this returns `{"Var1":"0"}` as expected. But on Tasmota 8.1. it returns `{"Var1":""}`. Also 6.6. prints the message `RUL: SYSTEM#BOOT performs "Backlog Var1 0; Var2 0"` on the web console but on 8.1 there is no such message.

Jason2866, 2020-01-19:
Your example rule is executed at boot. Cannot reproduce the issue.

mphilipp, 2020-01-19:
Thank you for looking into this!

Your observation left me quite puzzled. Therefore I completely reset the Shelly via esptool and flashed various firmware versions via serial. 
It turned out that I was previously using sonoff-basic.bin. Apparently I missed that this has been renamed to tasmota-lite.bin and therefore accidentally flashed tasmota.bin during OTA upgrade.
When using tasmota-lite.bin v8.1.0 the `System#Boot` rule trigger is working again.

I do not fully understand how using the wrong binary can lead to this strange symptom while everything else works fine.
I'm thinking about how to prevent that this happens again.
Would it be possible to issue a warning during the OTA update process if the binary name is changed?
Or could the default OTA URL be changed to contain the binary name that is currently installed?

Regards,
Matthias

ascillato2, 2020-01-19:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

mphilipp, 2020-01-24:
@ascillato2 could you please re-open this issue?

I did some debugging and it turned out that the the "on system#boot" rule is never triggered if you use the recommended release binary with its default settings and don't configure MQTT. 

The reason is this code in MqttReconnect() in xdrv_02_mqtt.ino

```
#ifdef USE_DISCOVERY
#ifdef MQTT_HOST_DISCOVERY
        if (!strlen(SettingsText(SET_MQTT_HOST)) && !Wifi.mdns_begun) { return; }
#endif  // MQTT_HOST_DISCOVERY
#endif  // USE_DISCOVERY
```

For some reason, the system boot event is generated by the MQTT driver in MqttConnected().
The binary tasmota.bin is compiled with USE_DISCOVERY enabled, while tasmota-lite.bin is compiled without this #define.
If you do not set an MQTT host and mDNS is disabled, the function MqttConnected() never gets called and hence the "on system#boot" rule is not triggered.
Unfortunately the default settings after a fresh install of tasmota.bin match this condition and the function returns before generating the the system boot event.

I think the solution would be to make the system boot event independent from MQTT. Please have another look and comment.

Thanks and regards,
Matthias

ascillato, 2020-01-24:
As designed. SYSTEM#BOOT only triggers if mqtt is on. If you want a trigger for an earlier event prior all the initialization, you can use ON POWER1#BOOT DO ....

ascillato, 2020-01-24:
Nevertheless, @arendst what do you think of @mphilipp proposal?

arendst, 2020-01-25:
I'll make sure there will be a SYSTEM#BOOT trigger when no MQTT is enabled.

arendst, 2020-01-25:
Been looking over the code and I found the code part you mention:

> The reason is this code in MqttReconnect() in xdrv_02_mqtt.ino
``
#ifdef USE_DISCOVERY
#ifdef MQTT_HOST_DISCOVERY
        if (!strlen(SettingsText(SET_MQTT_HOST)) && !Wifi.mdns_begun) { return; }
#endif  // MQTT_HOST_DISCOVERY
#endif  // USE_DISCOVERY
``

It's not in MqttReconnect() but the more obvious MqttCheck().

And still it does trigger the SYSTEM#BOOT on a restart as can be seen here:
```
00:00:00 SRC: Restart
00:00:00 Project tasmota Wemos4 Version 8.1.0.4(tasmota)-STAGE
00:00:00 WIF: Attempting connection...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Network 0, AP1, SSId indebuurt, Channel 1, BSSId 74:83:C2:2A:F1:AC, RSSI -80, Encryption 1
00:00:04 WIF: Network 1, AP1, SSId indebuurt, Channel 11, BSSId 18:E8:29:CA:17:C1, RSSI -34, Encryption 1
00:00:04 WIF: Connecting to AP1 indebuurt in mode 11N as wemos4...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Attempting connection...
00:00:06 QPC: Reset
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Connected
00:00:07 HTP: Web server active on wemos4 with IP address 192.168.2.223
12:41:30 NTP: Drift 0, (UTC) Sat Jan 25 11:41:29 2020, (DST) Sun Mar 29 02:00:00 2020, (STD) Sun Oct 25 03:00:00 2020
12:41:30 RSL: INFO1 = {"Module":"Generic","Version":"8.1.0.4(tasmota)","FallbackTopic":"cmnd/DVES_83BB10_fb/","GroupTopic":"cmnd/tasmotas/"}
12:41:30 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"wemos4","IPAddress":"192.168.2.223"}
12:41:30 RSL: INFO3 = {"RestartReason":"External System"}
12:41:30 RUL: SYSTEM#BOOT performs "Backlog Var1 0; Var2 0"
12:41:30 SRC: Rule
12:41:30 CMD: Group 0, Index 1, Command "BACKLOG", Data "Var1 0; Var2 0"
12:41:30 SRC: Backlog
12:41:30 CMD: Group 0, Index 1, Command "VAR", Data "0"
12:41:30 RSL: VAR = {"Var1":"0"}
12:41:30 APP: Boot Count 149
12:41:30 SRC: Backlog
12:41:30 CMD: Group 0, Index 2, Command "VAR", Data "0"
12:41:30 RSL: VAR = {"Var2":"0"}
12:41:31 CFG: Saved to flash at FA, Count 399, Bytes 4096
``
In the above case the ``mqqthost`` is set none and ``SetOption3`` disabled mqtt all together.

So although the top code in MqttCheck() does not execute function MqttConnected is executed once anyway by the bottom part of MqttCheck() at a restart which will also execute the SYSTEM#BOOT trigger.

So I connot reproduce your issue either.

ascillato2, 2020-01-25:
@mphilipp 

Hi, Theo has fixed this bug. Please test it. Thanks.

closed 2020-01-25 by ascillato2

mphilipp, 2020-01-26:
Thank you very much, Theo and Adrian! Now it works as expected.

montyfert, 2023-03-13:
hi am i doing something wrong ,
Rule1 ON Power1#boot DO POWER1 1 ENDON
RULE1 1

any help would be great thanks
martin

barbudor, 2023-03-13:
Your question as nothing to do with that closed issue
Please open a new discussion
Thanks

Gordus-Arrechus, 2025-03-17:
Hi, can we reopen this case please.

I Updated to ver 14.5.0 and if MQTT Server is down or not reacheable the variables called on boot "ON system#boot DO Var1"10" dont get the value until MQTT Server is up again.

Thanks in advance for your help and/or for providing workaround

sfromis, 2025-03-17:
Don't resurrect ancient threads, especially when you obviously did not even read the [docs](https://tasmota.github.io/docs/Rules/#examples-of-available-triggers) first.

`System#boot` is specifically when Tasmota is **fully** initialized, including both Wifi and MQTT.

`System#init` is the trigger if you want something earlier than that. No workaround needed.